thingsboard/thingsboard-gateway #1025: [HELP]getting data from external broker

dejagodic, 2022-12-09:
I've been using Thingsboard on my server installed in docker, I have a device that publish data to Mosquitto broker on my other server to topic v1/devices/me/telemetry. Now I want to extract data to Thingsboard with Thingsboard gateway. I've set it up, and everything works, but I have one question. For example, my device is sending a separate message for every data from list, theres 5 data on the list, message form is "key1" : "payload1", "key2" : "payload2", ...,"key5" : "payload5". Is there any way to keep the key and payload from the messages and show it on Thingsboard instance as they were sent on Mosquitto? As for now I succeed to set it up only in form "newkey1" : ${key1}.
So basicly, i want to use TB gateway to subscribe to mosquitto topic, gets all the messages, and push them to TB broker automatically, with same key as key on mosquitto.

MQTT Connector

 - OS: Ubuntu 20.04
 - Thingsboard IoT Gateway version : latest

samson0v, 2022-12-12:
Hi @dejagodic, if I understand you correctly you can use the following configuration:
```
"timeseries": "*"
```
For example:
[image]

Sent data:
[image]

Data on ThingsBoard:
[image]

dejagodic, 2022-12-12:
> Hi @dejagodic, if I understand you correctly you can use the following configuration:
> ```
> "timeseries": "*"
> ```
> For example:
> [image]
> 
> Sent data:
> [image]
> 
> Data on ThingsBoard:
> [image]
> 
> **(Data is abstract)**
> 

That was it, thank you so much! Now to testing other stuffs 😁

dejagodic, 2022-12-13:
> Hi @dejagodic, if I understand you correctly you can use the following configuration:
> ```
> "timeseries": "*"
> ```
> For example:
> [image]
> 
> Sent data:
> [image]
> 
> Data on ThingsBoard:
> [image]
> 
> **(Data is abstract)**
> 

Just one more question, sorry. Is there any way to filter keys to send only keys with matching ending?
For example:
I have keys like this:
Temp_41
Hum_41
...
Temp_42
Hum_42
...
And now i want to pass to TB only keys ending with _41
How can I do that? Thanks again!

samson0v, 2022-12-13:
@dejagodic, you can do it using rule nodes in ThingsBoard.

dejagodic, 2022-12-13:
> @dejagodic, you can do it using rule nodes in ThingsBoard.

So there is no way to configure it somehow on ThingsBoard gateway? Since I need that messages ending with _41 go to one ThingsBoard device, messages ending with _42 to the second ThingsBoard device, and so on...
I'm considering even setting up mosquitto broker in bridge mode to ThingsBoard if I can achieve the same result, but maybe is easier to filter messages and forward them to the correct device?
Thanks again!

samson0v, 2022-12-13:
@dejagodic, with a configuration like `*` you can't add additional filtering.

dejagodic, 2022-12-13:
> @dejagodic, with a configuration like `*` you can't add additional filtering.

Is there a way than to configure it somehow instead "*" to grab only keys ending with _41 a publish them to ThingsBoard?
I'm not sure how can I do with this type of configuration ("*") on the rule chain "extract" keys with _41 and send them to one device, then extract keys with _42 send them to the second device, and so on...

samson0v, 2022-12-13:
@dejagodic, there is no other way. 
You can write your [custom converter](https://thingsboard.io/docs/iot-gateway/custom/#custom-converter-methods-reference).

dejagodic, 2022-12-13:
> 
I'll try then with the rule node and after with the custom converter. Thank you again!

closed 2022-12-18 by samson0v