thingsboard/thingsboard-gateway #1028: [HELP] Request Connector RPC getValue JSON parse problem

peterandthejetsons, 2022-12-20:
Hello,

I have the Thingsboard Gateway installed from the latest branch in a docker environment on a Ubuntu 20.04 server. I have a AC unit and I'd like to use Knob widget with RPC and get the current temperature (exactly the min_temp key value pair) with default getValue method to set Knob temperature, but I'm always get the following error message:

```
Unable to parse response:
{
    "attributes": {
        "fan_modes": [
            "Silent",
            "Low",
            "Medium",
            "High",
            "Super high",
            "Auto"
        ],
        "friendly_name": "Air Conditioner",
        "hvac_modes": [
            "off",
            "auto",
            "cool",
            "dry",
            "heat",
            "fan_only"
        ],
        "icon": "mdi:air-conditioner",
        "max_temp": 30,
        "min_temp": 17,
        "preset_modes": [
            "none",
            "comfort",
            "eco",
            "boost"
        ],
        "supported_features": 121,
        "swing_modes": [
            "Off",
            "Vertical",
            "Horizontal",
            "Both"
        ],
        "target_temp_step": 0.5
    },
    "context": {
        "id": "01GMK8GQQRCFA22BN94A2TKW14",
        "parent_id": null,
        "user_id": null
    },
    "entity_id": "climate.145135534909349_climate",
    "last_changed": "2022-12-18T18:39:09.560607+00:00",
    "last_updated": "2022-12-18T18:39:09.560607+00:00",
    "state": "unavailable"
}
```
My tb_gateway.yaml config is the following:
```
...
  "serverSideRpc": [
    {
      "deviceNameFilter": ".*",
      "methodFilter": "getValue",
      "requestUrlExpression": "api/states/${deviceName}",
      "responseTimeout": 1,
      "httpMethod": "GET",
      "valueExpression": "${params}",
      "timeout": 0.5,
      "tries": 3,
      "httpHeaders": {
        "Content-Type": "application/json",
        "Authorization": "Bearer eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiI3MjVlNDM0ODgwNjk0ODBhYjE5MjJjZWFkY2MyNmNkZiIsImlhdCI6MTY3MDc5NTIxOSwiZXhwIjoxOTg2MTU1MjE5fQ.cqwfdWU-j0MEN6X5TXbWjocjH1Yc4MYpeEbrQnA-_qg"
      }
    }
  ]
...
```
I tried some ways like:
```
{\"attributes\":{\"min_temp\":${params}}}
{attributes.min_temp.params}
```
But unfortunately none of them helped, so I'd like to ask for your help in this issue. If you need further information, please let me know.

**Connector name:**
- Request connector
- Request connector's converter

**Versions (please complete the following information):**
 - OS: Ubuntu 20.04
 - Docker: Docker version 20.10.12
 - Thingsboard IoT Gateway 3.2 main version, build from latest branch
 - Python version 3.10.6

samson0v, 2022-12-21:
Hi @peterandthejetsons, I am trying to reproduce your bug but without success.
Have the next config:
[image]

HTTP server (Fast API):
[image]

Result:
[image]

peterandthejetsons, 2022-12-21:
Hi @samson0v,

Thank you for your answer. If i trying with RPC debug console, the result is the same for me too, but i'd like to parse from the response the key's value to set the right value on the Knob controller. If i understand well the valueExpression is the field where i can give regexp to parse the needed key's value. So I'd like to get only one key value to set the Knob controllers value to right value. 

Could you help me in this issue?

Best regards, 
Peter

samson0v, 2022-12-21:
@peterandthejetsons, ok, will add this feature.

closed 2022-12-21 by imbeacon

peterandthejetsons, 2022-12-21:
Hi @samson0v,

Thank you for the fast develepment process. I tried the new function, but maybe I'm doing something wrong because now I'm get the following error:

[image]

This is my request.json section:

```
  "serverSideRpc": [
    {
      "deviceNameFilter": ".*",
      "methodFilter": "getValue",
      "requestUrlExpression": "api/states/${deviceName}",
      "responseTimeout": 1,
      "httpMethod": "GET",
      "requestValueExpression": "${params}",
      "responseValueExpression": "${*.min_temp}",
      "timeout": 0.5,
      "tries": 3,
      "httpHeaders": {
        "Content-Type": "application/json",
        "Authorization": "Bearer eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiI3MjVlNDM0ODgwNjk0ODBhYjE5MjJjZWFkY2MyNmNkZiIsImlhdCI6MTY3MDc5NTIxOSwiZXhwIjoxOTg2MTU1MjE5fQ.cqwfdWU-j0MEN6X5TXbWjocjH1Yc4MYpeEbrQnA-_qg"
      }
    }

  ]
```

Am i doing something wrong? Could you help me in this issue?

Best regards,
Peter