thingsboard/thingsboard-gateway #1029: AttributeError: module 'puresnmp' has no attribute 'table'

ApsXpert, 2022-12-20:
**I am getting this error. Can anyone please guide me. I am using the default snmp.json*

 python3[175061]: [STREAM ONLY] 2022-12-21 00:22:55,919 - ERROR - [snmp_connector.py] - snmp_connector - 150 - module 'puresnmp' has no attribute 'table'
 python3[175061]: Traceback (most recent call last):
 python3[175061]:   File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/snmp/snmp_connector.py", line 144, in __process_data
 python3[175061]:     response = self.__process_methods(method, common_parameters, datatype_config)
python3[175061]:   File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/snmp/snmp_connector.py", line 219, in __process_methods
python3[175061]:     response = puresnmp.table(**common_parameters,
python3[175061]: AttributeError: module 'puresnmp' has no attribute 'table'
python3[175061]: [STREAM ONLY] 2022-12-21 00:22:55,950 - ERROR - [statistics_service.py] - statistics_service - 67 - Statistic parameter raise the exception: /bin>
python3[175061]: [STREAM ONLY] 2022-12-21 00:22:55,953 - ERROR - [statistics_service.py] - statistics_service - 67 - Statistic parameter raise the exception: /bin>
python3[175061]: [STREAM ONLY] 2022-12-21 00:22:55,958 - ERROR - [statistics_service.py] - statistics_service - 67 - Statistic parameter raise the exception: /bin>

 - OS: [e.g. Ubuntu 18.04]
 - Thingsboard IoT Gateway version [e.g. 3.2]
 - Python version[e.g. 3.8.10]

samson0v, 2022-12-21:
Hi @ApsXpert, what version of puresnmp is installed?

samson0v, 2022-12-26:
@ApsXpert, any updates?

ApsXpert, 2022-12-27:
Sorry for late reply. I fixed this by upgrading the gateway. Gateway was using old puresnmp library methods which were not there in puresnmp.

ApsXpert, 2022-12-27:
@ApsXpert thank you for your advice

closed 2022-12-28 by samson0v