thingsboard/thingsboard-gateway #1030: Cannot read a boolean on modbus TCP[BUG]

BjornDeGloire, 2022-12-21:
I have a modbus TCP connection to a device where coils can be read out as booleans.
The problem is that the TB-gateway modbus module always reads out an array instead of a boolean as described in the documentation. In the documentation it is metioned that when you read coils (functionCode 1) and type: "bits" and put objectsCount on 1. A boolean will be read out. This does not happen, I receive an array 'telemetry': [{'inputState': [0, 0, 0, 0, 0, 0, 0, 0]}]

My modbus config file
{
  "master": {
    "slaves": [
      {
        "host": "10.69.50.15",
        "port": 502,
        "type": "tcp",
        "method": "socket",
        "timeout": 35,
        "byteOrder": "BIG",
        "wordOrder": "BIG",
        "retries": true,
        "retryOnEmpty": true,
        "retryOnInvalid": true,
        "pollPeriod": 1000,
        "unitId": 1,
        "deviceName": "Camping Place 1",
        "deviceType": "SmartPowerSocket",
        "sendDataOnlyOnChange": false,
        "connectAttemptTimeMs": 5000,
        "connectAttemptCount": 5,
        "waitAfterFailedAttemptsMs": 30000,
        "timeseries": [
          {
            "byteOrder": "BIG",
            "tag": "inputState",
            "type": "bits",
            "functionCode": 1,
            "objectsCount": 1,
            "address": 101
          }
        ]
      }
    ]
  }
}

Versions
 - OS:
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye

 - Thingsboard IoT Gateway version 3.2
 - Python version 3.10.8

11208, 2022-12-25:
I am facing the same issue.

samson0v, 2022-12-26:
Hi @11208 and  @BjornDeGloire, trying to fix it...

closed 2022-12-29 by imbeacon