thingsboard/thingsboard-gateway #1032: [BUG]

charlilezar, 2022-12-22:
**Describe the bug**

Thank you very much for checking this question.

[WinError 10038]  from file  OPC-UA Connector.py .

i need call  rpc method from opcua-gateway , 
http://tb-ip:port/api/plugins/rpc/oneway/[deviceID]
http request/post whith these parameters: 
{
  "method": "ServerMethod",
  "params":["{ns=2;s=ServerMethod}",9527]
}
and 
{
  "method": "set",
  "params": "{ns=2;i=3};9527"
}
and the control tools : rpc debug shows request timeout .

**Connector name (If bug in the some connector):**
 in file : OPC-UA Connector.py  and line  321
                                result = method["node"].call_method(method[rpc_method], arguments)

**Error traceback (If available):**

Traceback (most recent call last):

called method ：  result = method["node"].call_method(method[rpc_method], arguments)
return exception:： 
[WinError 10038] 在一个非套接字上尝试了一个操作

**Versions (please complete the following information):**
 - OS: [Microsoft Windows 11 : version 10.0.22623]
 - Thingsboard IoT Gateway version [3.1]
 - Python version[3.10]

samson0v, 2022-12-30:
Hi @charlilezar, 
`[WinError 10038] 在一个非套接字上尝试了一个操作 (An operation was attempted on a non-socket)`
From this exception, it seems that problem on your side. Try to google it.

samson0v, 2023-01-26:
@charlilezar, any updates? Can we close the issue?

closed 2023-02-03 by samson0v