thingsboard/thingsboard-gateway #1033: I have an array comming in with booleans. I Want to assign every item of the array to a differant device

BjornDeGloire, 2022-12-27:
I have a PLC that I read the inputs with modbus TCP. I receive an array with boolean's 	
[0,1,0,0,0,0,0,0,1,0,0,0,1,0,1,0]
The problem is that I want to assign every input to a unique device.
I do not succeed to change the device in thingsboard with a rule chain.
I tried to change the deviceName in metadata but that did not work.
We are thinking to send it to an integration after separating the array but it feels like a dirty fix to do it in this way.
How do I do this in a correct way in Thingsboard?

With friendly regards,

ericprom, 2022-12-27:
Hopefully, after they fix issue #1030, we can work this out in the future.

samson0v, 2022-12-29:
Hi @BjornDeGloire, you can't do this dynamically using the default Modbus Connector. 
You can try to do it from the opposite: configure the devices, read the register with them and, according to the legend of RuleChains, allocate the data you need.
Another solution is to write [a custom connector](https://thingsboard.io/docs/iot-gateway/custom/).

devaskim, 2023-01-03:
@BjornDeGloire use `Change originator` rule chain node to assign data to another device

closed 2024-01-30 by imbeacon