thingsboard/thingsboard-gateway #1041: [BUG]

andonoc, 2023-01-07:
Hi, thank you for your product
I tried to install tb-gateway on a ubuntu server(xcp-ng)
As you mentioned on your installation instructions (This guide describes how to install ThingsBoard IoT Gateway on Ubuntu 18.04 LTS / 
Ubuntu 20.04 LTS) i used Ubuntu 20.04 and the installation failed
I tried with .deb pip and from sources (i tried on docker too but im a absolute docker noob, and it didn't work with the instructions)
I tried 3.2, 3.1, 3.0.1, 3.0, 2.9
2.9 finally worked.
Then i tried on Ubuntu 18,19,20,22
On Ubuntu 22.04 tb-gateway (with apt) finally started, is connected and working(By the way, i get the " statistics_service" error )
I have a Ubuntu 20.04 (on KVM) server running for a year and the tb-gateway is stuck on 2.8 and not upgradable.

imbeacon, 2023-01-07:
Hi, thank you for your interest in our product, please provide more information if it is possible to help us investigate and solve the issue, we need logs or information that you faced during installation. Also we will try to install on Ubuntu 20.04.
Probably there is an old Python version, such as newer version of the Python is required now than it was set up on Ubuntu 20.04 by default.

andonoc, 2023-01-09:
hi,
installing  on Ubuntu20.04 with your documentation from sources errors:
`.py
Traceback (most recent call last):
  File "./thingsboard_gateway/tb_gateway.py", line 17, in <module>
    from thingsboard_gateway.gateway.tb_gateway_service import TBGatewayService
`

samson0v, 2023-01-10:
Hi @andonoc, try to use the following command:
`pip3 install --upgrade protobuf`

andonoc, 2023-01-10:
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.1 LTS
Release:        22.04
Codename:       jammy
build from sources

for ``` python3 ./thingsboard_gateway/tb_gateway.py``` i get this output:

<details><summary>CLICK without: pip3 install --upgrade protobuf</summary>
<p>

```
Defaulting to user installation because normal site-packages is not writeable
Collecting jsonpath-rw
  Downloading jsonpath-rw-1.4.0.tar.gz (13 kB)
  Preparing metadata (setup.py) ... done
Collecting regex
  Downloading regex-2022.10.31-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (770 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 770.5/770.5 KB 10.6 MB/s eta 0:00:00
Requirement already satisfied: pip in /usr/lib/python3/dist-packages (from -r requirements.txt (line 3)) (22.0.2)
Requirement already satisfied: PyYAML in /usr/lib/python3/dist-packages (from -r requirements.txt (line 4)) (5.4.1)
Collecting simplejson
  Downloading simplejson-3.18.1-cp310-cp310-manylinux_2_5_x86_64.manylinux1_x86_64.manylinux_2_17_x86_64.manylinux2014_x86_64.whl (137 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 137.4/137.4 KB 15.2 MB/s eta 0:00:00
Requirement already satisfied: requests in /usr/lib/python3/dist-packages (from -r requirements.txt (line 6)) (2.25.1)
Collecting PyInquirer
  Downloading PyInquirer-1.0.3.tar.gz (27 kB)
  Preparing metadata (setup.py) ... done
Collecting pyfiglet
  Downloading pyfiglet-0.8.post1-py2.py3-none-any.whl (865 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 865.8/865.8 KB 21.2 MB/s eta 0:00:00
Collecting termcolor
  Downloading termcolor-2.2.0-py3-none-any.whl (6.6 kB)
Collecting grpcio==1.40.0
  Downloading grpcio-1.40.0.tar.gz (21.3 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 21.3/21.3 MB 6.9 MB/s eta 0:00:00
  Preparing metadata (setup.py) ... done
Collecting protobuf<=3.20.0
  Downloading protobuf-3.20.0-cp310-cp310-manylinux_2_12_x86_64.manylinux2010_x86_64.whl (1.1 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.1/1.1 MB 49.8 MB/s eta 0:00:00
Collecting cachetools
  Downloading cachetools-5.2.1-py3-none-any.whl (9.3 kB)
Requirement already satisfied: cryptography<3.5 in /usr/lib/python3/dist-packages (from -r requirements.txt (line 13)) (3.4.8)
Requirement already satisfied: six>=1.5.2 in /usr/lib/python3/dist-packages (from grpcio==1.40.0->-r requirements.txt (line 10)) (1.16.0)
Collecting decorator
  Downloading decorator-5.1.1-py3-none-any.whl (9.1 kB)
Collecting ply
  Downloading ply-3.11-py2.py3-none-any.whl (49 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 49.6/49.6 KB 7.9 MB/s eta 0:00:00
Collecting Pygments>=2.2.0
  Downloading Pygments-2.14.0-py3-none-any.whl (1.1 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.1/1.1 MB 89.8 MB/s eta 0:00:00
Collecting prompt_toolkit==1.0.14
  Downloading prompt_toolkit-1.0.14-py3-none-any.whl (248 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 248.3/248.3 KB 51.0 MB/s eta 0:00:00
Collecting wcwidth
  Downloading wcwidth-0.2.5-py2.py3-none-any.whl (30 kB)
Building wheels for collected packages: grpcio, jsonpath-rw, PyInquirer
  Building wheel for grpcio (setup.py) ... done
  Created wheel for grpcio: filename=grpcio-1.40.0-cp310-cp310-linux_x86_64.whl size=45686714 sha256=018e5440186fc0514210f838d8d96cf65b896106c394eb66d808b03139061db6
  Stored in directory: /home/ando/.cache/pip/wheels/8f/0f/e0/ef23fb7e36efeafd52f6b4a9db77a870b6a1cc333c8c4607f5
  Building wheel for jsonpath-rw (setup.py) ... done
  Created wheel for jsonpath-rw: filename=jsonpath_rw-1.4.0-py3-none-any.whl size=15149 sha256=1faaca1e3ffac871a9bb51cba66c3789d1e1aae343f3f8b062f0ddbe9f04f1a6
  Stored in directory: /home/ando/.cache/pip/wheels/e3/76/6f/c25be6a9e6cc9985b96e8c95997d46790242c6426ef68e754c
  Building wheel for PyInquirer (setup.py) ... done
  Created wheel for PyInquirer: filename=PyInquirer-1.0.3-py3-none-any.whl size=32862 sha256=ee097ea8126911f96e1493eabc1a20d293fec2d9fe8fc4e938fedaba3282e94f
  Stored in directory: /home/ando/.cache/pip/wheels/2c/e4/8e/81d8a92fe3a12404e9c6d991c567b7ad7c9ee09d542f7c1980
Successfully built grpcio jsonpath-rw PyInquirer
Installing collected packages: wcwidth, pyfiglet, ply, termcolor, simplejson, regex, Pygments, protobuf, prompt_toolkit, grpcio, decorator, cachetools, PyInquirer, jsonpath-rw
  WARNING: The script pyfiglet is installed in '/home/ando/.local/bin' which is not on PATH.
[... 74 lines cut ...]
  Building wheel for PyInquirer (setup.py) ... done
  Created wheel for PyInquirer: filename=PyInquirer-1.0.3-py3-none-any.whl size=32862 sha256=3a8631281fcf8c5ba7806e129fa0450cdd31eae0de9763b52770d2575afe8dad
  Stored in directory: /home/ando/.cache/pip/wheels/2c/e4/8e/81d8a92fe3a12404e9c6d991c567b7ad7c9ee09d542f7c1980
Successfully built grpcio jsonpath-rw PyInquirer
Installing collected packages: wcwidth, pyfiglet, ply, termcolor, simplejson, regex, Pygments, protobuf, prompt_toolkit, grpcio, decorator, cachetools, PyInquirer, jsonpath-rw
  WARNING: The script pyfiglet is installed in '/home/ando/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  WARNING: The script pygmentize is installed in '/home/ando/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  WARNING: The script jsonpath.py is installed in '/home/ando/.local/bin' which is not on PATH.
[... cut ...]
ando@thingsboard:~/thingsboard-gateway$ mkdir logs
ando@thingsboard:~/thingsboard-gateway$ python3 ./thingsboard_gateway/tb_gateway.py
Traceback (most recent call last):
  File "/home/ando/thingsboard-gateway/./thingsboard_gateway/tb_gateway.py", line 17, in <module>
    from thingsboard_gateway.gateway.tb_gateway_service import TBGatewayService
ModuleNotFoundError: No module named 'thingsboard_gateway'

```

</p>
</details>

And ```  Building wheel for grpcio (setup.py) ...``` takes several minutes

andonoc, 2023-01-10:
Are the scripts working on your machines?

andonoc, 2023-01-11:
Distributor ID: Ubuntu
Description: Ubuntu 22.04.1 LTS
Release: 22.04
Codename: jammy

Instlling with apt
Thingsboard-gateway  is running
the script is throwing a warning and a note
<details><summary>WARNINGS</summary>
<p>

```
   Setting up libc6-dev:amd64 (2.35-0ubuntu3.1) ...
Setting up libc-devtools (2.35-0ubuntu3.1) ...
Setting up python3-thingsboard-gateway (3.2-1) ...
Installing jsonschema
Collecting jsonschema==3.1.1
  Downloading jsonschema-3.1.1-py2.py3-none-any.whl (56 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 56.1/56.1 KB 2.5 MB/s eta 0:00:00
Requirement already satisfied: six>=1.11.0 in /usr/lib/python3/dist-packages (from jsonschema==3.1.1) (1.16.0)
Requirement already satisfied: attrs>=17.4.0 in /usr/lib/python3/dist-packages (from jsonschema==3.1.1) (21.2.0)
Requirement already satisfied: importlib-metadata in /usr/lib/python3/dist-packages (from jsonschema==3.1.1) (4.6.4)
Requirement already satisfied: pyrsistent>=0.14.0 in /usr/lib/python3/dist-packages (from jsonschema==3.1.1) (0.18.1)
Requirement already satisfied: setuptools in /usr/lib/python3/dist-packages (from jsonschema==3.1.1) (59.6.0)
Installing collected packages: jsonschema
  Attempting uninstall: jsonschema
    Found existing installation: jsonschema 3.2.0
    Not uninstalling jsonschema at /usr/lib/python3/dist-packages, outside environment /usr
    Can't uninstall 'jsonschema'. No files were found to uninstall.
Successfully installed jsonschema-3.1.1
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
Installation completed
Enabling daemon...
1065
Created symlink /etc/systemd/system/multi-user.target.wants/thingsboard-gateway.service → /etc/systemd/system/thingsboard-gateway.service.
1065
-e Thingsboard Gateway has been installed. Have a nice day :)
Setting up gcc (4:11.2.0-1ubuntu1) ...
Setting up libexpat1-dev:amd64 (2.4.7-1ubuntu0.2) ...
Setting up libstdc++-11-dev:amd64 (11.3.0-1ubuntu1~22.04) ...
Setting up zlib1g-dev:amd64 (1:1.2.11.dfsg-2ubuntu9.2) ...
Setting up g++-11 (11.3.0-1ubuntu1~22.04) ...
Setting up libpython3.10-dev:amd64 (3.10.6-1~22.04.2) ...
Setting up python3.10-dev (3.10.6-1~22.04.2) ...
Setting up g++ (4:11.2.0-1ubuntu1) ...
update-alternatives: using /usr/bin/g++ to provide /usr/bin/c++ (c++) in auto mode
Setting up build-essential (12.9ubuntu3) ...
Setting up libpython3-dev:amd64 (3.10.6-1~22.04) ...
Setting up python3-dev (3.10.6-1~22.04) ...
Processing triggers for man-db (2.10.2-1) ...
Processing triggers for libc-bin (2.35-0ubuntu3.1) ...
Scanning processes...
Scanning linux images...

Running kernel seems to be up-to-date.

No services need to be restarted.

No containers need to be restarted.

No user sessions are running outdated binaries.

No VM guests are running outdated hypervisor (qemu) binaries on this host.
N: Download is performed unsandboxed as root as file '/home/ando/python3-thingsboard-gateway.deb' couldn't be accessed by user '_apt'. - pkgAcquire::Run (13: Permission denied)
ando@thingsboard:~$ sudo systemctl status thingsboard-gateway.service
● thingsboard-gateway.service - ThingsBoard Gateway
     Loaded: loaded (/etc/systemd/system/thingsboard-gateway.service; enabled; vendor preset: enabled)
     Active: active (running) since Wed 2023-01-11 05:34:52 UTC; 1min 16s ago
   Main PID: 2926 (python3)
      Tasks: 11 (limit: 4574)
[... 130 lines cut ...]
Jan 11 05:34:53 thingsboard python3[2926]: ""2023-01-11 05:34:53" - |INFO| - [tb_updater.py] - tb_updater - check_for_new_version - 84 - v3.2"
Jan 11 05:34:53 thingsboard python3[2926]: ""2023-01-11 05:34:53" - |INFO| - [tb_updater.py] - tb_updater - check_for_new_version - 86 -
Jan 11 05:34:53 thingsboard python3[2926]: [===UPDATE===]
Jan 11 05:34:53 thingsboard python3[2926]:  New version v3.2 is available!
Jan 11 05:34:53 thingsboard python3[2926]: [===UPDATE===]
Jan 11 05:34:53 thingsboard python3[2926]: "
Jan 11 05:34:56 thingsboard python3[2926]: ""2023-01-11 05:34:56" - |ERROR| - [statistics_service.py] - statistics_service - run - 67 - Statistic parameter raise the exception: /bin/sh: 1:>
Jan 11 05:34:56 thingsboard python3[2926]: "
Jan 11 05:34:56 thingsboard python3[2926]: ""2023-01-11 05:34:56" - |ERROR| - [statistics_service.py] - statistics_service - run - 67 - Statistic parameter raise the exception: /bin/sh: 1:>
Jan 11 05:34:56 thingsboard python3[2926]: "
Jan 11 05:34:56 thingsboard python3[2926]: ""2023-01-11 05:34:56" - |ERROR| - [statistics_service.py] - statistics_service - run - 67 - Statistic parameter raise the exception: /bin/sh: 1:>
Jan 11 05:34:56 thingsboard python3[2926]: "
Jan 11 05:35:00 thingsboard python3[2926]: ""2023-01-11 05:35:00" - |ERROR| - [statistics_service.py] - statistics_service - run - 67 - Statistic parameter raise the exception: /bin/sh: 1:>
Jan 11 05:35:00 thingsboard python3[2926]: "
Jan 11 05:35:00 thingsboard python3[2926]: ""2023-01-11 05:35:00" - |ERROR| - [statistics_service.py] - statistics_service - run - 67 - Statistic parameter raise the exception: /bin/sh: 1:>
Jan 11 05:35:00 thingsboard python3[2926]: "
Jan 11 05:35:00 thingsboard python3[2926]: ""2023-01-11 05:35:00" - |ERROR| - [statistics_service.py] - statistics_service - run - 67 - Statistic parameter raise the exception: /bin/sh: 1:>
Jan 11 05:35:00 thingsboard python3[2926]: "
Jan 11 05:35:04 thingsboard python3[2926]: ""2023-01-11 05:35:04" - |ERROR| - [statistics_service.py] - statistics_service - run - 67 - Statistic parameter raise the exception: /bin/sh: 1:>
Jan 11 05:35:04 thingsboard python3[2926]: "

```

</p>
</details>

samson0v, 2023-01-11:
Hi @andonoc, in the tb_gateway.yaml file, simply comment the following line:
[image]

This bug was fixed, so in the next release, this error will not be raised.

samson0v, 2023-01-11:
@andonoc, any other questions or we can close the issue?

andonoc, 2023-01-11:
Hi @samson0v 
OK
And i have a question, but not related to the initially post.
The following connector worked on 2.8 but not on the actual installation with apt
No problem connecting to a mosquitto broker but is not connecting to TTN.
Did you change something on the configuration side that i missed?
Thank you for your time!!!!
<details><summary>TTN MQTT connector</summary>
<p>

```
   {
    "broker": {
        "name": "TTN Broker",
        "host": "eu1.cloud.thethings.network",
        "port": 1883,
        "clientId": "ThingsBoard_TTN_gateway",
        "security": {
            "type": "basic",
            "username": "xxxxxxx@ttn",
            "password": "NNSXS.xxxxxxxxxxx"
        }
    },
    "mapping": [
        {
            "topicFilter": "v3/+/devices/+/up",
            "converter": {
                "type": "json",
                "deviceNameJsonExpression": "${end_device_ids.dev_eui}",
                "deviceTypeJsonExpression": "${end_device_ids.application_ids.application_id}",
                "timeout": 60000,
                "attributes": [
                    {
                        "type": "string",
                        "key": "application_id",
                        "value": "${end_device_ids.application_ids.application_id}"
                    },
                    {
                        "type": "double",
                        "key": "latitude",
                        "value": "${uplink_message.locations.user.latitude}"
                    },
                    {
                        "type": "double",
                        "key": "longitude",
                        "value": "${uplink_message.locations.user.longitude}"
                    },
                    {
                        "type": "long",
                        "key": "altitude",
                        "value": "${uplink_message.locations.user.altitude}"
                    }
                ],
                "timeseries": [
                    {
                        "type": "double",
                        "key": "airTemperature",
                        "value": "${uplink_message.decoded_payload.airTemperature}"
                    },
                    {
                        "type": "long",
                        "key": "groundTemperature",
                        "value": "${uplink_message.decoded_payload.groundTemperature}"
                    },
                    {
                        "type": "double",
                        "key": "bulbTemperature",
                        "value": "${uplink_message.decoded_payload.bulbTemperature}"
                    },
                    {
                        "type": "long",
                        "key": "watermark",
                        "value": "${uplink_message.decoded_payload.watermark}"
                    },                    
                    {
                        "type": "long",
                        "key": "RSSI",
                        "value": "${uplink_message.rx_metadata[0].rssi}"
                    },
                    {
                        "type": "long",
                        "key": "bandwidth",
                        "value": "${uplink_message.settings.data_rate.lora.bandwidth}"
                    },
                    {
                        "type": "long",
                        "key": "spreading_factor",
                        "value": "${uplink_message.settings.data_rate.lora.spreading_factor}"
                    },
                    {
                        "type": "long",
                        "key": "frequency",
                        "value": "${uplink_message.settings.frequency}"
                    }
                ]
            }
        }
    ],
    "connectRequests": [],
    "disconnectRequests": [],
    "attributeUpdates": [],
    "serverSideRpc": []
}

```

</p>
</details>

samson0v, 2023-01-12:
@andonoc, send, please, the logs.

andonoc, 2023-01-12:
OK, i found : ```'attributeRequests' section missing from configuration```
On the documentation: ```Configuration in this section are optional```

<details><summary>systemd logs</summary>
<p>

```
Jan 12 20:41:19 thingsboard systemd[1]: Started ThingsBoard Gateway.
Jan 12 20:41:19 thingsboard python3[997]: ""2023-01-12 20:41:19" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 148 - Gateway starting..."
Jan 12 20:41:20 thingsboard python3[997]: ""2023-01-12 20:41:20" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 153 - ThingsBoard IoT gateway version: 3.2"
Jan 12 20:41:20 thingsboard python3[997]: ""2023-01-12 20:41:20" - |INFO| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - gw_subscribe_to_attribute - 185 - Subscribed to *|* with id 1 for device *"
Jan 12 20:41:20 thingsboard python3[997]: ""2023-01-12 20:41:20" - |INFO| - [tb_loader.py] - tb_loader - import_module - 68 - Import MqttConnector from /usr/lib/python3/dist-packages/thingsboard_gateway/connectors/mqtt."
Jan 12 20:41:20 thingsboard python3[997]: ""2023-01-12 20:41:20" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 243 - Number of accepted mapping handlers: 1"
Jan 12 20:41:20 thingsboard python3[997]: ""2023-01-12 20:41:20" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 247 - Number of rejected mapping handlers: 0"
Jan 12 20:41:20 thingsboard python3[997]: ""2023-01-12 20:41:20" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 243 - Number of accepted serverSideRpc handlers: 0"
Jan 12 20:41:20 thingsboard python3[997]: ""2023-01-12 20:41:20" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 247 - Number of rejected serverSideRpc handlers: 0"
Jan 12 20:41:20 thingsboard python3[997]: ""2023-01-12 20:41:20" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 243 - Number of accepted connectRequests handlers: 0"
Jan 12 20:41:20 thingsboard python3[997]: ""2023-01-12 20:41:20" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 247 - Number of rejected connectRequests handlers: 0"
Jan 12 20:41:20 thingsboard python3[997]: ""2023-01-12 20:41:20" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 243 - Number of accepted disconnectRequests handlers: 0"
Jan 12 20:41:20 thingsboard python3[997]: ""2023-01-12 20:41:20" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 247 - Number of rejected disconnectRequests handlers: 0"
Jan 12 20:41:20 thingsboard python3[997]: ""2023-01-12 20:41:20" - |ERROR| - [mqtt_connector.py] - mqtt_connector - load_handlers - 220 - 'attributeRequests' section missing from configuration"
Jan 12 20:41:20 thingsboard python3[997]: ""2023-01-12 20:41:20" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 243 - Number of accepted attributeUpdates handlers: 0"
Jan 12 20:41:20 thingsboard python3[997]: ""2023-01-12 20:41:20" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 247 - Number of rejected attributeUpdates handlers: 0"
Jan 12 20:41:20 thingsboard python3[997]: ""2023-01-12 20:41:20" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 252 - Gateway started."
Jan 12 20:41:20 thingsboard python3[997]: ""2023-01-12 20:41:20" - |INFO| - [tb_updater.py] - tb_updater - check_for_new_version - 84 - v3.2"
Jan 12 20:41:20 thingsboard python3[997]: ""2023-01-12 20:41:20" - |INFO| - [tb_updater.py] - tb_updater - check_for_new_version - 86 -
Jan 12 20:41:20 thingsboard python3[997]: [===UPDATE===]
Jan 12 20:41:20 thingsboard python3[997]:  New version v3.2 is available!
Jan 12 20:41:20 thingsboard python3[997]: [===UPDATE===]
Jan 12 20:41:20 thingsboard python3[997]: "
Jan 12 20:42:05 thingsboard python3[997]: ""2023-01-12 20:42:05" - |INFO| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - gw_subscribe_to_attribute - 185 - Subscribed to *|* with id 2 for device *"

```

</p>
</details>

<details><summary>tb_gateway.yaml</summary>
<p>

```
thingsboard:
  host: 127.0.0.1
  port: 1883
  remoteShell: false
  remoteConfiguration: false
  statistics:
    enable: true
    statsSendPeriodInSeconds: 3600
  #  configuration: statistics.json
  maxPayloadSizeBytes: 1024
  minPackSendDelayMS: 0
  checkConnectorsConfigurationInSeconds: 60
  handleDeviceRenaming: true
  checkingDeviceActivity:
    checkDeviceInactivity: false
    inactivityTimeoutSeconds: 120
    inactivityCheckPeriodSeconds: 10
  security:
    accessToken: V7b*********mnAoS
  qos: 1
storage:
  type: memory
  read_records_count: 100
  max_records_count: 100000
#  type: file
#  data_folder_path: ./data/
#  max_file_count: 10
#  max_read_records_count: 10
#  max_records_per_file: 10000
[... 100 lines cut ...]
                    },
                    {
                        "type": "long",
                        "key": "spreading_factor",
                        "value": "${uplink_message.settings.data_rate.lora.spreading_factor}"
                    },
                    {
                        "type": "long",
                        "key": "frequency",
                        "value": "${uplink_message.settings.frequency}"
                    }
                ]
            }
        }
    ],
    "connectRequests": [],
    "disconnectRequests": [],
    "attributeUpdates": [],
    "serverSideRpc": []
}

```

</p>
</details>

andonoc, 2023-01-12:
hi @samson0v 
installed on Centos and everything works fine
Thank you!!!

closed 2023-01-12 by andonoc