thingsboard/thingsboard-gateway #1044: [HELP] - Problem with modbus connector and - an error has occurred: 'RemoteStation' object has no attribute 'addrTuple'

tamerfs, 2023-01-12:
i was searching on other topics a help or solution for my problem, it came to me a similar issue but the solution proposed didn't work. (https://github.com/thingsboard/thingsboard-gateway/issues/712#issuecomment-1380672455)

**Error traceback (journalctl -fu thingsboard-gateway.service):**
```
 During handling of the above exception, another exception occurred:
Jan 12 14:39:56 raspberrypi python3[25033]: Traceback (most recent call last):
Jan 12 14:39:56 raspberrypi python3[25033]:   File "/usr/lib/python3/dist-packages/thingsboard_gateway/tb_utility/tb_loader.py", line 65, in import_module
Jan 12 14:39:56 raspberrypi python3[25033]:     module_spec.loader.exec_module(module)
Jan 12 14:39:56 raspberrypi python3[25033]:   File "<frozen importlib._bootstrap_external>", line 790, in exec_module
Jan 12 14:39:56 raspberrypi python3[25033]:   File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
Jan 12 14:39:56 raspberrypi python3[25033]:   File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 43, in <module>
Jan 12 14:39:56 raspberrypi python3[25033]:     from twisted.internet import reactor
Jan 12 14:39:56 raspberrypi python3[25033]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.9/site-packages/twisted/internet/reactor.py", line 38, in <module>
Jan 12 14:39:56 raspberrypi python3[25033]:     from twisted.internet import default
Jan 12 14:39:56 raspberrypi python3[25033]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.9/site-packages/twisted/internet/default.py", line 55, in <module>
Jan 12 14:39:56 raspberrypi python3[25033]:     install = _getInstallFunction(platform)
Jan 12 14:39:56 raspberrypi python3[25033]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.9/site-packages/twisted/internet/default.py", line 51, in _getInstallFunction
Jan 12 14:39:56 raspberrypi python3[25033]:     from twisted.internet.selectreactor import install
Jan 12 14:39:56 raspberrypi python3[25033]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.9/site-packages/twisted/internet/selectreactor.py", line 18, in <module>
Jan 12 14:39:56 raspberrypi python3[25033]:     from twisted.internet import posixbase
Jan 12 14:39:56 raspberrypi python3[25033]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.9/site-packages/twisted/internet/posixbase.py", line 16, in <module>
Jan 12 14:39:56 raspberrypi python3[25033]:     from twisted.internet import error, tcp, udp
Jan 12 14:39:56 raspberrypi python3[25033]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.9/site-packages/twisted/internet/tcp.py", line 99, in <module>
Jan 12 14:39:56 raspberrypi python3[25033]:     from twisted.internet import abstract, address, base, error, fdesc, main
Jan 12 14:39:56 raspberrypi python3[25033]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.9/site-packages/twisted/internet/base.py", line 34, in <module>
Jan 12 14:39:56 raspberrypi python3[25033]:     from twisted.internet import abstract, defer, error, fdesc, main, threads
Jan 12 14:39:56 raspberrypi python3[25033]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.9/site-packages/twisted/internet/defer.py", line 42, in <module>
Jan 12 14:39:56 raspberrypi python3[25033]:     from typing_extensions import Literal, ParamSpec, Protocol
Jan 12 14:39:56 raspberrypi python3[25033]: ImportError: cannot import name 'ParamSpec' from 'typing_extensions' (/usr/lib/python3/dist-packages/typing_extensions.py)
Jan 12 14:39:56 raspberrypi python3[25033]: ""2023-01-12 14:39:56" - |WARNING| - [tb_gateway_service.py] - tb_gateway_service - _load_connectors - 547 - Connector implementation not found for Modbus Connector"
```

I already tried to run 'sudo pip3 install pymodbus pyserial twisted' and got this in response:

```
pi@raspberrypi:/etc/thingsboard-gateway/config $ sudo pip3 install pymodbus pyserial twisted
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: pymodbus in /usr/local/lib/python3.9/dist-packages (3.0.2)
Requirement already satisfied: pyserial in /usr/lib/python3/dist-packages (3.5b0)
Requirement already satisfied: twisted in /usr/local/lib/python3.9/dist-packages (22.10.0)
Requirement already satisfied: zope.interface>=4.4.2 in /usr/local/lib/python3.9/dist-packages (from twisted) (5.5.2)
Requirement already satisfied: attrs>=19.2.0 in /usr/local/lib/python3.9/dist-packages (from twisted) (22.2.0)
Requirement already satisfied: hyperlink>=17.1.1 in /usr/local/lib/python3.9/dist-packages (from twisted) (21.0.0)
Requirement already satisfied: constantly>=15.1 in /usr/local/lib/python3.9/dist-packages (from twisted) (15.1.0)
Requirement already satisfied: typing-extensions>=3.6.5 in /usr/lib/python3/dist-packages (from twisted) (3.7.4.3)
Requirement already satisfied: incremental>=21.3.0 in /usr/local/lib/python3.9/dist-packages (from twisted) (22.10.0)
Requirement already satisfied: Automat>=0.8.0 in /usr/local/lib/python3.9/dist-packages (from twisted) (22.10.0)
Requirement already satisfied: six in /usr/lib/python3/dist-packages (from Automat>=0.8.0->twisted) (1.16.0)
Requirement already satisfied: idna>=2.5 in /usr/lib/python3/dist-packages (from hyperlink>=17.1.1->twisted) (2.10)
Requirement already satisfied: setuptools in /usr/lib/python3/dist-packages (from zope.interface>=4.4.2->twisted) (52.0.0)
```
and after that executed 'sudo pip3 install pymodbus pyserial twisted --upgrade' like in the comment (https://github.com/thingsboard/thingsboard-gateway/issues/712#issuecomment-1380741696)
and got this:

```
pi@raspberrypi:~ $ sudo pip3 install pymodbus pyserial twisted --upgrade
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: pymodbus in /usr/local/lib/python3.9/dist-packages (3.0.2)
Collecting pymodbus
  Using cached https://www.piwheels.org/simple/pymodbus/pymodbus-3.0.2-py2.py3-none-any.whl (142 kB)
  Using cached https://www.piwheels.org/simple/pymodbus/pymodbus-3.0.1-py2.py3-none-any.whl (142 kB)
Requirement already satisfied: pyserial in /usr/local/lib/python3.9/dist-packages (3.5)
Collecting pyserial
  Using cached https://www.piwheels.org/simple/pyserial/pyserial-3.5-py2.py3-none-any.whl (90 kB)
  Downloading https://www.piwheels.org/simple/pyserial/pyserial-3.4-py2.py3-none-any.whl (193 kB)
     |################################| 193 kB 168 kB/s
Requirement already satisfied: twisted in /usr/local/lib/python3.9/dist-packages (22.10.0)
Collecting twisted
  Using cached https://www.piwheels.org/simple/twisted/Twisted-22.10.0-py3-none-any.whl (3.1 MB)
  Using cached https://www.piwheels.org/simple/twisted/Twisted-22.8.0-py3-none-any.whl (3.1 MB)
[... 83 lines cut ...]
Jan 12 15:01:15 raspberrypi python3[25033]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.9/site-packages/bacpypes/appservice.py", line 1243, in confirmation
Jan 12 15:01:15 raspberrypi python3[25033]:     self.sap_request(apdu)
Jan 12 15:01:15 raspberrypi python3[25033]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.9/site-packages/bacpypes/comm.py", line 527, in sap_request
Jan 12 15:01:15 raspberrypi python3[25033]:     self.serviceElement.indication(*args, **kwargs)
Jan 12 15:01:15 raspberrypi python3[25033]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.9/site-packages/bacpypes/appservice.py", line 1471, in indication
Jan 12 15:01:15 raspberrypi python3[25033]:     self.sap_request(xpdu)
Jan 12 15:01:15 raspberrypi python3[25033]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.9/site-packages/bacpypes/comm.py", line 527, in sap_request
Jan 12 15:01:15 raspberrypi python3[25033]:     self.serviceElement.indication(*args, **kwargs)
Jan 12 15:01:15 raspberrypi python3[25033]:   File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/bacnet/bacnet_utilities/tb_gateway_bacnet_application.py", line 66, in indication
Jan 12 15:01:15 raspberrypi python3[25033]:     apdu.pduSource.addrTuple[0],
Jan 12 15:01:15 raspberrypi python3[25033]: AttributeError: 'RemoteStation' object has no attribute 'addrTuple'
```

**Versions :**

Raspbian GNU Linux 11 (bullseye)
Python 3.9.2
python3-thingsboard-gateway/now 3.2-1 all [installed,local]

can you guys help me?
thank you so much!

[error logs.txt](https://github.com/thingsboard/thingsboard-gateway/files/10404845/error.logs.txt)
[modbus.json.txt](https://github.com/thingsboard/thingsboard-gateway/files/10404846/modbus.json.txt)
[tb_gateway.txt](https://github.com/thingsboard/thingsboard-gateway/files/10404847/tb_gateway.txt)

samson0v, 2023-01-16:
Hi @tamerfs, from your tb_gateway.yaml you are trying to start OPC-UA Connector but the issue title about Modbus Connector, nay in the logs Gateway starts BACnet Connector.

Make sure that you are using right config files.

tamerfs, 2023-01-16:
@samson0v  sorry, acctualy i uploaded the wrong file.

this is the correct tb_gateway.yaml file.
[tb_gateway (1).txt](https://github.com/thingsboard/thingsboard-gateway/files/10425654/tb_gateway.1.txt)

with the bacnet and modbus connectors started..
and the bacnet its working.

thanks!!

samson0v, 2023-01-16:
@tamerfs ok, can we close thr issue?

tamerfs, 2023-01-16:
please, no.

i still have these problemes.
the modbus isint working and the system replys me these logs https://github.com/thingsboard/thingsboard-gateway/issues/1044#issue-1531159333

samson0v, 2023-01-16:
@tamerfs if you about addrTuple, this exception relates to BACnet Connector but it is working as you said.

samson0v, 2023-01-16:
@tamerfs about ParamSpec exception, try to run thr following command:
`sudo pip3 install pymodbus pyserial twisted --upgrade`

tamerfs, 2023-01-16:
i did befora, and just execute de cmd line in my raspi and i got this

```
pi@raspberrypi:/etc/thingsboard-gateway/config $ sudo pip3 install pymodbus pyserial twisted --upgrade
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: pymodbus in /usr/local/lib/python3.9/dist-packages (3.0.2)
Collecting pymodbus
  Downloading https://www.piwheels.org/simple/pymodbus/pymodbus-3.1.0-py2.py3-none-any.whl (163 kB)
     |################################| 163 kB 154 kB/s
Requirement already satisfied: pyserial in /usr/local/lib/python3.9/dist-packages (3.5)
Collecting pyserial
  Using cached https://www.piwheels.org/simple/pyserial/pyserial-3.5-py2.py3-none-any.whl (90 kB)
  Using cached https://www.piwheels.org/simple/pyserial/pyserial-3.4-py2.py3-none-any.whl (193 kB)
Requirement already satisfied: twisted in /usr/local/lib/python3.9/dist-packages (22.10.0)
Collecting twisted
  Using cached https://www.piwheels.org/simple/twisted/Twisted-22.10.0-py3-none-any.whl (3.1 MB)
  Using cached https://www.piwheels.org/simple/twisted/Twisted-22.8.0-py3-none-any.whl (3.1 MB)
Requirement already satisfied: Automat>=0.8.0 in /usr/local/lib/python3.9/dist-packages (from twisted) (22.10.0)
Requirement already satisfied: typing-extensions>=3.6.5 in /usr/lib/python3/dist-packages (from twisted) (3.7.4.3)
Requirement already satisfied: zope.interface>=4.4.2 in /usr/local/lib/python3.9/dist-packages (from twisted) (5.5.2)
Requirement already satisfied: attrs>=19.2.0 in /usr/local/lib/python3.9/dist-packages (from twisted) (22.2.0)
Requirement already satisfied: incremental>=21.3.0 in /usr/local/lib/python3.9/dist-packages (from twisted) (22.10.0)
Requirement already satisfied: hyperlink>=17.1.1 in /usr/local/lib/python3.9/dist-packages (from twisted) (21.0.0)
Requirement already satisfied: constantly>=15.1 in /usr/local/lib/python3.9/dist-packages (from twisted) (15.1.0)
Requirement already satisfied: six in /usr/lib/python3/dist-packages (from Automat>=0.8.0->twisted) (1.16.0)
Requirement already satisfied: idna>=2.5 in /usr/lib/python3/dist-packages (from hyperlink>=17.1.1->twisted) (2.10)
Requirement already satisfied: setuptools in /usr/lib/python3/dist-packages (from zope.interface>=4.4.2->twisted) (52.0.0)
Installing collected packages: pymodbus
  Attempting uninstall: pymodbus
    Found existing installation: pymodbus 3.0.2
    Uninstalling pymodbus-3.0.2:
      Successfully uninstalled pymodbus-3.0.2
Successfully installed pymodbus-3.1.0`
```
and the service.log replyed me this

```
""2023-01-16 09:44:26" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __stop_gateway - 355 - Stopping..."
""2023-01-16 09:44:26" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 148 - Gateway starting..."
""2023-01-16 09:44:27" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 153 - ThingsBoard IoT gateway version: 3.2"
""2023-01-16 09:44:27" - |INFO| - [tb_updater.py] - tb_updater - check_for_new_version - 84 - v3.2"
""2023-01-16 09:44:27" - |INFO| - [tb_updater.py] - tb_updater - check_for_new_version - 86 -

[===UPDATE===]

 New version v3.2 is available!

[===UPDATE===]
"
""2023-01-16 09:44:35" - |ERROR| - [tb_loader.py] - tb_loader - import_module - 72 - cannot import name 'ParamSpec' from 'typing_extensions' (/usr/lib/python3/dist-packages/typing_extensions.py)"
Traceback (most recent call last):
  File "/var/lib/thingsboard_gateway/.local/lib/python3.9/site-packages/twisted/internet/default.py", line 43, in _getInstallFunction
    from twisted.internet.epollreactor import install
  File "/var/lib/thingsboard_gateway/.local/lib/python3.9/site-packages/twisted/internet/epollreactor.py", line 19, in <module>
    from twisted.internet import posixbase
  File "/var/lib/thingsboard_gateway/.local/lib/python3.9/site-packages/twisted/internet/posixbase.py", line 16, in <module>
    from twisted.internet import error, tcp, udp
  File "/var/lib/thingsboard_gateway/.local/lib/python3.9/site-packages/twisted/internet/tcp.py", line 99, in <module>
    from twisted.internet import abstract, address, base, error, fdesc, main
  File "/var/lib/thingsboard_gateway/.local/lib/python3.9/site-packages/twisted/internet/base.py", line 34, in <module>
    from twisted.internet import abstract, defer, error, fdesc, main, threads
  File "/var/lib/thingsboard_gateway/.local/lib/python3.9/site-packages/twisted/internet/defer.py", line 42, in <module>
    from typing_extensions import Literal, ParamSpec, Protocol
ImportError: cannot import name 'ParamSpec' from 'typing_extensions' (/usr/lib/python3/dist-packages/typing_extensions.py)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/var/lib/thingsboard_gateway/.local/lib/python3.9/site-packages/twisted/internet/default.py", line 45, in _getInstallFunction
    from twisted.internet.pollreactor import install
[... 192 lines cut ...]
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 43, in <module>
    from twisted.internet import reactor
  File "/var/lib/thingsboard_gateway/.local/lib/python3.9/site-packages/twisted/internet/reactor.py", line 38, in <module>
    from twisted.internet import default
  File "/var/lib/thingsboard_gateway/.local/lib/python3.9/site-packages/twisted/internet/default.py", line 55, in <module>
    install = _getInstallFunction(platform)
[... cut ...]
""2023-01-16 09:44:43" - |INFO| - [tb_loader.py] - tb_loader - import_module - 68 - Import BACnetUplinkConverter from /usr/lib/python3/dist-packages/thingsboard_gateway/connectors/bacnet."
""2023-01-16 09:44:43" - |INFO| - [tb_loader.py] - tb_loader - import_module - 68 - Import BACnetDownlinkConverter from /usr/lib/python3/dist-packages/thingsboard_gateway/connectors/bacnet."
""2023-01-16 09:44:43" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 252 - Gateway started."
```

samson0v, 2023-01-17:
Hi @tamerfs, in what way are you installing Gateway?

tamerfs, 2023-01-17:
Hi @samson0v, via python3.
using this deb file 
wget https://github.com/thingsboard/thingsboard-gateway/releases/latest/download/python3-thingsboard-gateway.deb
and running the package in daemon 
sudo apt install ./python3-thingsboard-gateway.deb -y

samson0v, 2023-01-18:
@tamerfs, try to make freeze and see that versions of libs are right:
`pip3 freeze`
and send the result of the command.

tamerfs, 2023-01-18:
@samson0v 
 this was the results
```
pi@raspberrypi:~ $ pip3 freeze
arandr==0.1.10
astroid==2.5.1
asttokens==2.0.4
attrs==22.2.0
Automat==22.10.0
automationhat==0.2.0
beautifulsoup4==4.9.3
blinker==1.4
blinkt==0.1.2
buttonshim==0.0.2
cachetools==4.2.1
Cap1xxx==0.1.3
certifi==2020.6.20
chardet==4.0.0
click==7.1.2
colorama==0.4.4
colorzero==1.1
constantly==15.1.0
cryptography==3.3.2
cupshelpers==1.0
dbus-python==1.2.16
decorator==4.4.2
distro==1.5.0
docutils==0.16
drumhat==0.1.0
envirophat==1.0.0
ExplorerHAT==0.4.2
Flask==1.1.2
fourletterphat==0.1.0
gpiozero==1.6.2
grpcio==1.30.2
html5lib==1.1
hyperlink==21.0.0
idna==2.10
importlib==1.0.4
importlib-metadata==6.0.0
incremental==22.10.0
install==1.3.5
isort==5.6.4
itsdangerous==1.1.0
jedi==0.18.0
Jinja2==2.11.3
jsonpath-rw==1.4.0
jsonschema==3.1.1
lazy-object-proxy==0.0.0
logilab-common==1.8.1
lxml==4.6.3
MarkupSafe==1.1.1
mccabe==0.6.1
microdotphat==0.2.1
mmh3==3.0.0
mote==0.0.4
motephat==0.0.3
mypy==0.812
mypy-extensions==0.4.3
numpy==1.19.5
oauthlib==3.1.0
paho-mqtt==1.6.1
pantilthat==0.0.7
parso==0.8.1
pexpect==4.8.0
pgzero==1.2
phatbeat==0.1.1
pianohat==0.1.0
picamera==1.13
picamera2==0.3.8
pidng==4.0.9
piexif==1.1.3
piglow==1.2.5
pigpio==1.78
Pillow==8.1.2
ply==3.11
protobuf==3.12.4
psutil==5.8.0
pyasn1==0.4.8
pyasn1-modules==0.2.8
pycairo==1.16.2
pycups==2.0.1
pyfiglet==0.8.post0
pygame==1.9.6
Pygments==2.7.1
PyGObject==3.38.0
pyinotify==0.9.6
PyJWT==1.7.1
pylint==2.7.2
pymodbus==3.1.0
PyOpenGL==3.1.5
pyOpenSSL==20.0.1
PyQt5==5.15.2
PyQt5-sip==12.8.1
pyrsistent==0.19.3
pyserial==3.5
pysmbc==1.0.23
python-apt==2.2.1
python-prctl==1.7
PyYAML==5.3.1
rainbowhat==0.1.0
regex==2020.11.13
reportlab==3.5.59
requests==2.25.1
requests-oauthlib==1.0.0
responses==0.12.1
roman==2.0.0
RPi.GPIO==0.7.0
RTIMULib==7.2.1
scrollphat==0.0.7
scrollphathd==1.2.1
Send2Trash==1.6.0b1
sense-hat==2.4.0
service-identity==21.1.0
simplejpeg==1.6.4
simplejson==3.17.2
six==1.16.0
skywriter==0.0.7
sn3218==1.2.7
soupsieve==2.2.1
spidev==3.5
ssh-import-id==5.10
supervisor==4.2.2
tb-mqtt-client==1.4.1
termcolor==1.1.0
thingsboard-gateway==3.2
thonny==4.0.1
toml==0.10.1
touchphat==0.0.1
Twisted==22.10.0
twython==3.8.2
typed-ast==1.4.2
typing-extensions==4.4.0
unicornhathd==0.0.4
urllib3==1.26.5
v4l2-python3==0.3.1
webencodings==0.5.1
Werkzeug==1.0.1
wrapt==1.12.1
zipp==3.11.0
zope.interface==5.5.2
pi@raspberrypi:~ $

```

tamerfs, 2023-01-27:
@samson0v @imbeacon hi!

I have to freeze any version of python libraries?

Thanks!

samson0v, 2023-02-07:
Hi @tamerfs, it is a very strange problem because we tested Modbus on our raspberry pi and everything working well with the latest OS and default python 3.9.

tamerfs, 2023-02-14:
Hi @samson0v and @imbeacon!
im trying all again, from zero...

new raspiberry wiith new OS instalation on another enviroment with just one modbus device.
and im still having problems with the modbus connector....
can you guys help me?

thanks!!

```
Raspbian GNU/Linux 11 (bullseye)
TB-gateway v3.2
Python 3.9.2
```

the logs and de conf files =>
[14.02.23 syslog.txt](https://github.com/thingsboard/thingsboard-gateway/files/10733390/14.02.23.syslog.txt)
[modbus.json.txt](https://github.com/thingsboard/thingsboard-gateway/files/10733219/modbus.json.txt)
[tb_gateway.yaml.txt](https://github.com/thingsboard/thingsboard-gateway/files/10733220/tb_gateway.yaml.txt)

the configuration os modbus tcp/ip conversor =>
[image]

tamerfs, 2023-02-15:
i just made a new installation on another MSD CARD and i got same problem with modbus connector.
But in this time other problem =>
```
 File "/var/lib/thingsboard_gateway/.local/lib/python3.9/site-packages/twisted/internet/defer.py", line 42, in <module>
Feb 15 09:37:22 raspberrypi python3[1896]:     from typing_extensions import Literal, ParamSpec, Protocol
Feb 15 09:37:22 raspberrypi python3[1896]: ImportError: cannot import name 'ParamSpec' from 'typing_extensions' (/usr/lib/python3/dist-packages/typing_extensions.py)
Feb 15 09:37:22 raspberrypi python3[1896]: ""2023-02-15 09:37:22" - |WARNING| - [tb_gateway_service.py] - tb_gateway_service - _load_connectors - 547 - Connector implementation not found for Modbus Connector"
Feb 15 09:37:22 raspberrypi python3[1896]: ""2023-02-15 09:37:22" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 252 - Gateway started."

```

[15.02.23 modbus.json.txt](https://github.com/thingsboard/thingsboard-gateway/files/10742709/15.02.23.modbus.json.txt)
[15.02.23 syslog.txt](https://github.com/thingsboard/thingsboard-gateway/files/10742710/15.02.23.syslog.txt)
[15.02.23 tb_gateway.yaml.txt](https://github.com/thingsboard/thingsboard-gateway/files/10742711/15.02.23.tb_gateway.yaml.txt)

i made a diferent configuration on this modbus file, compared to the modbus file from yesterday. but both of thouse are based on TB documentation.

samson0v, 2023-02-16:
@tamerfs 
`pip install pyopenssl==22.0.0`

samson0v, 2023-03-01:
Hi @tamerfs, any updates?

wiseem, 2023-03-04:
@samson0v @tamerfs 
well, I solve this problem by downgrading the python version from 3.9.2 to 3.8.16

closed 2023-03-06 by samson0v

tamerfs, 2023-03-06:
@samson0v  in the end i have formated and reinstalled the gateway in another OS legacy and it worked...
@wiseem thks for the help!!