thingsboard/thingsboard-gateway #1045: [HELP] problem connection with TheThing Industries

emurilloEcourbes, 2023-01-17:
Hi,
I'm working with a MQTT connection from The thing industries, but I can not connect to Thingsboard IoT Gateway. The logs only show the message "MQTT Broker Connector has been stopped" and I don't know what happened.
If I write other host different to the thing industries, it connects correcty, the log shows 
```MQTT Broker Connector connected to broker.hivemq.com:1883 - successfully```

**tb_gateway.yaml** 
```thingsboard:
  host: mytb
  port: 1883
  remoteShell: false
  remoteConfiguration: false
  statistics:
    enable: true
    statsSendPeriodInSeconds: 3600
    configuration: statistics.json
  maxPayloadSizeBytes: 1024
  minPackSendDelayMS: 0
  checkConnectorsConfigurationInSeconds: 60
  handleDeviceRenaming: true
  checkingDeviceActivity:
    checkDeviceInactivity: false
    inactivityTimeoutSeconds: 120
    inactivityCheckPeriodSeconds: 10
  security:
    accessToken: XX
  qos: 1
storage:
  type: memory
  read_records_count: 100
  max_records_count: 100000
grpc:
  enabled: false
  serverPort: 9595
  keepaliveTimeMs: 10000
  keepaliveTimeoutMs: 5000
  keepalivePermitWithoutCalls: true
  maxPingsWithoutData: 0
  minTimeBetweenPingsMs: 10000
  minPingIntervalWithoutDataMs: 5000
connectors:
  -
    name: MQTT Broker Connector
    type: mqtt
    configuration: mqtt.json
```

**mqtt.json**
```
{
  "broker": {
    "name": "TTN broker",
    "host": "eu1.cloud.thethings.industries",
    "port": 1883,
    "security": {
      "type": "basic",
      "username": "sensors-air@tenant",
      "password": "NNSX"
    }
  },
  "mapping": [
    {
      "topicFilter": "v3/sensors-air@tenant/devices/id-plataforma-elevadora1/up",
      "converter": {
        "type": "json",
        "deviceNameTopicExpression": "Test device1",
        "timeout": 60000,
        "attributes": [
          {
            "type": "string",
            "key": "Modelo",
            "value": "Modelo 1"
          },
          {
            "type": "string",
            "key": "Serie",
            "value": "Serie 1"
          }
        ],
        "timeseries": [
          {
            "type": "double",
            "key": "temperature",
            "value": "21"
          }
        ]
      }
    }
  ],
  "connectRequests": [
    {
      "topicFilter": "v3/sensors-air@tenant/devices/id-plataforma-elevadora1/up",
      "deviceNameJsonExpression": "${payload.end_device_ids.device_id}"
    }
  ],
  "disconnectRequests": [],
  "attributeRequests": [],
  "attributeUpdates": [],
  "serverSideRpc": []
}
```

**Error traceback (If it was raised):**
```
""2023-01-17 06:53:50" - |INFO| - [mqtt_connector.py] - mqtt_connector - close - 295 - MQTT Broker Connector has been stopped."
""2023-01-17 06:53:50" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 243 - Number of accepted mapping handlers: 1"
""2023-01-17 06:53:50" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 247 - Number of rejected mapping handlers: 0"
```

**Versions (please complete the following information):**
 - Thingsboard IoT Gateway docker version 3.2
 - thingsboard/tb-postgres docker version 3.4.3

Thanks in advance.

samson0v, 2023-01-17:
Hi @emurilloEcourbes, replace all INFO to DEBUG in logs.conf file and send the logs.

emurilloEcourbes, 2023-01-17:
Hi @samson0v thanks for your quick response, I attached the log:

```
""2023-01-17 08:09:58" - |DEBUG| - [tb_logger.py] - tb_logger - __init__ - 39 - Added remote handler to log extension"
""2023-01-17 08:09:58" - |DEBUG| - [tb_logger.py] - tb_logger - __init__ - 39 - Added remote handler to log connector"
""2023-01-17 08:09:58" - |DEBUG| - [mqtt_connector.py] - mqtt_connector - load_handlers - 232 - Mandatory key 'topicFilter' found in mapping handler: {"topicFilter": "v3/sensors-airbus@airbustenant/devices/id-plataforma-elevadora1/up", "converter": {"type": "json", "deviceNameTopicExpression": "Test device1", "timeout": 60000, "attributes": [{"type": "string", "key": "Modelo", "value": "Modelo 1"}, {"type": "string", "key": "Serie", "value": "Serie 1"}], "timeseries": [{"type": "double", "key": "temperature", "value": "21"}]}}"
""2023-01-17 08:09:58" - |DEBUG| - [mqtt_connector.py] - mqtt_connector - load_handlers - 232 - Mandatory key 'converter' found in mapping handler: {"topicFilter": "v3/sensors-airbus@airbustenant/devices/id-plataforma-elevadora1/up", "converter": {"type": "json", "deviceNameTopicExpression": "Test device1", "timeout": 60000, "attributes": [{"type": "string", "key": "Modelo", "value": "Modelo 1"}, {"type": "string", "key": "Serie", "value": "Serie 1"}], "timeseries": [{"type": "double", "key": "temperature", "value": "21"}]}}"
""2023-01-17 08:09:58" - |DEBUG| - [mqtt_connector.py] - mqtt_connector - load_handlers - 240 - mapping handler has all mandatory keys => accepted: {"topicFilter": "v3/sensors-airbus@airbustenant/devices/id-plataforma-elevadora1/up", "converter": {"type": "json", "deviceNameTopicExpression": "Test device1", "timeout": 60000, "attributes": [{"type": "string", "key": "Modelo", "value": "Modelo 1"}, {"type": "string", "key": "Serie", "value": "Serie 1"}], "timeseries": [{"type": "double", "key": "temperature", "value": "21"}]}}"
""2023-01-17 08:09:58" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 243 - Number of accepted mapping handlers: 1"
""2023-01-17 08:09:58" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 247 - Number of rejected mapping handlers: 0"
""2023-01-17 08:09:58" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 243 - Number of accepted serverSideRpc handlers: 0"
""2023-01-17 08:09:58" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 247 - Number of rejected serverSideRpc handlers: 0"
""2023-01-17 08:09:58" - |DEBUG| - [mqtt_connector.py] - mqtt_connector - load_handlers - 232 - Mandatory key 'topicFilter' found in connectRequests handler: {"topicFilter": "v3/sensors-airbus@airbustenant/devices/id-plataforma-elevadora1/up", "deviceNameJsonExpression": "${payload.end_device_ids.device_id}"}"
""2023-01-17 08:09:58" - |DEBUG| - [mqtt_connector.py] - mqtt_connector - load_handlers - 240 - connectRequests handler has all mandatory keys => accepted: {"topicFilter": "v3/sensors-airbus@airbustenant/devices/id-plataforma-elevadora1/up", "deviceNameJsonExpression": "${payload.end_device_ids.device_id}"}"
""2023-01-17 08:09:58" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 243 - Number of accepted connectRequests handlers: 1"
""2023-01-17 08:09:58" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 247 - Number of rejected connectRequests handlers: 0"
""2023-01-17 08:09:58" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 243 - Number of accepted disconnectRequests handlers: 0"
""2023-01-17 08:09:58" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 247 - Number of rejected disconnectRequests handlers: 0"
""2023-01-17 08:09:58" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 243 - Number of accepted attributeRequests handlers: 0"
""2023-01-17 08:09:58" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 247 - Number of rejected attributeRequests handlers: 0"
""2023-01-17 08:09:58" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 243 - Number of accepted attributeUpdates handlers: 0"
""2023-01-17 08:09:58" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 247 - Number of rejected attributeUpdates handlers: 0"
""2023-01-17 08:09:58" - |DEBUG| - [mqtt_connector.py] - mqtt_connector - _on_disconnect - 397 - "MQTT Broker Connector" was disconnected. (<paho.mqtt.client.Client object at 0xffffad2ad4e0>, None, 7, None)"
""2023-01-17 08:09:59" - |DEBUG| - [mqtt_connector.py] - mqtt_connector - _on_disconnect - 397 - "MQTT Broker Connector" was disconnected. (<paho.mqtt.client.Client object at 0xffffad2ad4e0>, None, 7, None)"
""2023-01-17 08:10:01" - |DEBUG| - [mqtt_connector.py] - mqtt_connector - _on_disconnect - 397 - "MQTT Broker Connector" was disconnected. (<paho.mqtt.client.Client object at 0xffffad2ad4e0>, None, 7, None)"
""2023-01-17 08:10:05" - |DEBUG| - [mqtt_connector.py] - mqtt_connector - _on_disconnect - 397 - "MQTT Broker Connector" was disconnected. (<paho.mqtt.client.Client object at 0xffffad2ad4e0>, None, 7, None)"
""2023-01-17 08:10:14" - |DEBUG| - [mqtt_connector.py] - mqtt_connector - _on_disconnect - 397 - "MQTT Broker Connector" was disconnected. (<paho.mqtt.client.Client object at 0xffffad2ad4e0>, None, 7, None)"
""2023-01-17 08:10:31" - |DEBUG| - [mqtt_connector.py] - mqtt_connector - _on_disconnect - 397 - "MQTT Broker Connector" was disconnected. (<paho.mqtt.client.Client object at 0xffffad2ad4e0>, None, 7, None)"
""2023-01-17 08:11:03" - |DEBUG| - [mqtt_connector.py] - mqtt_connector - _on_disconnect - 397 - "MQTT Broker Connector" was disconnected. (<paho.mqtt.client.Client object at 0xffffad2ad4e0>, None, 7, None)"
```

samson0v, 2023-01-17:
@emurilloEcourbes, from logs I can see the next error code:
[image]

Does TheThing Industries support MQTT v.5?
If no, change the MQTT version in your MQTT Connector config file:
[image]

emurilloEcourbes, 2023-01-17:
Thanks @samson0v!! I changed the version and it was resolved.

Best!

closed 2023-01-17 by emurilloEcourbes

andonoc, 2023-01-17:
Thank you @samson0v has the same issue