thingsboard/thingsboard-gateway #1046: Get Data from Webservice SOAP (WSDL)

mamadouly12, 2023-01-17:
**Describe the issue**
We would like get datas using soap (wsdl file) as protocol of integration. But we are not sure how to get there 

**Configuration** (Attach your configuration file)
**Notate: Remove Access token from file if you want to attach a tb_gateway.yaml**

**Connector name (If you need help with some connector/converter):**
[e.g. MQTT Connector]

**Error traceback (If it was raised):**
```
'deviceName'
Traceback (most recent call last):
  File "<input>", line 2, in <module>
KeyError: 'deviceName'
```

**Versions (please complete the following information):**
 - OS: [e.g. Ubuntu 20.04]
 - Thingsboard IoT Gateway version [e.g. 3.0.1]
 - Python version[e.g. 3.9]

samson0v, 2023-01-18:
Hi @mamadouly12, Gateway doesn't support CoAP protocol, so you can use [ThingsBoard CoAP Integration](https://thingsboard.io/docs/user-guide/integrations/coap/).

mamadouly12, 2023-01-18:
Hi @samson0v : Thank you for your feedback, 
How do we call and consume the data from the wsdl file?

samson0v, 2023-01-18:
@mamadouly12 please, make a request to our support team
https://thingsboard.io/docs/contact-us/

closed 2023-01-24 by samson0v