thingsboard/thingsboard-gateway #1048: [HELP]Thingsboard-gateway can't connect to Modbus RTU device

HaoXuanSachi, 2023-01-29:
**Component**

 * Generic

**Description**
Thingsboard-gateway can't connect to Modbus RTU device

**Environment**

I got a problam in thingsboard-gateway that it cna't connect to device by Modbus RTU. I'm sure it can use app connect device and get respond in the same machine.  
```
root@gateway:/testmode# ./modpoll -b 19200 -p none -m rtu -a 1 -r 0x1b -c 1 /dev/ttyS0
modpoll 3.10 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright (c) 2002-2021 proconX Pty Ltd
Visit https://www.modbusdriver.com for Modbus libraries and tools.

Protocol configuration: Modbus RTU, FC3
Slave configuration...: address = 1, start reference = 27, count = 1
Communication.........: /dev/ttyS0, 19200, 8, 1, none, t/o 1.00 s, poll rate 1000 ms
Data type.............: 16-bit register, output (holding) register table

-- Polling slave... (Ctrl-C to stop)
[27]: 2590
```

Here is connector.log and modbus_serial.json
connector.log--
```
""2023-01-18 17:50:52" - |WARNING| - [backward_compability_adapter.py] - backward_compability_adapter - convert - 64 - You have to manually connect the new generated config file to tb_gateway.yaml!"
""2023-01-18 17:50:52" - |ERROR| - [modbus_connector.py] - modbus_connector - __process_slaves - 342 - int() argument must be a string, a bytes-like object or a number, not 'NoneType'"
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 311, in __process_slaves
    self.__connect_to_current_master(device)
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 357, in __connect_to_current_master
    device.config['master'], device.config['available_functions'] = self.__configure_master(device.config)
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 419, in __configure_master
    master = ModbusSerialClient(method=current_config["method"],
  File "/var/lib/thingsboard_gateway/.local/lib/python3.9/site-packages/pymodbus/client/serial.py", line 193, in __init__
    super().__init__(framer=framer, **kwargs)
  File "/var/lib/thingsboard_gateway/.local/lib/python3.9/site-packages/pymodbus/client/base.py", line 120, in __init__
    self.params.retries = int(retries)
TypeError: int() argument must be a string, a bytes-like object or a number, not 'NoneType'
```

modbus_serial.json--
```
{
  "master": {
    "server": {
    "name": "Modbus Default Server",
    "type": "serial",
    "method": "rtu",
    "port": "/dev/ttyS0",
    "baudrate": 19200,
    "timeout": 35,
    "devices": [
      {
        "unitId": 1,
        "deviceName": "iAeris 20",
        "attributesPollPeriod": 5000,
        "timeseriesPollPeriod": 5000,
        "sendDataOnlyOnChange": false,
        "attributes": [
          {
            "byteOrder": "BIG",
            "tag": "temperature",
            "type": "bytes",
            "functionCode": 3,
            "registerCount": 1,
            "address": 26
          }
        ]
      }
    ]
  }
  }
}
```

I'm not sure there is any problam in my setting
gateway status--
```
root@gateway:/testmode# systemctl status thingsboard-gateway.service
● thingsboard-gateway.service - ThingsBoard Gateway
     Loaded: loaded (/etc/systemd/system/thingsboard-gateway.service; enabled; vendor preset: enabled)
     Active: active (running) since Thu 2023-01-19 16:00:11 CST; 1s ago
   Main PID: 2742 (python3)
      Tasks: 13 (limit: 9294)
     Memory: 38.3M
        CPU: 1.463s
     CGroup: /system.slice/thingsboard-gateway.service
             └─2742 /usr/bin/python3 -c from thingsboard_gateway.tb_gateway import daemon; daemon()

Jan 19 16:00:12 gateway python3[2742]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.9/site-packages/pymodbus/client/base.py", line 120, in __init__
Jan 19 16:00:12 gateway python3[2742]:     self.params.retries = int(retries)
Jan 19 16:00:12 gateway python3[2742]: TypeError: int() argument must be a string, a bytes-like object or a number, not 'NoneType'
Jan 19 16:00:12 gateway python3[2742]: ""2023-01-19 16:00:12" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 252 - Gateway started."
Jan 19 16:00:12 gateway python3[2742]: ""2023-01-19 16:00:12" - |INFO| - [tb_updater.py] - tb_updater - check_for_new_version - 84 - v3.2"
Jan 19 16:00:12 gateway python3[2742]: ""2023-01-19 16:00:12" - |INFO| - [tb_updater.py] - tb_updater - check_for_new_version - 86 -
Jan 19 16:00:12 gateway python3[2742]: [===UPDATE===]
Jan 19 16:00:12 gateway python3[2742]:  New version v3.2 is available!
Jan 19 16:00:12 gateway python3[2742]: [===UPDATE===]
Jan 19 16:00:12 gateway python3[2742]: "

```

 * OS:  Debian11
 * ThingsBoard-gateway: v3.2
 * Browser: name and version

samson0v, 2023-02-03:
Hi @HaoXuanSachi, any updates?

HaoXuanSachi, 2023-02-06:
HI thanks you a lot. It can work when i change [service device] to  [master slaves].
But i still unknow what different about this two. It still cna't  work when i change [master slaves] to [service device] in same format.

samson0v, 2023-02-06:
Hi @HaoXuanSachi, read, please, the official [documentation](https://thingsboard.io/docs/iot-gateway/config/modbus/).

HaoXuanSachi, 2023-02-07:
thank you a lot. i know what happend in my  yaml. thank you.

closed 2023-02-07 by HaoXuanSachi