thingsboard/thingsboard-gateway #1051: [HELP] tb_gateway_service - __send_to_storage - 752 - 'str' object has no attribute 'get'

zlwjzf, 2023-02-01:
### Hi, I have a problem that causes loss of telemetry data received by Thingsboard. Please help with this.

### tb-gateway log errors are displayed as follows
`""2023-02-01 15:33:11" - |INFO| - [MqttHHYLConverter.py] - MqttHHYLConverter - convert - 12 - New data received: tele/CNOVIT_1EE464/SENSOR: {'ZbReceived': {'0xAD88': {'Device': '0xAD88', 'Name': '门磁', 'BatteryPercentage': 86.5, 'Endpoint': 1, 'LinkQuality': 102}}}"
""2023-02-01 15:33:11" - |INFO| - [MqttHHYLConverter.py] - MqttHHYLConverter - convert - 68 - Result: {'deviceName': '0xAD88', 'telemetry': [{'BatteryPercentage': 86.5, 'Endpoint': 1, 'LinkQuality': 102, 'Ts': 1675236791791}], 'attributes': [{'GateWay': 'CNOVIT_1EE464', 'Name': '门磁', 'Device': '0xAD88'}], 'deviceType': '门磁'}"
""2023-02-01 15:33:11" - |INFO| - [MqttHHYLConverter.py] - MqttHHYLConverter - convert - 12 - New data received: tele/CNOVIT_1EE464/SENSOR: {'ZbReceived': {'0xFC5B': {'Device': '0xFC5B', 'Name': '体征垫', '0704?01': '0000426461746132010000', 'Endpoint': 1, 'LinkQuality': 173}}}"
""2023-02-01 15:33:11" - |INFO| - [MqttHHYLConverter.py] - MqttHHYLConverter - convert - 68 - Result: {'deviceName': '0xFC5B', 'telemetry': [{'0704?01': '0000426461746132010000', 'Endpoint': 1, 'LinkQuality': 173, 'Ts': 1675236791791}], 'attributes': [{'GateWay': 'CNOVIT_1EE464', 'Name': '体征垫', 'Device': '0xFC5B'}], 'deviceType': '体征垫'}"
""2023-02-01 15:33:11" - |ERROR| - [tb_gateway_service.py] - tb_gateway_service - __send_to_storage - 752 - 'str' object has no attribute 'get'"`

I am using the MQTT custom pattern, and here is the configuration file
`from simplejson import dumps
from thingsboard_gateway.connectors.mqtt.mqtt_uplink_converter import MqttUplinkConverter, log
import time

class MqttHHYLConverter(MqttUplinkConverter):
    def __init__(self, config):
        self.__config = config.get('converter')
        self.dict_result = {}

    def convert(self, topic, body):
        try:
            log.info("New data received: %s: %s" % (topic,body))
            topicarray=topic.split("/") 
            telemetry={}
            attributes={}
            attributes['GateWay']= topicarray[1]
            if  ("ZbInfo" in  body) : 
                deviceinfo=body['ZbInfo'] 
                for data in deviceinfo: 
                  devicebody=deviceinfo[data]
                  self.dict_result["deviceName"] = data
                  attributes=devicebody 
            if  ("ZbReceived" in  body) : 
                deviceinfo=body['ZbReceived']
                for data in deviceinfo: 
                  devicebody=deviceinfo[data]
                  self.dict_result["deviceName"] = data
                  deviceType = devicebody['Name']
                  if deviceType != '体征垫' and deviceType !='如厕采集' and deviceType != '烟雾报警' and deviceType != '门磁' and deviceType != 'SOS按键' and deviceType != '语音呼救':
                      return []
                  attributes['Name']=devicebody['Name']
                  attributes['Device']=devicebody['Device']
                  devicebody.pop('Name')
                  devicebody.pop('Device')
                  telemetry=devicebody
                  now = int(round(time.time() * 1000))
                  telemetry['Ts'] =now
            else:
                 print('无')
                 return []
           
            self.dict_result["telemetry"] = [telemetry]
            self.dict_result["attributes"] = [attributes]
            self.dict_result["deviceType"] = deviceType
            log.info("Result: %s" % (self.dict_result))
            return self.dict_result

        except Exception as e:
            log.exception('Error in converter, for config: \n%s\n and message: \n%s\n', dumps(self.__config), body)
            log.exception(e)`

samson0v, 2023-02-01:
Hi @zlwjzf , thanks for your question. I will investigate your code and let you know about solution.

zlwjzf, 2023-02-03:
@samson0v  Excuse me, is there any progress? What is the problem?

samson0v, 2023-02-03:
Hi @zlwjzf, send, please, your config file.

zlwjzf, 2023-02-03:
Hi @samson0v, The following is the configuration of Mqtt.json

{
  "broker": {
    "name": "mqtt_hhyl",
    "host": "******",
    "port": 8888,
    "clientId": "mqttx_device_clientId_hhyl",
    "maxMessageNumberPerWorker": 10,
    "maxNumberOfWorkers": 100,
    "security": {
      "type": "basic",
      "username": "******",
      "password": "******"
    }
  },
  "mapping": [
    {
      "topicFilter":"tele/+/SENSOR",
      "converter": {
        "type": "custom",
        "extension": "MqttHHYLConverter",
	"extension-config":null
      }
    }
  ],
  "connectRequests": [],
  "disconnectRequests": [],
  "attributeRequests": [],
  "attributeUpdates": [],
  "serverSideRpc": []
}

samson0v, 2023-02-03:
@zlwjzf, I don't see any problem with data sending:
[image]
[image]

zlwjzf, 2023-02-06:
Hi @samson0v, When the error occurred in the Tb-gateway, I also tried to send the failed data separately. At this time, the TB could receive the data. However, when tracing logs, it is found that the first message fails to be sent because the tb-gateway receives two messages at the same time.

zlwjzf, 2023-02-06:
Hi @samson0v , Note that the TS (Ms timestamp) fields of the two data received in the tb-gateway log are the same. Is it possible that the error occurs only when the tb-gateway receives multiple messages at the same time?

samson0v, 2023-02-06:
> Hi @samson0v , Note that the TS (Ms timestamp) fields of the two data received in the tb-gateway log are the same. Is it possible that the error occurs only when the tb-gateway receives multiple messages at the same time?

@zlwjzf, no

samson0v, 2023-02-06:
@zlwjzf, from the data that you send to Gateway, Gateway doesn't use your timestamp, so it is use current receiving time.
If you want to use your ts, you have to change from `Ts` to `ts`.

zlwjzf, 2023-02-08:
Hi @samson0v ，I trace the logs of the tb-gateway for one week and find no other rule. The above error occurs only when the tb-gateway receives two messages at the same time. Could you please tell me the solution or preventive measures to this problem?

zlwjzf, 2023-02-08:
Hi @samson0v , I installed mosquitto and tb-gateway on a computer, and configured mqtt connection according to the above, and used two MQTTX clients to push two pieces of data in the error log periodically respectively (one for 5 seconds, one for 1 second), to simulate this problem, please try to simulate this way.

zlwjzf, 2023-02-08:
[error.zip](https://github.com/thingsboard/thingsboard-gateway/files/10683853/error.zip)
Hi @samson0v , I have located the problem, data.get("telemetry") in TBUtility.py, please check.

samson0v, 2023-02-08:
Hi @zlwjzf, it is very strange behavior, try to reinstall Gateway and use the latest version from the master branch.

zlwjzf, 2023-02-08:
Hi @samson0v , The Tb-gateway I am using is the latest version (3.2), and the gateway device on the TB is also new (specifically for testing).

samson0v, 2023-02-08:
@zlwjzf, okay, will try to simulate the same behavior

samson0v, 2023-02-10:
@zlwjzf, converter output data have to be in the next format:
`telemetry: {ts:..., values:[{"test":"test"}]}`

zlwjzf, 2023-02-13:
> telemetry: {ts:..., values:[{"test":"test"}]}

`if data.get("telemetry") is not None:`
    `    for entry in data.get("telemetry"):`
        `        if (entry.get("ts") is not None and len(entry.get("values")) > 0) or entry.get("ts") is None:`
Hi @samson0v , entry.get("ts") will report an error if you follow this format conversion; telemetry should be followed by [].

zlwjzf, 2023-02-13:
`{'deviceName': 'test_zl_tzd', 'telemetry': [{'ts': 1676256643030, 'values': {'0704?01': '0000426461746128010000', 'Endpoint': 1, 'LinkQuality': 160}}], 'attributes': {'GateWay': 'CNOVIT_1EE464', 'Name': 'test1', 'Device': 'test_zl_tzd'}, 'deviceType': 'test1'}`
Hi @samson0v , This is my output format, this format should not appear the problem of description, I need to observe for a few days.
Here is the log from the simulation (no error print):
`""2023-02-13 10:50:43" - |INFO| - [MqttHHYLConverter.py] - MqttHHYLConverter - convert - 12 - New data received: tele/CNOVIT_1EE464/SENSOR: {'ZbReceived': {'test_zl_tzd': {'Device': 'test_zl_tzd', 'Name': 'test1', '0704?01': '0000426461746128010000', 'Endpoint': 1, 'LinkQuality': 160}}}"`
`""2023-02-13 10:50:43" - |INFO| - [MqttHHYLConverter.py] - MqttHHYLConverter - convert - 53 - Result1: {'deviceName': 'test_zl_tzd', 'telemetry': [{'ts': 1676256643030, 'values': {'0704?01': '0000426461746128010000', 'Endpoint': 1, 'LinkQuality': 160}}], 'attributes': {'GateWay': 'CNOVIT_1EE464', 'Name': 'test1', 'Device': 'test_zl_tzd'}, 'deviceType': 'test1'}"
`
`""2023-02-13 10:50:43" - |INFO| - [MqttHHYLConverter.py] - MqttHHYLConverter - convert - 12 - New data received: tele/CNOVIT_1EE464/SENSOR: {'ZbReceived': {'test_zl_mc': {'Device': 'test_zl_mc', 'Name': 'test2', 'BatteryPercentage': 86.5, 'Endpoint': 1, 'LinkQuality': 102}}}"
`
`""2023-02-13 10:50:43" - |INFO| - [MqttHHYLConverter.py] - MqttHHYLConverter - convert - 53 - Result1: {'deviceName': 'test_zl_mc', 'telemetry': [{'ts': 1676256643231, 'values': {'BatteryPercentage': 86.5, 'Endpoint': 1, 'LinkQuality': 102}}], 'attributes': {'GateWay': 'CNOVIT_1EE464', 'Name': 'test2', 'Device': 'test_zl_mc'}, 'deviceType': 'test2'}"
`

zlwjzf, 2023-02-13:
Hi @samson0v , I occasionally found data overwriting problems during simulation tests, such as the following log printing:
`""2023-02-13 14:12:27" - |INFO| - [MqttHHYLConverter.py] - MqttHHYLConverter - convert - 12 - New data received: tele/CNOVIT_1EE464/SENSOR: {'ZbReceived': {'test_zl_mc': {'Device': 'test_zl_mc', 'Name': 'test2', 'BatteryPercentage': 86.5, 'Endpoint': 1, 'LinkQuality': 102}}}"
`
`""2023-02-13 14:12:27" - |INFO| - [MqttHHYLConverter.py] - MqttHHYLConverter - convert - 53 - Result1: {'deviceName': 'test_zl_mc', 'telemetry': [{'ts': 1676268747802, 'values': {'BatteryPercentage': 86.5, 'Endpoint': 1, 'LinkQuality': 102, 'Ts': 1676268747802}}], 'attributes': {'GateWay': 'CNOVIT_1EE464', 'Name': 'test2', 'Device': 'test_zl_mc'}, 'deviceType': 'test2'}"
`
`""2023-02-13 14:12:28" - |INFO| - [MqttHHYLConverter.py] - MqttHHYLConverter - convert - 12 - New data received: tele/CNOVIT_1EE464/SENSOR: {'ZbReceived': {'test_zl_tzd': {'Device': 'test_zl_tzd', 'Name': 'test1', '0704?01': '0000426461746128010000', 'Endpoint': 1, 'LinkQuality': 160}}}"
`
`""2023-02-13 14:12:28" - |INFO| - [MqttHHYLConverter.py] - MqttHHYLConverter - convert - 53 - Result1: {'deviceName': 'test_zl_tzd', 'telemetry': [{'ts': 1676268748004, 'values': {'0704?01': '0000426461746128010000', 'Endpoint': 1, 'LinkQuality': 160, 'Ts': 1676268748004}}], 'attributes': {'GateWay': 'CNOVIT_1EE464', 'Name': 'test1', 'Device': 'test_zl_tzd'}, 'deviceType': 'test1'}"
`
`""2023-02-13 14:12:28" - |INFO| - [tb_utility.py] - tb_utility - validate_converted_data - 56 - 1: {'deviceName': 'test_zl_tzd', 'telemetry': [{'ts': 1676268748004, 'values': {'0704?01': '0000426461746128010000', 'Endpoint': 1, 'LinkQuality': 160, 'Ts': 1676268748004}}], 'attributes': {'GateWay': 'CNOVIT_1EE464', 'Name': 'test1', 'Device': 'test_zl_tzd'}, 'deviceType': 'test1'}"
`
`""2023-02-13 14:12:28" - |INFO| - [tb_utility.py] - tb_utility - validate_converted_data - 58 - 2: {'deviceName': 'test_zl_tzd', 'telemetry': [{'ts': 1676268748004, 'values': {'0704?01': '0000426461746128010000', 'Endpoint': 1, 'LinkQuality': 160, 'Ts': 1676268748004}}], 'attributes': {'GateWay': 'CNOVIT_1EE464', 'Name': 'test1', 'Device': 'test_zl_tzd'}, 'deviceType': 'test1'}"
`
`""2023-02-13 14:12:28" - |INFO| - [tb_utility.py] - tb_utility - validate_converted_data - 60 - 3: {'deviceName': 'test_zl_tzd', 'telemetry': [{'ts': 1676268748004, 'values': {'0704?01': '0000426461746128010000', 'Endpoint': 1, 'LinkQuality': 160, 'Ts': 1676268748004}}], 'attributes': {'GateWay': 'CNOVIT_1EE464', 'Name': 'test1', 'Device': 'test_zl_tzd'}, 'deviceType': 'test1'}"
`
`""2023-02-13 14:12:28" - |INFO| - [tb_utility.py] - tb_utility - validate_converted_data - 56 - 1: {'deviceName': 'test_zl_tzd', 'telemetry': [{'ts': 1676268748004, 'values': {'0704?01': '0000426461746128010000', 'Endpoint': 1, 'LinkQuality': 160, 'Ts': 1676268748004}}], 'attributes': {'GateWay': 'CNOVIT_1EE464', 'Name': 'test1', 'Device': 'test_zl_tzd'}, 'deviceType': 'test1'}"
`
`""2023-02-13 14:12:28" - |INFO| - [tb_utility.py] - tb_utility - validate_converted_data - 58 - 2: {'deviceName': 'test_zl_tzd', 'telemetry': [{'ts': 1676268748004, 'values': {'0704?01': '0000426461746128010000', 'Endpoint': 1, 'LinkQuality': 160, 'Ts': 1676268748004}}], 'attributes': {'GateWay': 'CNOVIT_1EE464', 'Name': 'test1', 'Device': 'test_zl_tzd'}, 'deviceType': 'test1'}"
`
`""2023-02-13 14:12:28" - |INFO| - [tb_utility.py] - tb_utility - validate_converted_data - 60 - 3: {'deviceName': 'test_zl_tzd', 'telemetry': [{'ts': 1676268748004, 'values': {'0704?01': '0000426461746128010000', 'Endpoint': 1, 'LinkQuality': 160, 'Ts': 1676268748004}}], 'attributes': {'GateWay': 'CNOVIT_1EE464', 'Name': 'test1', 'Device': 'test_zl_tzd'}, 'deviceType': 'test1'}"
`
The data of the device name test_zl_tzd overwrites the data of the device name test_zl_mc，
Here is a screenshot of the telemetry data received by the test_zl_mc device on TB:
[image]

Here is a screenshot of the telemetry data received by the test_zl_tzd device on TB:
[image]

samson0v, 2023-03-01:
@zlwjzf, any updates? Can we close the issue?

zlwjzf, 2023-03-02:
Hi @samson0v 
Hello, I found in the reporting frequency of the test second level:

The amount of data received by the TB in a period is inconsistent with the actual data sent (The TB receives less data).

zlwjzf, 2023-03-03:
Hi @samson0v ,
I found that the problem may be TB, 610 records were successfully printed in the debug log of the save timeseries node in the rule chain, while only 577 were stored in the database
[image]

[image]

samson0v, 2023-03-13:
@zlwjzf, please, create an issue in the [ThingsBoard repo](https://github.com/thingsboard/thingsboard/issues) if the issue is on its side.

zlwjzf, 2023-03-23:
Hi @samson0v， Thank you. You can close the issue now。

closed 2023-03-23 by samson0v