thingsboard/thingsboard-gateway #1052: [BUG] ThingsBoard Gateway does not send telemetry

Steampunkery, 2023-02-02:
**Describe the bug**
More recent installations of thingsboard gateway are not sending telemetry from our BLE devices. The gateway will connect to the device. It will print a log saying that it can't find an attribute characteristic (normal), but it will never send any telemetry. Installations installed back in May of 2022 are still functional. 

Things we have tried:
 - Copying the thingsboard installation folder from an older working installation to a new machine. Result: no change
 - Trying the latest version 3.2. Result: No change
 - Fresh installing the entire OS: Result: No change
 - Getting another RPI4 to rule out hardware issue: No change
 

This leads us to believe that this is either a dependency issue or an OS issue. See logs below for more context. We have been working on this issue for many hours, so please don't hesitate to ask if we've tried something.

**Connector name (If bug in the some connector):**
BLE connector (others untested)

**LOGS**
https://pastebin.com/meN4yVn6
The issue starts around line 134 and again at 247.

**Versions (please complete the following information):**
 - OS: Raspberry Pi OS
 - Thingsboard IoT Gateway version 3.1 & 3.2
 - Python 3.9.2

samson0v, 2023-02-03:
Hi @Steampunkery, thanks for your interest in ThingsBoard IoT Gateway.
Send, please, your configuration files.

Steampunkery, 2023-02-06:
Here are my relevant configuration files.

[config.zip](https://github.com/thingsboard/thingsboard-gateway/files/10668151/config.zip)

samson0v, 2023-02-07:
Hi @Steampunkery, are you sure that you send the right config files? Because there is a difference between logs char and config char.

Steampunkery, 2023-02-08:
Those configuration files are identical to the configurations used with the exception of the characteristic value. The bug appears no matter the value of the characteristic. Even if you use one of the bluetooth standard device information characteristics.

samson0v, 2023-02-10:
@Steampunkery, make sure that your Gateway uses the right config files. Because of logs, I can see that you also use MQTT Connector but in the config file, you are using only BLE Connector.
I recommend you delete all Gateway folders and make a clean installation.

Steampunkery, 2023-02-10:
Hi, thanks for the suggestion. We have taken these steps more than 10 times. We are fairly sure it is an issue with dependencies or some other systemic issue because we are able to copy a working version from an older OS installation to a newer OS installation, and the problem still persists.

samson0v, 2023-02-16:
@Steampunkery, "clean installation", I mean this instruction: https://github.com/thingsboard/thingsboard-gateway/issues/977#issuecomment-1300244839

samson0v, 2023-03-01:
@Steampunkery, any updates?

Steampunkery, 2023-03-02:
Hi, I haven't had a chance to take those steps yet. Has anyone looked into the bug on your side? I'll try to execute the steps soon.

Steampunkery, 2023-03-02:
One of our team members has run the installation on a completely fresh version of Debian 11 (bullseye), and is experiencing the same error. We have some bullseye machines that are working, and some that aren't. The only discernable difference is that the working machines were set up before the non-working machines. This points to an error in the system or the way that thingsboard gateway is configured.

samson0v, 2023-03-13:
Hi @Steampunkery, let's continue our chat on the email. Please, send me a message to vbidochka@thingsboard.io

Mannetjie, 2023-03-20:
Have this been resolved as I am experiencing exactly the same issue. Clean install, following the getting started guide. The device gets created, all logs clean (no errors) BUT absolutely no telemetry data

Steampunkery, 2023-03-20:
No, it hasn't been resolved on my end yet.

Thomas
ASCII and ye shall receive

Mannetjie, 2023-03-20:
Hi Thomas

I got mine working. Here is what I did so far and I hope this helps you.
1: I stopped the gateway service.
2: Ran the thingsboard-gateway manually.
This showed me that there is some error in the statistics file. So I
disabled that for now in the config file and ran it again manually.
Then all of the sudden things started flowing.
I did get an error on one of the threads at this stage and it
stopped working again. Ran it again manually and all goes well.

I started the service again and it did run for a few messages and then just
stopped working again. Stopped the service and started it manually again
and all is fine again.
I am trying to reproduce the thread error and then I will have a look at
the statistics error.

I truly hope this helps you. I will keep you up to date on my findings.

Regards
Mannetjie

-- 

*Mannetjie Van Rooyen*
*Senior Developer*

011 317 4900
060 076 6750 Google Certified Professional Cloud Architect
Google Certified Professional Data Engineer
Google Certified Professional Developer
Certified Kubernetes Application Developer
monday.com Developers Badge
***@***.***
www.opennetworks.com
Johannesburg, Cape Town, South Africa

closed 2024-03-08 by imbeacon