thingsboard/thingsboard-gateway #1056: [HELP] Modbus Error: [Input/Output] [Errno 104] Connection reset by peer

WoofQuack, 2023-02-16:
**Describe the issue**
Receiving 'Connection Reset by Peer' and 'Timeout' errors in converter log after a couple hours of working fine.
```
""2023-02-15 17:58:28" - |ERROR| - [bytes_modbus_uplink_converter.py] - bytes_modbus_uplink_converter - convert - 84 - Modbus Error: [Input/Output] timed out"
NoneType: None
""2023-02-15 17:59:04" - |ERROR| - [bytes_modbus_uplink_converter.py] - bytes_modbus_uplink_converter - convert - 84 - Modbus Error: [Input/Output] [Errno 104] Connection reset by peer"
```
Restarting the gateway will fix the issue but error occurs and is consistent after a couple hours of running.
Data received from devices are set to poll every 5000ms, but when this error occurs it can take up to 10mins between receiving updates from device.

**Configuration** (Attach your configuration file)
modbus.json config:
```
{
  "master":{
    "slaves":[
      {
        "byteOrder":"BIG",
        "wordOrder":"LITTLE",
        "connectAttemptCount":5,
        "connectAttemptTimeMs":15000,
        "deviceName":"Panel 1 Gas Sensors",
        "deviceType":"Gas Sensor Panel",
        "host":"10.4.15.12",
        "method":"socket",
        "pollPeriod":5000,
        "port":502,
        "retries":true,
        "retryOnEmpty":true,
        "retryOnInvalid":true,
        "sendDataOnlyOnChange":false,
        "timeout":35,
        "attributes": [
          {
            "address":40,
            "functionCode":4,
            "objectsCount":2,
            "tag":"Gas1_MinTol",
            "type":"32float"
          },
          {
            "address":42,
            "functionCode":4,
            "objectsCount":2,
            "tag":"Gas2_MinTol",
            "type":"32float"
          },
          {
            "address":44,
            "functionCode":4,
            "objectsCount":2,
            "tag":"Gas3_MinTol",
            "type":"32float"
          },
          {
            "address":46,
            "functionCode":4,
            "objectsCount":2,
            "tag":"Gas4_MinTol",
            "type":"32float"
          },
          {
            "address":48,
            "functionCode":4,
            "objectsCount":2,
            "tag":"Gas5_MinTol",
            "type":"32float"
          },
          {
            "address":60,
[... 976 lines cut ...]

**Error traceback (If it was raised):**
```
""2023-02-15 17:58:28" - |ERROR| - [bytes_modbus_uplink_converter.py] - bytes_modbus_uplink_converter - convert - 84 - Modbus Error: [Input/Output] timed out"
NoneType: None
""2023-02-15 17:59:04" - |ERROR| - [bytes_modbus_uplink_converter.py] - bytes_modbus_uplink_converter - convert - 84 - Modbus Error: [Input/Output] [Errno 104] Connection reset by peer"
NoneType: None
""2023-02-15 17:59:39" - |ERROR| - [bytes_modbus_uplink_converter.py] - bytes_modbus_uplink_converter - convert - 84 - Modbus Error: [Input/Output] [Errno 104] Connection reset by peer"
NoneType: None
""2023-02-15 18:00:14" - |ERROR| - [bytes_modbus_uplink_converter.py] - bytes_modbus_uplink_converter - convert - 84 - Modbus Error: [Input/Output] [Errno 104] Connection reset by peer"
NoneType: None
""2023-02-15 18:07:14" - |ERROR| - [bytes_modbus_uplink_converter.py] - bytes_modbus_uplink_converter - convert - 84 - Modbus Error: [Input/Output] timed out"
NoneType: None
""2023-02-15 18:07:14" - |ERROR| - [bytes_modbus_uplink_converter.py] - bytes_modbus_uplink_converter - convert - 84 - Modbus Error: [Input/Output] timed out"
NoneType: None
```

**Versions (please complete the following information):**
 - OS: Ubuntu 22.04.1 LTS 
 - Docker version 20.10.21
 - Thingsboard IoT Gateway version 3.2
 - Thingsboard Edge version 3.4.3EDGEPE
 - Python version: 3.10.6

Thanks! Any help would be appreciated.

WoofQuack, 2023-02-17:
It seems when the error occurs, the first telemetry key is missing from the post telemetry message when viewed in the rule chain.
For example, 'Fan1Run' is missing from the message received in the rule chain.
This missing first key in the timeseries message is evident across ALL the devices defined in the modbus config.
```
{
    "Fan2Run": 1,
    "Fan3Run": 1,
    "Fan1Overload": 0,
    "Fan2Overload": 0,
    "Fan3Overload": 0,
    "Fan1RuntimeThisDay": 525.0,
    "Fan1PowerConsumedThisDay": 962.5,
    "Fan2RuntimeThisDay": 525.0,
    "Fan2PowerConsumedThisDay": 962.5,
    "Fan3RuntimeThisDay": 525.0,
    "Fan3PowerConsumedThisDay": 962.5,
    "FanSumPowerConsumedThisDay": 2887.5,
    "Sent": false
}
```

samson0v, 2023-02-20:
Hi @WoofQuack, thanks for your interest in ThingsBoard IoT Gateway!
Currently, I investigate your problem and let you know about the result.

closed 2023-02-23 by imbeacon