thingsboard/thingsboard-gateway #1059: [HELP] OCPP Connector: Sending Call requests to Charge Points

DriftingShadows, 2023-02-17:
**Describe the issue**
I have successfully added charge points to thingsboard via the OCPP connector. I am having trouble figuring out how to send requests (eg. StartTransaction) from thingsboard to the Charge Point. Any help appreciated.

**Connector name (If you need help with some connector/converter):**
OCPP Connector

samson0v, 2023-02-20:
Hi @DriftingShadows, thanks for your interest in ThingsBoard IoT Gateway!
Currently, it is unavailable, OCPP Connector is in the early development stage, so we planning to add RPCs with your request.
For now, OCPP Connector can automatically authorize Charged Points and accept BootNotification, Heartbeat, MeterValues and DataTransfer.

imbeacon, 2023-02-20:
Hi @samson0v, I think we should add this ability to 3.3 or 3.4.

Anmirazik, 2023-03-19:
hi @DriftingShadows  , can you please share your ocpp.json and tb_gateway.yaml file ? I've tried configure the json file but it seems like I'm not able to connect the charger to the thingsboard so that the thingsboard will act as the central system for the ev charger , btw Im using docker image of tb-gateway which i pulled from docker hub

closed 2026-09-11 by samson0v