thingsboard/thingsboard-gateway #1060: [HELP]How to get telemetry data through rule chain node，I tried, but it didn't work。

a-jiang, 2023-02-17:
**Describe the issue**
[image]
[image]
[image]
[image]

**Connector name (If you need help with some connector/converter):**
[e.g. MQTT Connector]

**Error traceback (If it was raised):**
```
'deviceName'
Traceback (most recent call last):
  File "<input>", line 2, in <module>
KeyError: 'deviceName'
```

**Versions (please complete the following information):**
 - OS: [e.g. Ubuntu 20.04]
 - Thingsboard IoT Gateway version [e.g. 3.0.1]
 - Python version[e.g. 3.9]

imbeacon, 2023-02-17:
Hi @a-jiang ,

You can enable debug on rule node, save rule chain, wait for the data and open **Events** tab for this rule-node in rule chain, it will give you ability to see the incoming messages and then build your script. [Debugging](https://thingsboard.io/docs/user-guide/rule-engine-2-0/overview/#debugging)

a-jiang, 2023-02-17:
> Hi @a-jiang ,
>
> You can enable debug on rule node, save rule chain, wait for the data and open **Events** tab for this rule-node in rule chain, it will give you ability to see the incoming messages [...]

Thank you very much, I found out that the data I got is not the raw data I want through debug mode.
[image]
**my raw data like this**
`{'body': {'serviceId': 'analog', 'nodeId': 'kd_huawei_pvi1', 'serviceProperties': {'PV_DC': '60.44', 'PV_TC': '18048.98', 'ERROR_CODE': '0.00', 'PV_ST': '1676621429.00', 'PV_T': '34.00', 'PV_EF': '98.24', 'PV_FQ': '50.03', 'PV_PF': '1.00', 'PV_OQP': '-0.02', 'PV_OAP': '6.34', 'PV_HP': '24.62', 'PV_UIC': '9.04', 'PV_UIB': '9.06', 'PV_UIA': '9.05', 'PV_UVC': '232.40', 'PV_UVB': '234.00', 'PV_UVA': '232.50', 'PV_CA': '401.10', 'PV_BC': '402.60', 'PV_AB': '402.10', 'PV_IP': '6.45', 'PV4_A': '2.78', 'PV4_V': '577.20', 'PV3_A': '2.74', 'PV3_V': '574.60', 'PV2_A': '2.84', 'PV2_V': '577.10', 'PV1_A': '2.78', 'PV1_V': '581.30', 'warn2': '0.00', 'warn1': '0.00', 'warn': '0.00', 'staus3': '7.00', 'staus2': '0.00', 'status': '6.00'}}, 'timestamp': '2016-09-05T17:53:22Z'}`

imbeacon, 2023-02-20:
Hi @a-jiang ,

It is a statistic data, you should see there different messages, for devices. This message is just a gateway statistic data.

a-jiang, 2023-02-22:
> Hi @a-jiang ,
> 
> It is a statistic data, you should see there different messages, for devices. This message is just a gateway statistic data.
 thank you for your answer. I got an error ，See the logs below
[image]
below is the output data which from coustom
`{'body': {'serviceId': 'analog', 'nodeId': 'kd_huawei_pvi1', 'serviceProperties': {'PV_DC': '60.44', 'PV_TC': '18048.98', 'ERROR_CODE': '0.00', 'PV_ST': '1676621429.00', 'PV_T': '34.00', 'PV_EF': '98.24', 'PV_FQ': '50.03', 'PV_PF': '1.00', 'PV_OQP': '-0.02', 'PV_OAP': '6.34', 'PV_HP': '24.62', 'PV_UIC': '9.04', 'PV_UIB': '9.06', 'PV_UIA': '9.05', 'PV_UVC': '232.40', 'PV_UVB': '234.00', 'PV_UVA': '232.50', 'PV_CA': '401.10', 'PV_BC': '402.60', 'PV_AB': '402.10', 'PV_IP': '6.45', 'PV4_A': '2.78', 'PV4_V': '577.20', 'PV3_A': '2.74', 'PV3_V': '574.60', 'PV2_A': '2.84', 'PV2_V': '577.10', 'PV1_A': '2.78', 'PV1_V': '581.30', 'warn2': '0.00', 'warn1': '0.00', 'warn': '0.00', 'staus3': '7.00', 'staus2': '0.00', 'status': '6.00'}}, 'timestamp': '2016-09-05T17:53:22Z'}`

**how can i solve this error,please help me,thank you very much.**

imbeacon, 2024-12-11:
Hi,

Timestamps supports only unix timestamp, not a date string.
Ability to use date string as timestamp will be added after implementing functionality, described in https://github.com/thingsboard/thingsboard-gateway/issues/1256

closed 2024-12-11 by imbeacon