thingsboard/thingsboard-gateway #1063: Can IoT gateway group devices before TB receives data?

Jonnik84, 2023-02-24:
Hi there, please help me to understand.
 I have multiple of sensors on my machine and I want this machine to appear in TB as single device, but with many parameters pls see pic.  
 is it possible gateway group sensors?

[image]

imbeacon, 2023-02-24:
Hi @Jonnik84 ,

It depends on which connector you want to use and how the connector configured. Theoretically you can just put the same name  for all sensors in connect configuration and the data will be stored for the device with provided name.

eeibhn, 2023-03-01:
Same question for OPC does the device get created only when data has been received. I have the following log but no device creation

Mar  1 09:51:58 tb-iot-gw python3[591006]: [STREAM ONLY] 2023-03-01 09:51:58,512 - INFO - [tb_gateway_service.py] - tb_gateway_service - 415 - Remote logging has being updated. Current logging level is: DEBUG
Mar  1 09:51:58 tb-iot-gw python3[591006]: [STREAM ONLY] 2023-03-01 09:51:58,513 - DEBUG - [tb_gateway_service.py] - tb_gateway_service - 391 - Shared attributes received (RemoteLoggingLevel).
Mar  1 09:51:58 tb-iot-gw python3[591006]: [STREAM ONLY] 2023-03-01 09:51:58,520 - DEBUG - [opcua_connector.py] - opcua_connector - 125 - 3
Mar  1 09:51:58 tb-iot-gw python3[591006]: [STREAM ONLY] 2023-03-01 09:51:58,551 - INFO - [tb_updater.py] - tb_updater - 84 - v3.2
Mar  1 09:51:58 tb-iot-gw python3[591006]: [STREAM ONLY] 2023-03-01 09:51:58,552 - INFO - [tb_updater.py] - tb_updater - 86 -
Mar  1 09:51:58 tb-iot-gw python3[591006]: [===UPDATE===]
Mar  1 09:51:58 tb-iot-gw python3[591006]:  New version v3.2 is available!
Mar  1 09:51:58 tb-iot-gw python3[591006]: [===UPDATE===]
Mar  1 09:51:58 tb-iot-gw python3[591006]: [STREAM ONLY] 2023-03-01 09:51:58,591 - DEBUG - [opcua_connector.py] - opcua_connector - 607 - SHOW MAP: Current node path: Root\.Objects\.Server
Mar  1 09:51:58 tb-iot-gw python3[591006]: [STREAM ONLY] 2023-03-01 09:51:58,604 - DEBUG - [opcua_connector.py] - opcua_connector - 607 - SHOW MAP: Current node path: Topology
Mar  1 09:51:58 tb-iot-gw python3[591006]: [STREAM ONLY] 2023-03-01 09:51:58,605 - DEBUG - [opcua_connector.py] - opcua_connector - 622 - SHOW MAP: Current node path: Topology - NODE FOUND
Mar  1 09:51:58 tb-iot-gw python3[591006]: [STREAM ONLY] 2023-03-01 09:51:58,605 - DEBUG - [opcua_connector.py] - opcua_connector - 625 - SHOW MAP: Search in Topology
Mar  1 09:51:58 tb-iot-gw python3[591006]: [STREAM ONLY] 2023-03-01 09:51:58,636 - DEBUG - [opcua_connector.py] - opcua_connector - 607 - SHOW MAP: Current node path: Topology\.Test gruppering
Mar  1 09:51:58 tb-iot-gw python3[591006]: [STREAM ONLY] 2023-03-01 09:51:58,636 - DEBUG - [opcua_connector.py] - opcua_connector - 622 - SHOW MAP: Current node path: Topology\.Test gruppering - NODE FOUND
Mar  1 09:51:58 tb-iot-gw python3[591006]: [STREAM ONLY] 2023-03-01 09:51:58,637 - DEBUG - [opcua_connector.py] - opcua_connector - 625 - SHOW MAP: Search in Topology\.Test gruppering
Mar  1 09:51:58 tb-iot-gw python3[591006]: [STREAM ONLY] 2023-03-01 09:51:58,675 - DEBUG - [opcua_connector.py] - opcua_connector - 607 - SHOW MAP: Current node path: Topology\.Test gruppering\.NET0001
Mar  1 09:51:58 tb-iot-gw python3[591006]: [STREAM ONLY] 2023-03-01 09:51:58,695 - DEBUG - [opcua_connector.py] - opcua_connector - 607 - SHOW MAP: Current node path: Topology\.AlarmInfo
Mar  1 09:51:58 tb-iot-gw python3[591006]: [STREAM ONLY] 2023-03-01 09:51:58,702 - DEBUG - [opcua_connector.py] - opcua_connector - 607 - SHOW MAP: Current node path: Topology\.Test gruppering
Mar  1 09:51:58 tb-iot-gw python3[591006]: [STREAM ONLY] 2023-03-01 09:51:58,703 - DEBUG - [opcua_connector.py] - opcua_connector - 622 - SHOW MAP: Current node path: Topology\.Test gruppering - NODE FOUND
Mar  1 09:51:58 tb-iot-gw python3[591006]: [STREAM ONLY] 2023-03-01 09:51:58,703 - DEBUG - [opcua_connector.py] - opcua_connector - 625 - SHOW MAP: Search in Topology\.Test gruppering
Mar  1 09:51:58 tb-iot-gw python3[591006]: [STREAM ONLY] 2023-03-01 09:51:58,716 - DEBUG - [opcua_connector.py] - opcua_connector - 607 - SHOW MAP: Current node path: Topology\.Test gruppering\.NET0001
Mar  1 09:51:58 tb-iot-gw python3[591006]: [STREAM ONLY] 2023-03-01 09:51:58,716 - DEBUG - [opcua_connector.py] - opcua_connector - 375 - [{'Root\\.Objects\\.Topology\\.Test gruppering\\.Digitalt\\.DV - DV\\.NET0001.DUC001.DV01': {'deviceNodePattern': 'Root\\.Objects\\.Topology\\.Test gruppering\\.Digitalt\\.DV - DV\\.NET0001.DUC001.DV01', 'deviceNamePattern': 'Device ${Root\\.Objects\\.Topology\\.Test gruppering\\.Digitalt\\.DV - DV\\.NET0001.DUC001.DV01.Value}', 'attributes': [{'key': '', 'path': '${ns=2;s=NET0001.DUC001.DV01.Value}'}], 'timeseries': [{'key': '', 'path': '${ns=2;s=NET0001.DUC001.DV01.Value}'}], 'rpc_methods': [{'key': '', 'path': '${ns=2;s=NET0001.DUC001.DV01.Value}'}], 'attributes_updates': [{'key': '', 'path': '${ns=2;s=NET0001.DUC001.DV01.Value}'}]}}]
Mar  1 09:51:58 tb-iot-gw python3[591006]: [STREAM ONLY] 2023-03-01 09:51:58,717 - DEBUG - [opcua_connector.py] - opcua_connector - 347 - Subscriptions: {}
Mar  1 09:51:58 tb-iot-gw python3[591006]: [STREAM ONLY] 2023-03-01 09:51:58,717 - DEBUG - [opcua_connector.py] - opcua_connector - 348 - Available methods: {}
Mar  1 09:51:58 tb-iot-gw python3[591006]: [STREAM ONLY] 2023-03-01 09:51:58,722 - INFO - [opcua_connector.py] - opcua_connector - 133 - OPC-UA connector OPC-UA Connector connected to server opc.tcp://212.247.187.118:7000
Mar  1 09:52:00 tb-iot-gw python3[591006]: [STREAM ONLY] 2023-03-01 09:52:00,874 - INFO - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 185 - Subscribed to *|* with id 2 for device *
Mar  1 09:52:03 tb-iot-gw python3[591006]: [STREAM ONLY] 2023-03-01 09:52:03,879 - INFO - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 185 - Subscribed to *|* with id 3 for device *
Mar  1 09:52:06 tb-iot-gw python3[591006]: [STREAM ONLY] 2023-03-01 09:52:06,884 - INFO - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 185 - Subscribed to *|* with id 4 for device *
[... cut ...]
Mar  1 09:52:48 tb-iot-gw python3[591006]: [STREAM ONLY] 2023-03-01 09:52:48,750 - INFO - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 185 - Subscribed to *|* with id 18 for device *
Mar  1 09:52:51 tb-iot-gw python3[591006]: [STREAM ONLY] 2023-03-01 09:52:51,754 - INFO - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 185 - Subscribed to *|* with id 19 for device *
Mar  1 09:52:54 tb-iot-gw python3[591006]: [STREAM ONLY] 2023-03-01 09:52:54,759 - INFO - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 185 - Subscribed to *|* with id 20 for device *

closed 2024-06-20 by imbeacon