thingsboard/thingsboard-gateway #1066: [HELP] Error occur in "systemctl status" and upgrade issue

samChanFarm66, 2023-03-06:
*issue**
Hi everyone, I am using a diy mqtt topic to update device data to thingsboard and it gets some issues
It works fine, each latest telemetry can been seen.

1.However, error will occur for each update

- |ERROR| - [mqtt_connector.py] - mqtt_connector - on_attributes_update - 695 - Cannot find deviceName by filter in message with data: {'device': '0x0a02', 'data': {'element_1': False}}

It seems that the element  [ 'deviceName': '0x0a02']  changed to ['device': '0x0a02'] in my case can lead to the error.
Is there any method the fix this?

2. Also, when I check  "systemctl status", tb_updater reminds me there is a v3.2 version update.
I tried "sudo pip3 install thingsboard-gateway --upgrade", but the  tb_updater remind still exists.
Am I using a wrong update method?

Thank you

**Connector **
 MQTT Connector - I add a new element into mqtt.json
{
      "topicFilter": "gateway/test/+",
      "converter": {
        "type": "json",
        "deviceNameTopicExpression": "(?<=test\/)(.*)",
        "deviceTypeTopicExpression": "device",
        "timeout": 60000,
        "attributes": [
          {
            "type": "string",
            "key": "last_seen",
            "value": "${last_seen}"
          }
        ],
        "timeseries": [
          {
            "type": "boolean",
            "key": "element_1",
            "value": "${element_1}"
          }
        ]
      }
    },

**Error traceback *
 - |ERROR| - [mqtt_connector.py] - mqtt_connector - on_attributes_update - 695 - Cannot find deviceName by filter in message with data: {'device': '0x0a02', 'data': {'element_1': False}}

**Versions**
 - OS: raspberry pi os (raspbian)
- thingsboard gateway version: I install it yesterday with DEB, should be latest one
 - Python version[ 3.9.2]

samson0v, 2023-03-06:
Hi @samChanFarm66, thanks for your interest in ThingsBoard IoT Gateway!
Send, please, the entire MQTT connector config file with attributesUpdate section.

samChanFarm66, 2023-03-06:
> Hi @samChanFarm66, thanks for your interest in ThingsBoard IoT Gateway! Send, please, the entire MQTT connector config file with attributesUpdate section.

hi @samson0v , this the full content of the mqtt.json

{
  "broker": {
    "name": "Default Local Broker",
    "host": "127.0.0.1",
    "port": 1883,
    "clientId": "ThingsBoard_gateway",
    "version": 5,
    "maxMessageNumberPerWorker": 15,
    "maxNumberOfWorkers": 100,
    "security": {
      "type": "anonymous"
    }
  },
  "mapping": [
    {
      "topicFilter": "gateway/test/+",
      "converter": {
        "type": "json",
        "deviceNameTopicExpression": "(?<=test\/)(.*)",
        "deviceTypeTopicExpression": "device",
        "timeout": 60000,
        "attributes": [
          {
            "type": "string",
            "key": "last_seen",
            "value": "${last_seen}"
          }
        ],
        "timeseries": [
          {
            "type": "boolean",
            "key": "data_1",
            "value": "${data_1}"
          },
          {
            "type": "double",
            "key": "data_2",
            "value": "${data_2}"
          },
          {
            "type": "double",
            "key": "data_3",
            "value": "${data_3}"
          }
        ]
      }
    },
    {
      "topicFilter": "/sensor/data",
      "converter": {
        "type": "json",
        "deviceNameJsonExpression": "${serialNumber}",
        "deviceTypeJsonExpression": "${sensorType}",
        "timeout": 60000,
        "attributes": [
          {
            "type": "string",
            "key": "model",
            "value": "${sensorModel}"
          },
          {
            "type": "string",
            "key": "${sensorModel}",
            "value": "on"
          }
        ],
        "timeseries": [
          {
            "type": "double",
            "key": "temperature",
            "value": "${temp}"
          },
          {
            "type": "double",
            "key": "humidity",
            "value": "${hum}"
          },
          {
            "type": "string",
            "key": "combine",
            "value": "${hum}:${temp}"
          }
        ]
      }
    },
    {
      "topicFilter": "/sensor/+/data",
      "converter": {
        "type": "json",
        "deviceNameTopicExpression": "(?<=sensor\/)(.*?)(?=\/data)",
        "deviceTypeTopicExpression": "Thermometer",
        "timeout": 60000,
        "attributes": [
          {
            "type": "string",
            "key": "model",
            "value": "${sensorModel}"
          }
        ],
        "timeseries": [
          {
            "type": "double",
            "key": "temperature",
            "value": "${temp}"
          },
          {
            "type": "double",
            "key": "humidity",
            "value": "${hum}"
          }
        ]
      }
    },
    {
      "topicFilter": "/sensor/raw_data",
      "converter": {
        "type": "bytes",
        "deviceNameExpression": "[0:4]",
        "deviceTypeExpression": "default",
        "timeout": 60000,
        "attributes": [
          {
            "type": "raw",
            "key": "rawData",
            "value": "[:]"
          }
        ],
        "timeseries": [
          {
            "type": "raw",
            "key": "temp",
            "value": "[4:]"
          }
        ]
      }
    },
    {
      "topicFilter": "/custom/sensors/+",
      "converter": {
        "type": "custom",
        "extension": "CustomMqttUplinkConverter",
        "extension-config": {
          "temperatureBytes": 2,
          "humidityBytes": 2,
          "batteryLevelBytes": 1
        }
      }
    }
  ],
  "connectRequests": [
    {
      "topicFilter": "sensor/connect",
      "deviceNameJsonExpression": "${SerialNumber}"
    },
    {
      "topicFilter": "sensor/+/connect",
      "deviceNameTopicExpression": "(?<=sensor\/)(.*?)(?=\/connect)"
    }
  ],
  "disconnectRequests": [
    {
      "topicFilter": "sensor/disconnect",
      "deviceNameJsonExpression": "${SerialNumber}"
    },
    {
      "topicFilter": "sensor/+/disconnect",
      "deviceNameTopicExpression": "(?<=sensor\/)(.*?)(?=\/disconnect)"
    }
  ],
  "attributeRequests": [
    {
      "retain": false,
      "topicFilter": "v1/devices/me/attributes/request",
      "deviceNameJsonExpression": "${serialNumber}",
      "attributeNameJsonExpression": "${versionAttribute}, ${pduAttribute}",
      "topicExpression": "devices/${deviceName}/attrs",
      "valueExpression": "${attributeKey}: ${attributeValue}"
    }
  ],
  "attributeUpdates": [
    {
      "retain": true,
      "deviceNameFilter": "SmartMeter.*",
      "attributeFilter": "uploadFrequency",
      "topicExpression": "sensor/${deviceName}/${attributeKey}",
      "valueExpression": "{\"${attributeKey}\":\"${attributeValue}\"}"
    }
  ],
  "serverSideRpc": [
    {
      "deviceNameFilter": ".*",
      "methodFilter": "echo",
      "requestTopicExpression": "sensor/${deviceName}/request/${methodName}/${requestId}",
      "responseTopicExpression": "sensor/${deviceName}/response/${methodName}/${requestId}",
      "responseTimeout": 10000,
      "valueExpression": "${params}"
    },
    {
      "deviceNameFilter": ".*",
      "methodFilter": "no-reply",
      "requestTopicExpression": "sensor/${deviceName}/request/${methodName}/${requestId}",
      "valueExpression": "${params}"
    }
  ]
}

samson0v, 2023-03-06:
@samChanFarm66, I can't see that you configured the attributesUpdate section:
[image]
So, from the config above, Gateway will allow accepting every attribute request whose device name is SmartMeter, but in your case it is different.
Please, correct `deviceNameFilter` and, also, make sure that `attributeFilter` is correct.

samChanFarm66, 2023-03-08:
> @samChanFarm66, I can't see that you configured the attributesUpdate section: [image] So, from the config above, Gateway will allow accepting every attribute request whose device name is SmartMeter, [...]

Thank you very much, it works after editting.

closed 2023-03-08 by samson0v