thingsboard/thingsboard-gateway #1073: How to only update  "deviceName" in modbus connector not insert a new one

11208, 2023-03-11:
**Describe the issue**
When ever I tried to change the  "deviceName" in modbus.json then restart gateway, it add new device to thingsboard instead of update the name.

**Configuration** 
```
thingsboard:
  host: XX.XXX.XXX.XXX
  port: 1883
  remoteShell: false
  remoteConfiguration: false
  statistics:
    enable: true
    statsSendPeriodInSeconds: 3600
  maxPayloadSizeBytes: 1024
  minPackSendDelayMS: 0
  checkConnectorsConfigurationInSeconds: 60
  handleDeviceRenaming: true
  checkingDeviceActivity:
    checkDeviceInactivity: false
    inactivityTimeoutSeconds: 120
    inactivityCheckPeriodSeconds: 10
  security:
    accessToken: XXX
  qos: 1
storage:
  type: memory
  read_records_count: 100
  max_records_count: 100000
grpc:
  enabled: false
  serverPort: 9595
  keepaliveTimeMs: 10000
  keepaliveTimeoutMs: 5000
  keepalivePermitWithoutCalls: true
  maxPingsWithoutData: 0
  minTimeBetweenPingsMs: 10000
  minPingIntervalWithoutDataMs: 5000
connectors:
  -
    name: Modbus Connector for SHT20
    type: modbus
    configuration: sht20.json
    ```

**Connector name (If you need help with some connector/converter):**
Modbus

**Versions (please complete the following information):**
 - OS: Ubuntu 22.04
 - Thingsboard IoT Gateway version v3.2
 - Python version 3.9

samson0v, 2023-03-11:
Hi @11208, thanks for your interest in ThingsBoard IoT Gateway!
It is a normal situation for Gateway, you can change the device name only on the ThingsBoard side. 
For this, you have to configure the following parameter (`handleDeviceRenaming`):
[image]

11208, 2023-03-12:
> Hi @11208, thanks for your interest in ThingsBoard IoT Gateway! It is a normal situation for Gateway, you can change the device name only on the ThingsBoard side. For this, you have to configure the following parameter (`handleDeviceRenaming`): [image]

After setting handleDeviceRenaming: true
The device name in gateway not change when changing name from ThingsBoard Side.
[image]
[image]
[image]

samson0v, 2023-03-21:
Hi @11208,
Tested `handleDeviceRenaming` again and it works fine:
1. Created a new device and send some data:
    [image]
2. Renamed it on ThingsBoard
3. Get a renaming notification
    [image]
4. Send data to the renamed device
    [image]
5. Data updated on ThingsBoard
    [image]

11208, 2023-03-21:
> Hi @11208, Tested `handleDeviceRenaming` again and it works fine:
> 
> 1. Created a new device and send some data:
>    [image]
> 2. Renamed it on ThingsBoard
> 3. Get a renaming notification
>    [image]
> 4. Send data to the renamed device
>    [image]
> 5. Data updated on ThingsBoard
>    [image]

Thanks you. You can close the issue.

closed 2023-03-21 by samson0v