thingsboard/thingsboard-gateway #1074: [HELP] Thingsboard Gateway MQTT connector is not reconnecting until the Gateway container is restarted

akseerali, 2023-03-13:
**Describe the issue**
I am sending the data from MQTT broker to TB Edge via TB Gateway. At gateway, I am using MQTT 8883 TLS based connection to join the broker. The gateway is working fine and I am also able to receive the data at Edge Devices; however, when I am restarting the remote MQTT broker, the gateway is throwing an error and it can no longer subscribe as a MQTT client until the container is restarted. I have also tried to use Thingsboard Gateway as a DEB package on Ubuntu, but it has the same issue.

Below are the attached configurations for Thingsboard Gateway, MQTT connector, and Docker Compose.
**tb_gateway.yaml configuration:**
``` 
thingsboard:
  host: 10.5.0.11
  port: 1883
  remoteShell: false
  remoteConfiguration: false
  statistics:
    enable: true
    statsSendPeriodInSeconds: 3600
  maxPayloadSizeBytes: 1024
  minPackSendDelayMS: 100
  checkConnectorsConfigurationInSeconds: 60
  handleDeviceRenaming: false
  checkingDeviceActivity:
    checkDeviceInactivity: true
    inactivityTimeoutSeconds: 120
    inactivityCheckPeriodSeconds: 10
  security:
    accessToken: #access token
  qos: 1
storage:
#  type: memory
#  read_records_count: 100
#  max_records_count: 100000
#  type: file  
#  data_folder_path: /home/iiotedge/.tb-gateway/data
#  max_file_count: 10
#  max_read_records_count: 10
#  max_records_per_file: 1000
  type: sqlite
  data_file_path: /home/iiotedge/.tb-gateway/data/data.db
  messages_ttl_check_in_hours: 1
  messages_ttl_in_days: 7
grpc:
  enabled: false
  serverPort: 9595
  keepaliveTimeMs: 10000
  keepaliveTimeoutMs: 5000
  keepalivePermitWithoutCalls: true
  maxPingsWithoutData: 0
  minTimeBetweenPingsMs: 10000
  minPingIntervalWithoutDataMs: 5000
connectors:
   - name: mqtt connector
     type: mqtt
     configuration: mqtt-connector.json 
```

**mqtt connector configuration: **
``` 
{
   "broker":{
      "name":"MQTT broker",
      "host":"192.168.1.100",
      "port":8883,
      "clientId":"mqttconnector",
      "maxMessageNumberPerWorker":10,
      "maxNumberOfWorkers":100,
      "security":{
         "type":"tls",
         "username":"mqttconnector",
         "password":"**mqttconnector**"
      }
   },
   "mapping":[
[... 143 lines cut ...]
tb-gw  |   File "/root/.local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3591, in _thread_main
tb-gw  |     self.loop_forever(retry_first_connection=True)
tb-gw  |   File "/root/.local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1756, in loop_forever
tb-gw  |     rc = self._loop(timeout)
tb-gw  |   File "/root/.local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1164, in _loop
tb-gw  |     rc = self.loop_read()
tb-gw  |   File "/root/.local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1556, in loop_read
tb-gw  |     rc = self._packet_read()
tb-gw  |   File "/root/.local/lib/python3.10/site-packages/paho/mqtt/client.py", line 2439, in _packet_read
tb-gw  |     rc = self._packet_handle()
tb-gw  |   File "/root/.local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3039, in _packet_handle
tb-gw  |     return self._handle_connack()
tb-gw  |   File "/root/.local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3135, in _handle_connack
tb-gw  |     on_connect(self, self._userdata,
tb-gw  |   File "/thingsboard_gateway/connectors/mqtt/mqtt_connector.py", line 389, in _on_connect
tb-gw  |     if result_code in result_codes:
tb-gw  | TypeError: unhashable type: 'ReasonCodes'
``` 

**Versions (please complete the following information):**
 - OS: [Ubuntu 22.04]
 - Thingsboard IoT Gateway version [latest]
 - Thingsboard Edge PE version [3.4.3EDGEPE]
 - Docker Engine version [23.0.1]
 - Docker Compose version [v2.16.0]

samson0v, 2023-03-21:
Hi @akseerali, thanks for your interest in ThingsBoard IoT Gateway!
We are investigating your problem.

mdameda, 2023-09-04:
Hello @samson0v we are experiencing the same error. 
We are using a thingsboard gateway with a MQTT connection to a remote broken, and when the broker disconnect we see same logs as previous post:

` if result_code in result_codes:
tb-gw  | TypeError: unhashable type: 'ReasonCodes'`

The solution is to restart the IoT Gateway docker container to restablish the connection. 

Any news ?

cos-gl, 2024-07-29:
I have the same problem for the OPCUA connector and we use the same solution (which is inefficient) .
I'm using commit b684e46cc23469865ed2b45c984cbd67455d24b9 (12 July 2024)
Do you have any news about this?

imbeacon, 2024-11-14:
Hi,

This issue relates to the SDK version. To solve the issue, we recommend to use the latest version of the gateway, before updating we recommend to create a backup of your configuration files.

closed 2024-11-14 by imbeacon