thingsboard/thingsboard-gateway #1075: IoT Gateway should not automatically install dependencies

pbrkr, 2023-03-14:
Automatic installation of dependencies at runtime is not compatible with many of the places where we want to use the IoT Gateway. Embedded systems may have flash space pre-allocated at build time (so there is no spare space for package installation at runtime), may use a read-only rootfs filesystem or may not have `pip` available. We also want to use the IoT Gateway in environments with limited or no access to the internet (where it's talking to an on-site instance of Thingsboard). Furthermore, installation of additional packages at runtime breaks the reproducibility of our system and causes our tracking of CVEs and licenses to be incomplete.

I recommend expressing dependencies in setup.py where possible. If you don't want to put dependencies for all connectors into the top-level setup.py file, it may be better to break out each connector into its own module with proper dependencies.

At the least, please make this configurable at runtime so we can disable it in our environments.

Also see https://github.com/thingsboard/thingsboard-python-client-sdk/issues/39 for similar concerns.

samson0v, 2023-03-21:
Hi @pbrkr, thanks for your interest in ThingsBoard IoT Gateway!
We decided to add a new parameter to the configuration, that will block any package installation at runtime.

samson0v, 2023-05-04:
Hi @pbrkr, for now, we are busy with higher-priority tasks. So, we will be grateful if you are prepared PR for this feature.
So, the design is the following:
- Add a new parameter to the main config tb_gateway.yaml (in the earest feature tb_gateway.json) - `dynRequirements` (by default True);
- Provide it to the implemented connectors and services.

closed 2024-01-26 by imbeacon