thingsboard/thingsboard-gateway #1077: [HELP] How to add ocpp charger using thingsboard gateway

Anmirazik, 2023-03-18:
**Describe the issue**
How to add ocpp charging station using thingsboard iot gateway  ?

```
thingsboard:
  host: 10.0.X.X
  port: 2000
  remoteShell: false
  remoteConfiguration: false
  statistics:
    enable: true
    statsSendPeriodInSeconds: 3600
    configuration: statistics.json
  maxPayloadSizeBytes: 1024
  minPackSendDelayMS: 0
  checkConnectorsConfigurationInSeconds: 60
  handleDeviceRenaming: true
  checkingDeviceActivity:
    checkDeviceInactivity: false
    inactivityTimeoutSeconds: 120
    inactivityCheckPeriodSeconds: 10
  security:
    accessToken:XXXXXXXXXXXXXXX
  qos: 1
storage:
  type: memory
  read_records_count: 100
  max_records_count: 100000
grpc:
  enabled: false
  serverPort: 9595
  keepaliveTimeMs: 10000
  keepaliveTimeoutMs: 5000
  keepalivePermitWithoutCalls: true
  maxPingsWithoutData: 0
  minTimeBetweenPingsMs: 10000
  minPingIntervalWithoutDataMs: 5000
connectors:
  -
    name: MQTT Broker Connector
    type: mqtt
    configuration: mqtt.json

  -
   name: OCPP Connector
   type: ocpp
   configuration: ocpp.json
```

**OCPP.JSON File**

```
{
  "centralSystem": {
    "name": "Central System",
    "host": "10.0.X.X",
    "port": 9000,
    "connection": {
      "type": "insecure"
    },
    "security": []
  },
  "chargePoints": [
    {
      "idRegexpPattern": "bidon/hello/CP_1",
      "deviceNameExpression": "${Vendor} ${Model}",
      "deviceTypeExpression": "default",
      "attributes": [
        {
          "messageTypeFilter": "MeterValues,",
          "key": "temp1",
          "value": "${meter_value[:].sampled_value[:].value}"
        },
        {
          "messageTypeFilter": "MeterValues,",
          "key": "vendorId",
          "value": "${connector_id}"
        }
      ],
      "timeseries": [
        {
          "messageTypeFilter": "DataTransfer,",
          "key": "temp",
          "value": "${data.temp}"
        }
      ],
      "attributeUpdates": [
        {
          "attributeOnThingsBoard": "shared",
          "valueExpression": "{\"${attributeKey}\":\"${attributeValue}\"}"
        }
      ],
      "serverSideRpc": [
        {
          "methodRPC": "rpc1",
          "withResponse": true,
          "valueExpression": "${params}"
        }
      ]
    }
  ]
}

```

Note : I changed the thingsboard mqtt port from 1883 to 2000 due to I have installed mqtt broker (mosquito) which run on port 1883 on my server 

back to my question/issue :-

I Believe the sections of **"idRegexpPattern": "bidon/hello/CP_1"** is the path and id right ?

hence the full websocket URL looks like this 

ws://10.0.X.X:9000/bidon/hello/CP_1 

where the bidon/hello is the path and CP_1 is the charger ID 

am I right ? did I do it wrong ? 

it seems like the charger wont connect to thingsboard ocpp connector . I'm using ocpp 1.6 charging station simulator btw from here https://github.com/victormunoz/OCPP-1.6-Chargebox-Simulator

**Connector name (If you need help with some connector/converter):**
[OCPP Connector]

**Versions (please complete the following information):**
 - OS: Ubuntu 20.04 , Im using docker images of thingboard gateway on ubuntu 20.04
 - Thingsboard IoT Gateway image version latest

samson0v, 2023-03-21:
Hi @Anmirazik, thanks for your interest in ThingsBoard IoT Gateway!

I downloaded the fork(https://github.com/victormunoz/OCPP-1.6-Chargebox-Simulator) from the original repo and it seems there is a bug (CP reconnect all time).
But, I also, downloaded the original repo (https://github.com/kubarskii/OCPP-J-CP-Simulator) and it works fine:
[image]
[image]

closed 2023-03-22 by Anmirazik

Vaghulkumar95, 2024-02-05:
Hi @samson0v,

I have this simulator running in a windows machine and in the same network I have a pi connected(thingsboard running here) and I'm trying to talk to the simulator? What would be the hostname in ocpp.json and What would be the endpoint URL I should be giving in the simulator? And in the simulator should I fill the other details like chargepoint Vendor, chargepoint model etc?