thingsboard/thingsboard-gateway #1079: [HELP] Error while handling ocpp request (Status Notification) & How to extract charging consumption data ?

Anmirazik, 2023-03-22:
**Describe the issue**
Error while handling message request for status notification from real life charger , is there any changes that i need to made in the configuration file to accept status notification request ? How do i extract charging consumption data from the charger ? do i need to modified the ocpp.json , and if yes how should i modified the ocpp.json file ? 

**Configuration** (ocpp.json) 

```
{
  "centralSystem": {
    "name": "Central System",
    "host": "0.0.0.0",
    "port": 9000,
    "connection": {
      "type": "insecure"
    },
    "security": []
  },
  "chargePoints": [
    {
      "idRegexpPattern": "bidon/hello/EXICOM_SP",
      "deviceNameExpression": "${Vendor} ${Model}",
      "deviceTypeExpression": "default",
      "attributes": [
        {
          "messageTypeFilter": "MeterValues,",
          "key": "temp1",
          "value": "${meter_value[:].sampled_value[:].value}"
        },
        {
          "messageTypeFilter": "MeterValues,",
          "key": "vendorId",
          "value": "${connector_id}"
        }
      ],
      "timeseries": [
        {
          "messageTypeFilter": "DataTransfer,",
          "key": "temp",
          "value": "${data.temp}"
        }
      ],
      "attributeUpdates": [
        {
          "attributeOnThingsBoard": "shared",
          "valueExpression": "{\"${attributeKey}\":\"${attributeValue}\"}"
        }
      ],
      "serverSideRpc": [
        {
          "methodRPC": "rpc1",
          "withResponse": true,
          "valueExpression": "${params}"
        }
      ]
    }
  ]
}
```

**Connector name (If you need help with some connector/converter):**
[OCPP Connector]

**Error :**
```
""2023-03-22 09:03:36" - |INFO| - [ocpp_connector.py] - ocpp_connector - on_connect - 153 - Protocols Matched: ocpp1.6"
""2023-03-22 09:03:36" - |INFO| - [ocpp_connector.py] - ocpp_connector - on_connect - 181 - Connected Charge Point with id: bidon/hello/EXICOM_SP"
""2023-03-22 09:03:37" - |ERROR| - [charge_point.py] - charge_point - route_message - 155 - Error while handling request '<Call - unique_id=8230, action=StatusNotification, payload={'connectorId': 1, 'errorCode': 'NoError', 'info': 'SIM_Absent', 'status': 'Available', 'timestamp': '2023-03-22T09:03:35Z'}>'"
Traceback (most recent call last):
  File "/root/.local/lib/python3.10/site-packages/ocpp/charge_point.py", line 174, in _handle_call
    handlers = self.route_map[msg.action]
KeyError: 'StatusNotification'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/root/.local/lib/python3.10/site-packages/ocpp/charge_point.py", line 153, in route_message
    await self._handle_call(msg)
  File "/root/.local/lib/python3.10/site-packages/ocpp/charge_point.py", line 176, in _handle_call
    raise NotSupportedError(
ocpp.exceptions.NotSupportedError: NotSupportedError: Request Action is recognized but not supported by the receiver, {'cause': 'No handler for StatusNotification registered.'}
""2023-03-22 09:03:37" - |ERROR| - [charge_point.py] - charge_point - route_message - 155 - Error while handling request '<Call - unique_id=15718, action=StatusNotification, payload={'connectorId': 1, 'errorCode': 'NoError', 'info': 'WiFiConnected', 'status': 'Available', 'timestamp': '2023-03-22T09:03:35Z'}>'"
Traceback (most recent call last):
  File "/root/.local/lib/python3.10/site-packages/ocpp/charge_point.py", line 174, in _handle_call
    handlers = self.route_map[msg.action]
KeyError: 'StatusNotification'

```

**Versions (please complete the following information):**
 - OS: [Ubuntu 20.04]
 - Local Docker Image thingsboard gateway based on github repo

Anmirazik, 2023-03-22:
i read from https://github.com/thingsboard/thingsboard-gateway/issues/1059 , current available ocpp features are BootNotification, Heartbeat, MeterValues and DataTransfer . I believed from the MeterValues we can extract the charging consumption data right if I'm not mistaken . How can i get that data , is there any configuration changes that i must made in the ocpp.json file ? Thank you

julianck2011, 2024-02-07:
I would like to confirm whether the ocpp features like BootNotification, Heartbeat, MeterValues and DataTransfer etc already implemented or waiting for development since we find message like "IOT gateway is not available to handler the OCPP call request". Thanks.

closed 2026-09-10 by samson0v