thingsboard/thingsboard-gateway #1091: [FEATURE] Connector RPC

devaskim, 2023-04-02:
**Describe the issue**
This feature request is a part of more bigger feature - **configuration management of custom converters**.

Our business case is dictionary based mapping of MQTT incoming data. Dictionaries are dynamically changed and provided by MQTT broker as well, then converter persists them to local files. So to map incoming timeseries we need to [share dictionaries between dictionary converter and timeseries converter](https://github.com/thingsboard/thingsboard-gateway/pull/1090).

```
    {
      "topicFilter": "Boilers/+/KPI_meta",
      "converter": {
        "extension": "TimeseriesConverter",
        "deviceNameTopicExpression": "(?<=Boilers\/)(.*?)(?=\/KPI_meta)",
        "deviceTypeTopicExpression": "Boiler",
        "namePrefix": "KPI_",
        "dumpFile": "kpi_meta.json"
      }
    },
    {
      "topicFilter": "Boilers/+/KPI",
      "converter": {
        "extension": "TimeseriesConverter",
        "deviceNameTopicExpression": "(?<=Boilers\/)(.*?)(?=\/KPI)",
        "deviceTypeTopicExpression": "Boiler",
        "namePrefix": "KPI_",
        "dumpFile": "kpi_meta.json"
      }
    }

```

From time to time we faces with timeseries lack because of absent of some dictionary's mapping keys. In such cases we need to `ssh` to gateway's machine to manually check the dump files of dictionaries.

Our idea is to leverage and extend gateway RPC feature to send per connector requests. Then we could poll [cached custom converters ](https://github.com/thingsboard/thingsboard-gateway/pull/1090) to remotely check their configuration or even update it.

I am ready to implement all this stuff. But need some feedback from the core team @imbeacon @samson0v 
Thanks in advance

samson0v, 2023-04-14:
Hi @devaskim, thanks for your interest in ThingsBoard IoT Gateway!
We added a feature that allows change converters config using Shared Attribute Update Request (see PR #1087). Maybe it can solve your problem, so let us know.

devaskim, 2023-04-15:
@samson0v thanks for information, interesting feature but it is about static converter configuration (i.e. described in file or shared attribute).
Our converters use dynamic configuration - receive mapping rules from specific MQTT topics and persist them in specific file (not TB Gateway's mqtt.json)

devaskim, 2023-04-15:
@samson0v Thanks, looks very promising
Let's extend this feature to send existing converter configuration to server.
If you agree, what preferred way will be for this feature: client attribute or RPC response?

samson0v, 2023-04-17:
Hi @devaskim, ok, I will add this functionality.

devaskim, 2023-04-22:
Hi @samson0v any ETA?

closed 2023-04-28 by imbeacon