thingsboard/thingsboard-gateway #1094: [HELP] Using FTP to send log messages to thingsboard

rajeshAnt, 2023-04-13:
I am trying to use FTP to read a local .txt file and send it to the thingsboard. But I keep getting an error, any help to resolve this and implement ftp would be great help :)

**ftp.json**
```
{
    "host": "0.0.0.0",
    "port": 21,
    "TLSSupport": false,
    "security": {
        "type": "basic",
        "username": "admin",
        "password": "admin"
    },
    "paths": [
        {
            "devicePatternName": "asd",
            "devicePatternType": "[0:1]",
            "delimiter": ":",
            "path": "./path/log.txt",
            "readMode": "FULL",
            "maxFileSize": 5,
            "pollPeriod": 60,
            "txtFileDataView": "SLICED",
            "withSortingFiles": true,
            "attributes": [
                {
                    "type": "string",
                    "key": "LOG",
                    "value": "[1:]"
                }
            ],
            "timeseries": [
                {
                    "type": "string",
                    "key": "[0:1]",
                    "value": "[1:]"
                }
            ]
        }
    ]
}
```
**log.txt file**
```
LOG: Hi, this is a log message
```
**tb_gateway.yaml**
```thingsboard:
  host: thingsboard.cloud
  port: 1883
  remoteShell: true
  remoteConfiguration: true
  security:
    accessToken: ****************************
storage:
  type: memory
  read_records_count: 100
  max_records_count: 100000
#  type: file
#  data_folder_path: ./data/
#  max_file_count: 10
#  max_read_records_count: 10   
#  max_records_per_file: 10000
connectors:

#  -
#    name: MQTT Broker Connector
#    type: mqtt
#    configuration: mqtt.json

[... 80 lines cut ...]
''2023-04-13 15:27:43' - DEBUG - tb_gateway_service - 391 - Shared attributes received (RemoteLoggingLevel).'
''2023-04-13 15:27:43' - DEBUG - tb_gateway_service - 394 - Client attributes received (allBytesSentToDevices, allBytesSentToTB, allReceivedBytesFromTB, convertedBytesFromDevice, CPU, current_configuration, Memory, receivedBytesFromDevices).'
''2023-04-13 15:27:43' - ERROR - tb_gateway_service - 752 - 'attributes''
''2023-04-13 15:27:53' - ERROR - ftp_connector - 104 - 'NoneType' object has no attribute 'sendall''
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/thingsboard_gateway/connectors/ftp/ftp_connector.py", line 95, in run
    path.find_files(ftp)
  File "/usr/local/lib/python3.8/dist-packages/thingsboard_gateway/connectors/ftp/path.py", line 79, in find_files
    current_dir = ftp.pwd()
  File "/usr/lib/python3.8/ftplib.py", line 646, in pwd
    resp = self.voidcmd('PWD')
  File "/usr/lib/python3.8/ftplib.py", line 281, in voidcmd
    self.putcmd(cmd)
  File "/usr/lib/python3.8/ftplib.py", line 203, in putcmd
    self.putline(line)
  File "/usr/lib/python3.8/ftplib.py", line 198, in putline
    self.sock.sendall(line.encode(self.encoding))
AttributeError: 'NoneType' object has no attribute 'sendall'
''2023-04-13 15:27:53' - ERROR - tb_gateway_service - 752 - 'attributes''
```

**Versions:**
 - OS: Ubuntu [20.04]
 - Thingsboard IoT Gateway version [3.2]
 - Python version [3.8]

samson0v, 2023-04-14:
Hi @rajeshAnt, thanks for your interest in ThingsBoard IoT Gateway!
We will investigate your problem as soon as possible.

samson0v, 2023-04-14:
@rajeshAnt, from your logs, it seems Gateway can't connect to your FTP Server.
Please, make user that your credentials are correct.

rajeshAnt, 2023-04-17:
Hi @samson0v , thanks for your reply. I checked the credentials, it is correct. But the issue is still there. 
This is how my ftp.json file looks, and I have checked the credentials to right

```
{
    "host": "0.0.0.0",
    "port": 21,
    "TLSSupport": false,
    "security": {
        "type": "basic",
        "username": "*********",
        "password": "**********"
    },
    "paths": [
        {
            "devicePatternName": "asd",
            "devicePatternType": "[0:1]",
            "delimiter": ":",
            "path": "./path/log.txt",
            "readMode": "FULL",
            "maxFileSize": 5,
            "pollPeriod": 60,
            "txtFileDataView": "SLICED",
            "withSortingFiles": true,
            "attributes": [
                {
                    "type": "string",
                    "key": "LOG",
                    "value": "[1:]"
                }
            ],
            "timeseries": [
                {
                    "type": "string",
                    "key": "[0:1]",
                    "value": "[1:]"
                }
            ]
        }
    ]
}
```

samson0v, 2023-04-24:
@rajeshAnt, why are you using `"host": "thingsboard.cloud"`? You have to use the host and credentials of your FTP Server!
From the following log you can see that connector can't connect to FTP Server:
`ftp_connector - __connect - 126 - [Errno 110] Connection timed out"`

samson0v, 2023-04-25:
@rajeshAnt, as I can see from your logs, connect connected to your FTP Server.
Change your delimiter from `,` to ` ` (space).

rajeshAnt, 2023-04-26:
@samson0v  thanks for your reply. The issue was with the ftp server and I fixed it by using VSFTPD server for Ubuntu instead of my custom server, now I can access the files but whereas I still get some error which I need help in debugging.

ftp.json
```
{
  "host": "0.0.0.0",
  "port": 21,
  "TLSSupport": false,
  "security": {
      "type": "basic",
      "username": "******************",
      "password": "*******************"
  },
  "paths": [
      {
          "devicePatternName": "****",
          "devicePatternType": "Device",
          "delimiter": ":",
          "path": "./logs/log.txt",
          "readMode": "PARTIAL",
          "maxFileSize": 5,
          "pollPeriod": 60,
          "txtFileDataView": "SLICED",
          "withSortingFiles": true,
          "attributes": [
          ],
          "timeseries": [
              {
                  "type": "string",
                  "key": "[0:1]",
                  "value": "[1:]"
              }
          ]
      }
  ]
}
```

Error log
```

thingsboard-gateway 
[STREAM ONLY] 2023-04-26 12:03:02,324 - WARNING - [tb_gateway_service.py] - tb_gateway_service - 203 - Remote shell is enabled. Please be carefully with this feature.
[STREAM ONLY] 2023-04-26 12:03:02,617 - ERROR - [ftp_uplink_converter.py] - ftp_uplink_converter - 111 - Error in converter, for config: 
{"delimiter": ":", "devicePatternName": "*****", "devicePatternType": "Device", "timeseries": [{"type": "string", "key": "[0:1]", "value": "[1:]"}], "attributes": [], "txt_file_data_view": "SLICED"}
 and message: 

[STREAM ONLY] 2023-04-26 12:03:02,618 - ERROR - [ftp_uplink_converter.py] - ftp_uplink_converter - 112 - list index out of range
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/thingsboard_gateway/connectors/ftp/ftp_uplink_converter.py", line 99, in _convert_slices_view_data
    val = self._get_key_or_value(information['value'], arr)
  File "/usr/local/lib/python3.8/dist-packages/thingsboard_gateway/connectors/ftp/ftp_uplink_converter.py", line 88, in _get_key_or_value
    return arr[int(first_val_index):int(last_val_index)][0]
IndexError: list index out of range
[STREAM ONLY] 2023-04-26 12:03:02,725 - ERROR - [tb_utility.py] - tb_utility - 71 - No telemetry and attributes in data: {"deviceName": "valera 1", "deviceType": "Device", "attributes": [], "telemetry": []}
[STREAM ONLY] 2023-04-26 12:03:02,725 - ERROR - [tb_gateway_service.py] - tb_gateway_service - 670 - Data from FTP Connector connector is invalid.
[STREAM ONLY] 2023-04-26 12:03:06,034 - INFO - [tb_gateway_service.py] - tb_gateway_service - 415 - Remote logging has being updated. Current logging level is: DEBUG 
[STREAM ONLY] 2023-04-26 12:03:06,035 - DEBUG - [tb_gateway_service.py] - tb_gateway_service - 391 - Shared attributes received (RemoteLoggingLevel).
[STREAM ONLY] 2023-04-26 12:03:06,035 - DEBUG - [tb_gateway_service.py] - tb_gateway_service - 394 - Client attributes received (allBytesSentToDevices, allBytesSentToTB, allReceivedBytesFromTB, convertedBytesFromDevice, CPU, current_configuration, key, Memory, message, receivedBytesFromDevices).
[STREAM ONLY] 2023-04-26 12:03:06,131 - ERROR - [tb_gateway_service.py] - tb_gateway_service - 752 - 'attributes'
[STREAM ONLY] 2023-04-26 12:03:07,397 - INFO - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 185 - Subscribed to *|* with id 5 for device *
[STREAM ONLY] 2023-04-26 12:03:09,000 - INFO - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 185 - Subscribed to *|* with id 6 for device *
[STREAM ONLY] 2023-04-26 12:03:10,603 - INFO - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 185 - Subscribed to *|* with id 7 for device *
[STREAM ONLY] 2023-04-26 12:03:12,406 - INFO - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 185 - Subscribed to *|* with id 8 for device *
[STREAM ONLY] 2023-04-26 12:03:14,009 - INFO - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 185 - Subscribed to *|* with id 9 for device *
[STREAM ONLY] 2023-04-26 12:03:15,613 - INFO - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 185 - Subscribed to *|* with id 10 for device *
[STREAM ONLY] 2023-04-26 12:03:17,616 - INFO - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 185 - Subscribed to *|* with id 11 for device *
[STREAM ONLY] 2023-04-26 12:03:19,220 - INFO - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 185 - Subscribed to *|* with id 12 for device *
[STREAM ONLY] 2023-04-26 12:03:20,823 - INFO - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 185 - Subscribed to *|* with id 13 for device *
[STREAM ONLY] 2023-04-26 12:03:22,426 - INFO - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 185 - Subscribed to *|* with id 14 for device *
[STREAM ONLY] 2023-04-26 12:03:24,229 - INFO - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 185 - Subscribed to *|* with id 15 for device *
[STREAM ONLY] 2023-04-26 12:03:24,264 - ERROR - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 66 - Service subscription to topic v1/gateway/attributes/response - failed.
[STREAM ONLY] 2023-04-26 12:03:26,233 - INFO - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 185 - Subscribed to *|* with id 16 for device *
[STREAM ONLY] 2023-04-26 12:03:27,837 - INFO - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 185 - Subscribed to *|* with id 17 for device *
[STREAM ONLY] 2023-04-26 12:03:29,440 - INFO - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 185 - Subscribed to *|* with id 18 for device *
```
The file I am accessing looks like this 
```
ROS_LOG: [DEBUG] - Hi this is a log message
ROS_LOG: [ERROR] - Hi this is a error message
```

samson0v, 2023-04-26:
@rajeshAnt, we will investigate this problem

rajeshAnt, 2023-04-26:
@samson0v, Also when I get this error. remote shell stops responding. 
Remote works perfect if I remove the ftp connector.
[image]

rajeshAnt, 2023-04-27:
> @samson0v thanks for your reply. The issue was with the ftp server and I fixed it by using VSFTPD server for Ubuntu instead of my custom server, now I can access the files but whereas I still get [...]

@samson0v this error seems to occur when processing empty lines in the file, I tried removing all the empty lines but the error still persists.

closed 2023-05-02 by imbeacon