thingsboard/thingsboard-gateway #1102: Data Forwarding [HELP]

Vaghulkumar95, 2023-04-19:
Just wanted to check if you have a sample guide which I can follow to forward the modbus values to a bacnet device(Bacnet data forwarding) using rules? Basically data forwarding from one device to another. Or do you have any suggestions on how to implement the same?

Vaghulkumar95, 2023-04-19:
[image]
[image]
[image]
[image]

I used the change originator to test this and I wanted to know how to update the device after saving the timeseries.

Vaghulkumar95, 2023-05-10:
Hi,

Any updates on the above 2 requests?

Vaghulkumar95, 2023-05-15:
Hi Team,
Any suggestions on the above two requests?

Vaghulkumar95, 2023-06-07:
Hi Team,

Just wanted to follow up.
Any suggesstions on this?

Vaghulkumar95, 2023-06-20:
Hi @samson0v @imbeacon ,

Any updates on the above 2 questions? And do we have any documentation on how to integrate Mbus into thingsboard?

Vaghulkumar95, 2023-06-27:
Hi @samson0v @imbeacon ,

Just wanted to follow up on this. Any suggestions from your end is highly appreciated.

samson0v, 2023-07-06:
Hi @Vaghulkumar95, thanks for your interest in ThingsBoard IoT Gateway!

> msg.SetPoint_1 is taking the value from modbus.json, can this name be same for different telemetry data in modbus.json or should we name it different inorder for the rule to identify?

It can be the same, depending on your configuration.

> Assume I have 4 modbus devices from which I need to get the data and forward it to bacnet(1device), should I name the telemetry data different or can it be the same? And how does the value flow from all 4 modbus devices to the bacnet in this scenario?

It can be the same but you have to send them to one BACnet device.

> eg. the "tag": "Discharge_Temp" is common for "deviceName": "Heatpump_MFT_1", "deviceName": "Heatpump_MFT_2"...
Is this fine? Or should it be unique for the rule to identify?

It can be whatever you want.

samson0v, 2023-07-07:
@Vaghulkumar95, 

> we are looking for an option, where we are getting notified when the device is inactive(https://thingsboard.io/docs/user-guide/rule-engine-2-0/tutorials/create-inactivity-alarm/ ) but once the data starts to flow again we are wanting some kind of notification that those devices which were offline are now back.

You can send your specific client attribute and use it to determine whether the device is active.

> Do we have any documentation on how to integrate Mbus into thingsboard?

Unfortunately, no.

Vaghulkumar95, 2023-07-07:
Then in the MBus Scenario should I go with custom connector? https://thingsboard.io/docs/iot-gateway/custom/
Or do you have any other suggestions?

samson0v, 2023-07-07:
@Vaghulkumar95 I think will be enough custom converter.

Vaghulkumar95, 2023-07-19:
Hi @samson0v,

Quick question, I'm trying to connect to a gateway via ssh and to execute a set of commands like below to get pitunnel access I need to be a root user.
(For devices running Ubuntu Desktop: wget -qO- https://pitunnel.com/get/PPtyrTELhs | sudo bash)

Can you let me know which password is it expecting in the below image, I tried giving my thingsboard.cloud password and the gateway(pi) password as well, but nothing is working. So just wanted to check with you which is the correct password I need to be using and is the correct way I'm using the -S option?

[image]

[image]

Vaghulkumar95, 2023-07-20:
Hi @samson0v / @imbeacon ,

Any suggesstions on the above question? Is it possible to enable an ssh connection from the above terminal?

Vaghulkumar95, 2023-07-20:
> 
Currently we are not using any attributes, we are just using telemetry, Is it possible to do with the latest telemetry, and where/how should I go ahead and add the specific attribute/telemetry data, is it in activity event Clear alarm or is it somewhere else?
[image]

Vaghulkumar95, 2023-07-21:
Hi @samson0v/@imbeacon,

Just wanted to follow up on this. Any suggestions from your end is highly appreciated.

Vaghulkumar95, 2023-07-24:
Hi @samson0v,
Any suggesstions on the above?

samson0v, 2023-07-24:
Hi @Vaghulkumar95, for an answer to your question, please, contact [support](https://thingsboard.io/docs/contact-us/).

Vaghulkumar95, 2023-08-31:
Hi @samson0v,

I tried contacting support but couldn't get any response? Is there anyone else I can reach out to for both the issues? 
1) Regarding the remote shell 
2) Regarding the alerts

> * 
> *

Vaghulkumar95, 2023-09-08:
Hi @samson0v,

Any updates on the above request?

Vaghulkumar95, 2023-09-18:
Hi @samson0v ,

Any suggesstions on the above?

Vaghulkumar95, 2023-09-28:
Hi @samson0v @imbeacon ,

Quick question, I was trying to install thingsboard-gateway in Debian 11 with python 3.9.2 installed. And got this error. Is this something new or is this a known issue?

rock@rock-3c:~$ sudo apt install ./python3-thingsboard-gateway.deb -y
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'python3-thingsboard-gateway' instead of './python3-thingsboard-gateway.deb'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 python3-thingsboard-gateway : Depends: python3-cryptography (>= 3.4.0) but 3.3.2-1 is to be installed
E: Unable to correct problems, you have held broken packages.

Vaghulkumar95, 2023-09-28:
Hi @samson0v @imbeacon ,

This is something a little urgent which needs to be completed for 50 devices and dispatched. Could you please share your comments on how to proceed with this?

Vaghulkumar95, 2023-10-04:
Hi @samson0v @imbeacon ,

Just wanted to follow up, do you have any suggesstions on the above issues?

HarvUK, 2023-10-26:
@Vaghulkumar95 Did you manage to get TB gateway installed - I have the same error installing on a Raspberry Pi.

HarvUK, 2023-10-26:
> Hi @samson0v, Thanks for your response, I will check on that and update you. Anything on 2nd and 3rd questions?
> 
> 2. we are looking for an option, where we are getting notified when the device is inactive(https://thingsboard.io/docs/user-guide/rule-engine-2-0/tutorials/create-inactivity-alarm/ ) but once the data starts to flow again we are wanting some kind of notification that those devices which were offline are now back.
> 3. Do we have any documentation on how to integrate Mbus into thingsboard?

@Vaghulkumar95 Did you have any success with Mbus - I am very interested in how you got on.

Vaghulkumar95, 2023-10-27:
@HarvUK I was able to install it from pip. https://thingsboard.io/docs/iot-gateway/install/pip-installation/.

For MBus we need to go with https://thingsboard.io/docs/iot-gateway/custom/ but I have not started this yet.

@samson0v do you have any suggesstions on the above issues?

HarvUK, 2023-10-30:
> @HarvUK I was able to install it from pip. https://thingsboard.io/docs/iot-gateway/install/pip-installation/.
> 
> For MBus we need to go with https://thingsboard.io/docs/iot-gateway/custom/ but I have not started this yet.
> 
> @samson0v do you have any suggesstions on the above issues?

@Vaghulkumar95 

I've had some success with loading from PIP - many thanks.

I've been having a play with MBus on a raspberry pi and managed to have some success, although I am sending through MQTT directly rather than using TB gateway.

Let me know how you get on with TB-GW.

Thanks,

Harv

Vaghulkumar95, 2023-12-13:
Hi @samson0v @imbeacon ,

Just wanted to let you know that after upgrading to 3.4.3.1 I'm not getting the data flowing in thingsboard, when I revert back to 3.4.3 it is working fine. I have modbus_serial data. Could you please have a look at this?

[image]

closed 2025-04-16 by imbeacon