thingsboard/thingsboard-gateway #1103: [BUG] Lost data from queue

mircopz, 2023-04-19:
**Describe the bug**
My custom MQTT converter receives two different messages very close in time (about 2 milliseconds). They are both processed and correctly put in `__converted_data_queue` (I add some logs).
Meanwhile, I suppose `__send_to_storage` thread is in its `sleep(0.2)`. 
When `__send_to_storage` wakes up, it found a queue of size 2, but both times the element returned is the last put in the queue. The first element in the queue seems overwritten by the last one.
It seems some concurrency problem, I don't know if the problem is the get or put method. 
Any advice to solve this?
Thank you

**Connector name (If bug in the some connector):**
MQTT Connector

**Error traceback (If available):**
```
2023-04-19 15:03:55" - |INFO| - [my_custom_mqtt_uplink_convert.py] - my_custom_mqtt_uplink_convert - convert - 36 - New message received: {'attributes': [], 'telemetry': [{'ts': 1681916634816, 'values': {'key1': 0, 'key2': 'idle'}}], 'deviceName': 'TEST_DEV', 'deviceType': 'DEVICE'}
2023-04-19 15:03:55" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - send_to_storage - 603 - __converted_data_queue: current size 0, put message {'attributes': [], 'telemetry': [{'ts': 1681916634816, 'values': {'key1': 0, 'key2': 'idle'}}], 'deviceName': 'TEST_DEV', 'deviceType': 'DEVICE'}
2023-04-19 15:03:55" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - send_to_storage - 606 - __converted_data_queue: new size 1
2023-04-19 15:03:55" - |INFO| - [my_custom_mqtt_uplink_convert.py] - my_custom_mqtt_uplink_convert - convert - 36 - New message received: {'attributes': [], 'telemetry': [{'ts': 1681916634818, 'values': {'key3': 1, 'key4': 'running'}}], 'deviceName': 'TEST_DEV', 'deviceType': 'DEVICE'}
2023-04-19 15:03:55" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - send_to_storage - 603 - __converted_data_queue: current size 1, put message {'attributes': [], 'telemetry': [{'ts': 1681916634818, 'values': {'key3': 1, 'key4': 'running'}}], 'deviceName': 'TEST_DEV', 'deviceType': 'DEVICE'}
2023-04-19 15:03:55" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - send_to_storage - 606 - __converted_data_queue: new size 2
2023-04-19 15:03:55" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __send_to_storage - 616 - __converted_data_queue: current size 2, get message [{'attributes': [], 'telemetry': [{'ts': 1681916634818, 'values': {'key3': 1, 'key4': 'running'}}], 'deviceName': 'TEST_DEV', 'deviceType': 'DEVICE'}]
2023-04-19 15:03:55" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __send_to_storage - 616 - __converted_data_queue: current size 1, get message [{'attributes': [], 'telemetry': [{'ts': 1681916634818, 'values': {'key3': 1, 'key4': 'running'}}], 'deviceName': 'TEST_DEV', 'deviceType': 'DEVICE'}]
```

**Versions (please complete the following information):**
 - OS: Ubuntu 18.04
 - Thingsboard IoT Gateway version: 3.1
 - Python version: 3.9

devaskim, 2023-04-30:
@mircopz Did you find the root cause?

mircopz, 2023-05-02:
@devaskim No, I cannot reproduce it using a simple script, so I don't think it's a python problem... Maybe some wrong usage of queue in thingsboard-gateway, I don't know... 
Did you have any similar problem? I don't want to insert some workaround, but I'd like to fix it

devaskim, 2023-05-02:
Sometimes pieces of important data disappear, but I cannot reproduce it as well.

samson0v, 2023-05-02:
Hi @mircopz, thanks for your interest in ThingsBoard IoT Gateway!
Send, please, your config files (tb_gateway.yml and mqtt.json) and your custom converter.

mircopz, 2023-05-02:
I found the problem: the value returned by a converter must not be a reference to a static variable.

My connector is similar to the original `CustomMqttUplinkConverter`: the `convert` function is using a class variable (`self.dict_result`) to append the processed telemetry and then returning it at the end.
In this way the converter is using a class variable, queuing it twice with different values. When I dequeue them, I found two elements with the same reference, but every time its value is the last value assigned to the variable (the last processed message).
The solution is to use a local variable inside the `convert` function (`dict_result` instead of `self.dict_result`). Now everything is working fine.
Otherwise, I think a copy of the message should be done before putting it in the queue.

closed 2023-05-02 by mircopz