thingsboard/thingsboard-gateway #1105: [HELP] run generate_deb_package.sh error

Kevin-ox, 2023-04-27:
**Describe the issue**

generate_deb_package.sh error, sudo python3 ./thingsboard_gateway/tb_gateway.py is normal,I've tried many times.

Thanks

**Error traceback (If it was raised):**
```
thingsboard_gateway@DESKTOP-DTJC5LC:~/thingsboard-gateway$ sudo python3 ./thingsboard_gateway/tb_gateway.py
[sudo] password for thingsboard_gateway:
""2023-04-27 11:29:35" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 180 - Gateway starting..."
""2023-04-27 11:29:35" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 185 - ThingsBoard IoT gateway version: 3.2"
""2023-04-27 11:29:35" - |INFO| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - gw_subscribe_to_attribute - 185 - Subscribed to *|* with id 1 for device *"
""2023-04-27 11:29:35" - |ERROR| - [tb_gateway_service.py] - tb_gateway_service - _load_connectors - 661 - Connectors - not found! Check your configuration!"
""2023-04-27 11:29:35" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - _load_connectors - 663 - Remote configuration is enabled forcibly!"
""2023-04-27 11:29:35" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 300 - Gateway started."
""2023-04-27 11:29:35" - |ERROR| - [tb_device_mqtt.py] - tb_device_mqtt - _on_connect - 223 - connection FAIL with error 5 not authorised"
""2023-04-27 11:29:36" - |INFO| - [tb_updater.py] - tb_updater - check_for_new_version - 86 - v3.2"
""2023-04-27 11:29:36" - |INFO| - [tb_updater.py] - tb_updater - check_for_new_version - 88 -

[===UPDATE===]

 New version v3.2 is available!

[===UPDATE===]
"
^C""2023-04-27 11:29:41" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __stop_gateway - 427 - Stopping..."
""2023-04-27 11:29:41" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __stop_gateway - 436 - The gateway has been stopped."
""2023-04-27 11:29:41" - |INFO| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - gw_unsubscribe - 193 - Unsubscribed from *|*, subscription id '*'"
thingsboard_gateway@DESKTOP-DTJC5LC:~/thingsboard-gateway$
thingsboard_gateway@DESKTOP-DTJC5LC:~/thingsboard-gateway$
thingsboard_gateway@DESKTOP-DTJC5LC:~/thingsboard-gateway$
thingsboard_gateway@DESKTOP-DTJC5LC:~/thingsboard-gateway$
thingsboard_gateway@DESKTOP-DTJC5LC:~/thingsboard-gateway$
thingsboard_gateway@DESKTOP-DTJC5LC:~/thingsboard-gateway$ sudo sh generate_deb_package.sh 3.2
Installing libraries for building deb package.
Reading package lists... Done
Building dependency tree
Reading state information... Done
fakeroot is already the newest version (1.24-1).
dh-python is already the newest version (4.20191017ubuntu7).
python-all is already the newest version (2.7.17-2ubuntu4).
python3-stdeb is already the newest version (0.8.5-3).
zstd is already the newest version (1.4.4+dfsg-3ubuntu0.1).
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
Building DEB package
Creating sources for DEB package...
running bdist_deb
running sdist_dsc
running egg_info
writing thingsboard_gateway.egg-info/PKG-INFO
writing dependency_links to thingsboard_gateway.egg-info/dependency_links.txt
writing entry points to thingsboard_gateway.egg-info/entry_points.txt
writing requirements to thingsboard_gateway.egg-info/requires.txt
writing top-level names to thingsboard_gateway.egg-info/top_level.txt
reading manifest file 'thingsboard_gateway.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'thingsboard_gateway.egg-info/SOURCES.txt'
running sdist
running check
creating thingsboard-gateway-3.2
creating thingsboard-gateway-3.2/thingsboard_gateway
creating thingsboard-gateway-3.2/thingsboard_gateway.egg-info
creating thingsboard-gateway-3.2/thingsboard_gateway/config
creating thingsboard-gateway-3.2/thingsboard_gateway/config/statistics
creating thingsboard-gateway-3.2/thingsboard_gateway/connectors
creating thingsboard-gateway-3.2/thingsboard_gateway/connectors/bacnet
creating thingsboard-gateway-3.2/thingsboard_gateway/connectors/bacnet/bacnet_utilities
creating thingsboard-gateway-3.2/thingsboard_gateway/connectors/ble
creating thingsboard-gateway-3.2/thingsboard_gateway/connectors/can
[... 332 lines cut ...]
chown: cannot access 'deb_dist/thingsboard-gateway-/debian/python3-thingsboard-gateway/DEBIAN/preinst': No such file or directory
sh: 1: cannot create deb_dist/thingsboard-gateway-/debian/python3-thingsboard-gateway/DEBIAN/control: Directory nonexistent
dpkg-deb: error: failed to open package info file 'deb_dist/thingsboard-gateway-/debian/python3-thingsboard-gateway//DEBIAN/control' for reading: No such file or directory
ar: ../deb_dist/thingsboard-gateway-/debian/python3-thingsboard-gateway.deb: No such file or directory
zstd: can't stat *.zst : No such file or directory -- ignored
rm: cannot remove '*.zst': No such file or directory
xz: *.tar: No such file or directory
ar: creating ../python3-thingsboard-gateway.deb
ar: debian-binary: No such file or directory
thingsboard_gateway@DESKTOP-DTJC5LC:~/thingsboard-gateway$ curl www.google.com
<HTML><HEAD><meta http-equiv="content-type" content="text/html;charset=utf-8">
<TITLE>302 Moved</TITLE></HEAD><BODY>
<H1>302 Moved</H1>
The document has moved
<A HREF="http://www.google.com.hk/url?sa=p&amp;hl=zh-CN&amp;pref=hkredirect&amp;pval=yes&amp;q=http://www.google.com.hk/&amp;ust=1682562293683807&amp;usg=AOvVaw3dNgV5wLKUdcVaBCuFPrQr">here</A>.
</BODY></HTML>
thingsboard_gateway@DESKTOP-DTJC5LC:~/thingsboard-gateway$
thingsboard_gateway@DESKTOP-DTJC5LC:~/thingsboard-gateway$
thingsboard_gateway@DESKTOP-DTJC5LC:~/thingsboard-gateway$
```

**Versions (please complete the following information):**
 - OS: [e.g. Ubuntu 20.04]
 - Thingsboard IoT Gateway version master
 - Python version[e.g. 3.8.10]

samson0v, 2023-04-27:
Hi @Kevin-ox, thanks for your interest in ThingsBoard IoT Gateway!
You have to clone the repo using the following command:
`git clone --recurse-submodules https://github.com/thingsboard/thingsboard-gateway.git`

Also, you can read about source installation in the [official documentation](https://thingsboard.io/docs/iot-gateway/install/source-installation/).

Kevin-ox, 2023-04-27:
Yes, the operation is fine, but the packaging is wrong.

```
tb-mqtt-client library not found - installing...
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProxyError('Cannot connect to proxy.', NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x7f87df786220>: Failed to establish a new connection: [Errno 111] Connection refused'))': /simple/tb-mqtt-client/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProxyError('Cannot connect to proxy.', NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x7f87df786340>: Failed to establish a new connection: [Errno 111] Connection refused'))': /simple/tb-mqtt-client/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProxyError('Cannot connect to proxy.', NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x7f87df786670>: Failed to establish a new connection: [Errno 111] Connection refused'))': /simple/tb-mqtt-client/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProxyError('Cannot connect to proxy.', NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x7f87df7867c0>: Failed to establish a new connection: [Errno 111] Connection refused'))': /simple/tb-mqtt-client/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProxyError('Cannot connect to proxy.', NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x7f87df786f10>: Failed to establish a new connection: [Errno 111] Connection refused'))': /simple/tb-mqtt-client/
ERROR: Could not find a version that satisfies the requirement tb-mqtt-client (from versions: none)
ERROR: No matching distribution found for tb-mqtt-client
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProxyError('Cannot connect to proxy.', NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x7f7cd74d7d90>: Failed to establish a new connection: [Errno 111] Connection refused'))': /simple/tb-mqtt-client/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProxyError('Cannot connect to proxy.', NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x7f7cd74d7a90>: Failed to establish a new connection: [Errno 111] Connection refused'))': /simple/tb-mqtt-client/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProxyError('Cannot connect to proxy.', NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x7f7cd74d7550>: Failed to establish a new connection: [Errno 111] Connection refused'))': /simple/tb-mqtt-client/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProxyError('Cannot connect to proxy.', NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x7f7cd74d7190>: Failed to establish a new connection: [Errno 111] Connection refused'))': /simple/tb-mqtt-client/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProxyError('Cannot connect to proxy.', NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x7f7cd74d7880>: Failed to establish a new connection: [Errno 111] Connection refused'))': /simple/tb-mqtt-client/
ERROR: Could not find a version that satisfies the requirement tb-mqtt-client (from versions: none)
ERROR: No matching distribution found for tb-mqtt-client
Traceback (most recent call last):
  File "/home/thingsboard_gateway/project/thingsboard-gateway/deb_dist/thingsboard-gateway-3.2/thingsboard_gateway/gateway/tb_client.py", line 28, in <module>
    from tb_gateway_mqtt import TBGatewayMqttClient, TBDeviceMqttClient
ModuleNotFoundError: No module named 'tb_gateway_mqtt'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/thingsboard_gateway/project/thingsboard-gateway/deb_dist/thingsboard-gateway-3.2/thingsboard_gateway/tb_utility/tb_utility.py", line 146, in install_package
    result = check_call([executable, "-m", "pip", "install", package, "--upgrade", "--user"])
  File "/usr/lib/python3.8/subprocess.py", line 364, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/bin/python3.8', '-m', 'pip', 'install', 'tb-mqtt-client', '--upgrade', '--user']' returned non-zero exit status 1.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "setup.py", line 20, in <module>
    from thingsboard_gateway.gateway.constants import VERSION
  File "/home/thingsboard_gateway/project/thingsboard-gateway/deb_dist/thingsboard-gateway-3.2/thingsboard_gateway/__init__.py", line 15, in <module>
    from thingsboard_gateway.gateway.tb_gateway_service import TBGatewayService
  File "/home/thingsboard_gateway/project/thingsboard-gateway/deb_dist/thingsboard-gateway-3.2/thingsboard_gateway/gateway/tb_gateway_service.py", line 42, in <module>
    from thingsboard_gateway.gateway.tb_client import TBClient
[... cut ...]
dpkg-deb: error: failed to open package info file 'deb_dist/thingsboard-gateway-/debian/python3-thingsboard-gateway//DEBIAN/control' for reading: No such file or directory
ar: ../deb_dist/thingsboard-gateway-/debian/python3-thingsboard-gateway.deb: No such file or directory
zstd: can't stat *.zst : No such file or directory -- ignored 
rm: cannot remove '*.zst': No such file or directory
xz: *.tar: No such file or directory
ar: creating ../python3-thingsboard-gateway.deb
ar: debian-binary: No such file or directory
```

samson0v, 2023-05-02:
@Kevin-ox, try to set the Python path:
`PYTHONPATH=.` (in project root folder)

Kevin-ox, 2023-05-04:
I tried PYTHONPATH=. And export PYTHONPATH=. The problem still exists，
docker packaging is also normal
Only generate_deb_package.sh is abnormal

NikolinaKote, 2023-05-04:
Hello, i am facing the same issue, @Kevin-ox did you manage to find a solution?

imbeacon, 2023-05-05:
Hi,
Please try to run the following command: `git pull origin master --recurse-submodules`
And `pip install tb-mqtt-client`

Kevin-ox, 2023-05-05:
The problem still exists.

`thingsboard_gateway@DESKTOP-KOJPFOG:~$ git clone --recurse-submodules https://github.com/thingsboard/thingsboard-gateway.git
Cloning into 'thingsboard-gateway'...
remote: Enumerating objects: 12907, done.
remote: Counting objects: 100% (173/173), done.
remote: Compressing objects: 100% (74/74), done.
remote: Total 12907 (delta 104), reused 147 (delta 99), pack-reused 12734
Receiving objects: 100% (12907/12907), 10.75 MiB | 11.01 MiB/s, done.
Resolving deltas: 100% (8666/8666), done.
Submodule 'tb_mqtt_client' (https://github.com/thingsboard/thingsboard-python-client-sdk.git) registered for path 'tb_mqtt_client'
Cloning into '/home/thingsboard_gateway/thingsboard-gateway/tb_mqtt_client'...
remote: Enumerating objects: 898, done.        
remote: Counting objects: 100% (280/280), done.        
remote: Compressing objects: 100% (84/84), done.        
remote: Total 898 (delta 223), reused 229 (delta 196), pack-reused 618        
Receiving objects: 100% (898/898), 192.13 KiB | 690.00 KiB/s, done.
Resolving deltas: 100% (497/497), done.
Submodule path 'tb_mqtt_client': checked out '8dfcac37b7fb9b6054eec69b8c1e1a0a775ae8e8'
thingsboard_gateway@DESKTOP-KOJPFOG:~$ 
thingsboard_gateway@DESKTOP-KOJPFOG:~$ ls
ligoothings  project  thingsboard-gateway
thingsboard_gateway@DESKTOP-KOJPFOG:~$ 
thingsboard_gateway@DESKTOP-KOJPFOG:~$ 
thingsboard_gateway@DESKTOP-KOJPFOG:~$ cd thingsboard-gateway/
thingsboard_gateway@DESKTOP-KOJPFOG:~/thingsboard-gateway$ ls
LICENSE  MANIFEST.in  README.md  docker  for_build  generate_deb_package.sh  make_packages.sh  requirements.txt  setup.py  tb_mqtt_client  tests  thingsboard-gateway.spec  thingsboard_gateway
thingsboard_gateway@DESKTOP-KOJPFOG:~/thingsboard-gateway$ 
thingsboard_gateway@DESKTOP-KOJPFOG:~/thingsboard-gateway$ 
thingsboard_gateway@DESKTOP-KOJPFOG:~/thingsboard-gateway$ sudo python3 setup.py install
[sudo] password for thingsboard_gateway: 
running install
running bdist_egg
running egg_info
creating thingsboard_gateway.egg-info
writing thingsboard_gateway.egg-info/PKG-INFO
writing dependency_links to thingsboard_gateway.egg-info/dependency_links.txt
writing entry points to thingsboard_gateway.egg-info/entry_points.txt
writing requirements to thingsboard_gateway.egg-info/requires.txt
writing top-level names to thingsboard_gateway.egg-info/top_level.txt
writing manifest file 'thingsboard_gateway.egg-info/SOURCES.txt'
reading manifest file 'thingsboard_gateway.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'thingsboard_gateway.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
creating build
creating build/lib
creating build/lib/thingsboard_gateway
copying thingsboard_gateway/__init__.py -> build/lib/thingsboard_gateway
copying thingsboard_gateway/tb_gateway.py -> build/lib/thingsboard_gateway
creating build/lib/thingsboard_gateway/gateway
copying thingsboard_gateway/gateway/constants.py -> build/lib/thingsboard_gateway/gateway
copying thingsboard_gateway/gateway/__init__.py -> build/lib/thingsboard_gateway/gateway
copying thingsboard_gateway/gateway/configuration_wizard.py -> build/lib/thingsboard_gateway/gateway
copying thingsboard_gateway/gateway/constant_enums.py -> build/lib/thingsboard_gateway/gateway
copying thingsboard_gateway/gateway/tb_client.py -> build/lib/thingsboard_gateway/gateway
copying thingsboard_gateway/gateway/tb_gateway_service.py -> build/lib/thingsboard_gateway/gateway
copying thingsboard_gateway/gateway/device_filter.py -> build/lib/thingsboard_gateway/gateway
copying thingsboard_gateway/gateway/duplicate_detector.py -> build/lib/thingsboard_gateway/gateway
copying thingsboard_gateway/gateway/hot_reloader.py -> build/lib/thingsboard_gateway/gateway
copying thingsboard_gateway/gateway/statistics_service.py -> build/lib/thingsboard_gateway/gateway
creating build/lib/thingsboard_gateway/gateway/proto
copying thingsboard_gateway/gateway/proto/messages_pb2_grpc.py -> build/lib/thingsboard_gateway/gateway/proto
copying thingsboard_gateway/gateway/proto/messages_pb2.py -> build/lib/thingsboard_gateway/gateway/proto
copying thingsboard_gateway/gateway/proto/__init__.py -> build/lib/thingsboard_gateway/gateway/proto
creating build/lib/thingsboard_gateway/gateway/grpc_service
copying thingsboard_gateway/gateway/grpc_service/__init__.py -> build/lib/thingsboard_gateway/gateway/grpc_service
copying thingsboard_gateway/gateway/grpc_service/grpc_connector.py -> build/lib/thingsboard_gateway/gateway/grpc_service
[... 1119 lines cut ...]
    process_command(args,cwd=cwd)
  File "/usr/lib/python3/dist-packages/stdeb/util.py", line 183, in process_command
    check_call(args, cwd=cwd)
  File "/usr/lib/python3/dist-packages/stdeb/util.py", line 46, in check_call
    raise CalledProcessError(retcode)
stdeb.util.CalledProcessError: 2
Adding the files, scripts and permissions in the package
cp: cannot create directory 'deb_dist/thingsboard-gateway-/debian/python3-thingsboard-gateway': No such file or directory
cp: cannot create directory 'deb_dist/thingsboard-gateway-/debian/python3-thingsboard-gateway': No such file or directory
cp: cannot create directory 'deb_dist/thingsboard-gateway-/debian/python3-thingsboard-gateway': No such file or directory
chown: cannot access 'deb_dist/thingsboard-gateway-/debian/python3-thingsboard-gateway/': No such file or directory
[... cut ...]
dpkg-deb: error: failed to open package info file 'deb_dist/thingsboard-gateway-/debian/python3-thingsboard-gateway//DEBIAN/control' for reading: No such file or directory
ar: ../deb_dist/thingsboard-gateway-/debian/python3-thingsboard-gateway.deb: No such file or directory
zstd: can't stat *.zst : No such file or directory -- ignored 
rm: cannot remove '*.zst': No such file or directory
xz: *.tar: No such file or directory
ar: creating ../python3-thingsboard-gateway.deb
ar: debian-binary: No such file or directory
thingsboard_gateway@DESKTOP-KOJPFOG:~/thingsboard-gateway$ 
`

imbeacon, 2023-05-26:
Please try the version from the master branch.

Kevin-ox, 2023-05-27:
Thanks. It worked, but there were a few mistakes at the end
```
dpkg-deb: building package 'python3-thingsboard-gateway' in '../python3-thingsboard-gateway_3.3-1_all.deb'.
 dpkg-genbuildinfo --build=binary
 dpkg-genchanges --build=binary >../thingsboard-gateway_3.3-1_amd64.changes
dpkg-genchanges: info: binary-only upload (no source code included)
 dpkg-source --after-build .
dpkg-buildpackage: info: binary-only upload (no source included)
Adding the files, scripts and permissions to the package
dpkg-deb: building package 'python3-thingsboard-gateway' in 'deb_dist/thingsboard-gateway-3.3/debian/python3-thingsboard-gateway.deb'.
zstd: can't stat *.zst : No such file or directory -- ignored
rm: cannot remove '*.zst': No such file or directory
xz: *.tar: No such file or directory
```

Kevin-ox, 2023-06-01:
Run successfully!
Executed after the installation is complete
```
sudo pip3 install --upgrade protobuf==3.20.0
```

samson0v, 2023-06-01:
Hi @Kevin-ox, can we close the issue?

Kevin-ox, 2023-06-01:
Of course. Thank you.

closed 2023-06-01 by samson0v