thingsboard/thingsboard-gateway #1106: [BUG] Server side RPC: data instead of params

devaskim, 2023-04-30:
**Describe the bug**

!!! Note !!!  For some reasons `data` key is used in others method of the MQTT connector. Is it breaking change?

[image]

**Connector name (If bug in the some connector):**
MQTT Connector

**Versions (please complete the following information):**
 - Thingsboard 3.4.1
 - Thingsboard IoT Gateway version: master

samson0v, 2023-05-02:
Hi @devaskim, thanks for your interest in ThingsBoard IoT Gateway!
Send, please, your MQTT Connector config file.

devaskim, 2023-05-02:
Thanks @samson0v for your attention

My MQTT Connector configuration isn't a secret, but I think, it doesn't help, because there is **only** `mapping` section.
To send RPC, I use default `RPC button` widget.

[image]

devaskim, 2023-05-02:
I patched a bit code of MQTT connector, so once the gateway gets this RPC, it will call [self._send_current_converter_config()](https://github.com/thingsboard/thingsboard-gateway/blob/master/thingsboard_gateway/connectors/mqtt/mqtt_connector.py#L445) which you have added recently.

samson0v, 2023-05-02:
@devaskim
[image]
[image]

devaskim, 2023-05-02:
Strange, what is your server version? Mine is 3.4.1

samson0v, 2023-05-02:
@devaskim, I can't see `device` in your content.
It won't work even if you change data to params.

samson0v, 2023-05-02:
@devaskim try to check your Entity Alias, maybe it is the cause of the problem.

samson0v, 2023-05-02:
@devaskim 
[image]
[image]
[image]

devaskim, 2023-05-02:
I got my RPC on gateway side, so entity alias is correct and the same is yours. Here is the code of gateway with my fix and patch for new RPC method for connector.

[image]

devaskim, 2023-05-02:
@samson0v So (my) problem occurs on the step 1

samson0v, 2023-05-02:
@devaskim, ok, I understand how it is working.
You can't simply change data to params because RPC for the device and gateway are different.
I will create a PR.

devaskim, 2023-05-02:
Thanks @samson0v for help
Could you add RPC for synchronizing converters' config as I did? If yes, the name is up to you.

samson0v, 2023-05-02:
@devaskim, we decided that client attributes will update every time when the converter config updates.

devaskim, 2023-05-02:
> we decided that client attributes will update every time when the converter config updates

@samson0v It may be even more convenient, but how does connector know if **custom** converter config is changed? Would it be some callback that converter could call on its config update? Just to remind, our converter has 2 part of config: static (from `mqtt.json`) and dynamic, so called, dictionaries which converter receives from MQTT broker.

devaskim, 2023-05-02:
Sad, it doesn't cover our case.

Do you think the flow you have described is needed at all? I mean, you just duplicates data from shared attribute to client attribute.

imbeacon, 2023-05-03:
@devaskim These parameters depends on the entitiy that you uses to send RPC for. You can easily change in your version of the MQTT connector and use it. We will think about mechanism that we can develop to give ability to implement custom RPC for connectors, But it will be in the future, after releasing a new version of the gateway remote configuration.

devaskim, 2023-05-03:
Ok, thanks team

closed 2023-05-03 by devaskim