thingsboard/thingsboard-gateway #1115: [HELP]

a-jiang, 2023-05-09:
**Describe the issue**
i got a log error .
[image]

and log files are empty 
[image]

**Configuration** (Attach your configuration file)
**Notate: Remove Access token from file if you want to attach a tb_gateway.yaml**

**Connector name (If you need help with some connector/converter):**
[e.g. MQTT Connector]

**Error traceback (If it was raised):**
```
'deviceName'
Traceback (most recent call last):
  File "<input>", line 2, in <module>
KeyError: 'deviceName'
```

**Versions (please complete the following information):**
 - OS: [e.g. Ubuntu 20.04]
 - Thingsboard IoT Gateway version [e.g. 3.0.1]
 - Python version[e.g. 3.9]

samson0v, 2023-07-06:
Hi @a-jiang, thanks for your interest in ThingsBoard IoT Gateway!
Please, send configuration files (connector config file, tb_gateway.yml, logs.conf) and log files.

samson0v, 2023-09-29:
Hi @a-jiang, can we close the issue?

closed 2023-10-13 by samson0v