thingsboard/thingsboard-gateway #1117: Thingsboard Gateway does not send ALL data to server.

santobedi, 2023-05-16:
**Describe the issue**
Thingsboard Gateway [docker container] is able to create devices and keys (telemetry and attribute) at the Thingsboard server. However, it does not report every change in attribute and telemetry values on the server. The data source to the gateway is another Thingsboard server (production server) wherein `External mqtt` node is used to publish `Post attributes` and `Post telemetry` messages of some devices at our MQTT Broker. The gateway subscribes to those messages and forwards them to our dev server. The production server publishes under the `Production/${deviceName}/data` topic. The MQTT connector (mqtt.json) uses `topicFilter` as `Production/+/data` to subscribe to those messages. All the devices are created at the dev server via the gateway and attributes/telemetry are updated. The issue is we don't see every change on the telemetry of a device at the Production server in the dev server.  The number of devices is 70 and each device has two attributes and a telemetry (test case). Is there any limitation on the number of devices that the gateway supports? Please note that we have the same issue when the gateway was installed in Raspberry Pi 3.

**Configuration** 
```
thingsboard:
  host: server ip
  port: 8883
  remoteShell: false
  remoteConfiguration: false
  statistics:
    enable: true
    statsSendPeriodInSeconds: 3600
    configuration: statistics.json
  maxPayloadSizeBytes: 1024
  minPackSendDelayMS: 0
  checkConnectorsConfigurationInSeconds: 60
  handleDeviceRenaming: true
  checkingDeviceActivity:
    checkDeviceInactivity: false
    inactivityTimeoutSeconds: 1200
    inactivityCheckPeriodSeconds: 600
  security:
    accessToken: device access token
    caCert: myfile.pub.pem
  qos: 1
storage:
#  type: memory
#  read_records_count: 100
#  max_records_count: 100000
#  type: file
#  data_folder_path: ./data/
#  max_file_count: 10
#  max_read_records_count: 10
#  max_records_per_file: 10000
  type: sqlite
  data_file_path: /thingsboard_gateway/data/data.db
  messages_ttl_check_in_hours: 1
  messages_ttl_in_days: 2
grpc:
  enabled: false
  serverPort: 9595
  keepaliveTimeMs: 10000
  keepaliveTimeoutMs: 5000
  keepalivePermitWithoutCalls: true
  maxPingsWithoutData: 0
  minTimeBetweenPingsMs: 10000
  minPingIntervalWithoutDataMs: 5000
connectors:
  -
    name: MQTT Broker Connector
    type: mqtt
    configuration: mqtt.json
```

**Connector name (If you need help with some connector/converter):**
MQTT Connector

```
{
  "broker": {
    "name": "my Broker",
    "host": "[broker's IP]",
    "port": 8883,
    "clientId": "ThingsBoard_gateway_docker",
    "version": 5,
    "maxMessageNumberPerWorker": 200,
    "maxNumberOfWorkers": 100,
    "security": {
      "type": "tls",
      "username": "username",
      "password": "pw"
    }
  },
  "mapping": [
    {
      "topicFilter": "Production/+/data",
      "converter": {
        "type": "json",
        "deviceNameTopicExpression": "(?<=Production\/)(.*?)(?=\/data)",
        "deviceTypeTopicExpression": "default",
        "timeout": 1200000,
        "attributes": [
          {
            "type": "string",
            "key": "ts",
            "value": "${ts}"
          },
          {
            "type": "double",
            "key": "status",
            "value": "${status}"
          }
        ],
        "timeseries": [
          {
            "type": "double",
            "key": "ac",
            "value": "${ac}"
          }
        ]
      }
    }
  ]
}

```

**Error traceback (If it was raised):**
```
""2023-05-16 08:41:59" - |ERROR| - [statistics_service.py] - statistics_service - run - 67 - Statistic parameter raise the exception: /bin/sh: 1: sw_vers: not found
"
""2023-05-16 08:41:59" - |ERROR| - [statistics_service.py] - statistics_service - run - 67 - Statistic parameter raise the exception: /bin/sh: 1: uptime: not found
"
""2023-05-16 08:41:59" - |ERROR| - [statistics_service.py] - statistics_service - run - 67 - Statistic parameter raise the exception: /bin/sh: 1: system_profiler: not found
```

**Versions (please complete the following information):**
 - OS: Debian GNU/Linux 11 (bullseye)
 - Thingsboard IoT Gateway version: Latest Docker Image
 - Python version: 3.10.8

mahmutsalman, 2023-05-29:
I have the exact problem.

santobedi, 2023-06-05:
@imbeacon @samson0v  Could you look into this issue, please?

mursisoy, 2023-09-05:
Hi there,
I was experiencing a similar issue. The data saved was not sent. There where to issues here I was able to detect:

- From a restart, old data was never read
- While running, data could be inserted after `read_data` call and `self.last_read = time()` causing that data to never be read ([sqlite_event_storage.py#L63](https://github.com/thingsboard/thingsboard-gateway/blob/a2da5b836bddaa15a9e8e5db8ee959bddf9796a0/thingsboard_gateway/storage/sqlite/sqlite_event_storage.py#L63)).

Another issue I detected is that `qos = 1` in `tb_gateway.yaml` configuration file slows down the performance of the gateway. As a result, depending on the velocity of the data, the gateway is not able to flush all the data. But this is out of scope.

I'll drop a pull request with the fix I came out.

Regards

closed 2023-09-12 by imbeacon