thingsboard/thingsboard-gateway #1118: [HELP] Is there an easy way to read data from multiple devices?

UprightNeether, 2023-05-18:
I have to read about 40 timesery datas from single Device
and there is more then 100 device need to Configuration

Its makes modbus.json too large to maintain

it there any keywords like include or import?so I can write timeseries in another file?

samson0v, 2023-05-18:
Hi @UprightNeether, thanks for your interest in ThingsBoard IoT Gateway!
We will add additional options for parameter `unitId` values: '*', (1, 10), [1, 2, 10, 45].

UprightNeether, 2023-05-19:
Thanks for your quick reply~

How long will take your guys to finish this feature?
If this feature will takes several weeks,I will make a simple patch by my own,(*'▽')

samson0v, 2023-05-19:
@UprightNeether if you can, please make a PR for this feature.

If the user sets the parameter: `unitId: "*"` - connector iterates from 0 to 255 and appends found devices to the separate array (in order to know exactly which devices are there and not to perform the search function every time).
The same logic for `unitId: "(1,13)"` - connector iterates from 1 to 13 unit ids.
and for this one `unitId: "[1, 4, 5, 6, 12]"`

Feel free to ask questions.

UprightNeether, 2023-05-19:
I will try to do this, thanks for help ('ω')ノ

cheng521314, 2023-08-11:
我也遇到了这个问题，请问解决了吗？

UprightNeether, 2023-08-11:
> 我也遇到了这个问题，请问解决了吗？

没有，直接在excel里生成得了

cheng521314, 2023-08-11:
> > 我也遇到了这个问题，请问解决了吗？
> 
> 没有，直接在excel里生成得了

在2.X版本时候，如果设备同一个ip，可以配置一次，通过devices中的unitId来区分不同的设备，这样只占用了一个连接，为什么到3.X版本不支持这样配置了，每一个modbus设备都需要配置一次IP，这样很不友好，比如我们的设备最大只支持10个tcp连接数量，这样在3.x中，就无法达到一个ip地址连接，控制多个modbus设备了，您有什么更好的建议吗？？

UprightNeether, 2023-08-11:
> > > 我也遇到了这个问题，请问解决了吗？
> > 
> > 
> > 没有，直接在excel里生成得了
> 
> 在2.X版本时候， [image] 如果设备同一个ip，可以配置一次，通过devices中的unitId来区分不同的设备，这样只占用了一个连接，为什么到3.X版本不支持这样配置了 [image] ，每一个modbus设备都需要配置一次IP，这样很不友好，比如我们的设备最大只支持10个tcp连接数量，这样在3.x中，就无法达到一个ip地址连接，控制多个modbus设备了，您有什么更好的建议吗？？

我的设备只支持1个连接,所以后一个会踢走前一个的连接。但是可以看下源码，是放在queue中处理的，所以除了每次都要建立新的连接外没啥问题

UprightNeether, 2023-08-11:
可以在这里加了个变量
[image]
然后在这里存取
[image]

cheng521314, 2023-08-11:
> 可以在这里加了个变量 [image] 然后在这里存取 [image]

十分感谢。。对我来说太难了，我是做java的，看不懂这，等下个版本更新吧。。。

LukaszDuss, 2023-08-31:
> You can add a variable here and then access it here [image] [image]

can you elaborate?  what file is this? what kind variable can i add there?

UprightNeether, 2023-08-31:
what file is this?->thingsboard_gateway/connectors/modbus/modbus_connector.py
what kind variable can i add there?->I add an dictionary
can you elaborate?->add a dictionary to cache ModbusTcpClient to avoid create a new ModbusTcpClient every time

zhangdayeIMAX, 2024-01-29:
我也碰到同样问题，一台modbus TCP上有11个网关，每个网关上有19台空调，每个空调有10个地址，需要209个IP连接才能完成，modbus连接，CPU资源190+%占用非常高，gateway稳定性极差。
谢谢@UprightNeether的思路。

判断现在的ip和端口是否在字典库中，有就使用现有连接，没有就新增一个连接，这样减少了连接数，CPU资源60%左右。
我是这种修改的，基于3.4.1版本gateway
新建一个类

`class ModbusClientManager:
    _instance = None

    def __new__(cls):
        if cls._instance is None:
            cls._instance = super(ModbusClientManager, cls).__new__(cls)
            cls._instance.clients = {}
        return cls._instance
    def get_client(self, config):
        key = f"{config['host']}:{config['port']}"
        if key not in self.clients:
            # 根据不同的配置类型创建不同的客户端
            if config.get('type') == 'tcp':
                if config.get('tls'):
                    self.clients[key] = ModbusTlsClient(config["host"],
                                                        config["port"],
                                                        FRAMER_TYPE[config['method']],
                                                        timeout=config["timeout"],
                                                        retry_on_empty=config["retry_on_empty"],
                                                        retry_on_invalid=config["retry_on_invalid"],
                                                        retries=config["retries"],
                                                        **config['tls'])
                else:
                    self.clients[key] = ModbusTcpClient(config["host"],
                                                        config["port"],
                                                        FRAMER_TYPE[config['method']],
                                                        timeout=config["timeout"],
                                                        retry_on_empty=config["retry_on_empty"],
                                                        retry_on_invalid=config["retry_on_invalid"],
                                                        retries=config["retries"])
            elif config.get('type') == 'udp':
                self.clients[key] = ModbusUdpClient(config["host"],
                                                    config["port"],
                                                    FRAMER_TYPE[config['method']],
                                                    timeout=config["timeout"],
                                                    retry_on_empty=config["retry_on_empty"],
                                                    retry_on_invalid=config["retry_on_invalid"],
                                                    retries=config["retries"])
            elif config.get('type') == 'serial':
                self.clients[key] = ModbusSerialClient(method=config["method"],
                                                       port=config["port"],
                                                       timeout=config["timeout"],
                                                       retry_on_empty=config["retry_on_empty"],
                                                       retry_on_invalid=config["retry_on_invalid"],
                                                       retries=config["retries"],
                                                       baudrate=config["baudrate"],
                                                       stopbits=config["stopbits"],
                                                       bytesize=config["bytesize"],
                                                       parity=config["parity"],
                                                       strict=config["strict"])
        return self.clients[key]
 `
修改def __configure_master，去掉装饰器后

`    def __configure_master(self, config):
        current_config = config
        current_config["rtu"] = FRAMER_TYPE[current_config['method']]
        master = self.client_manager.get_client(current_config)
        available_functions = {
            1: master.read_coils,
            2: master.read_discrete_inputs,
            3: master.read_holding_registers,
            4: master.read_input_registers,
            5: master.write_coil,
            6: master.write_register,
            15: master.write_coils,
            16: master.write_registers,
        }
        return master, available_functions`
目前使用正常

closed 2024-06-19 by imbeacon