thingsboard/thingsboard-gateway #1119: [HELP]MQTT connector attributeUpdates problem

coder7675, 2023-05-21:
**Describe the issue**

I am learning and using the gateway, and I tested the connectRequests, disconnectRequests, and attributeRequests sections, and it worked properly. However, the attributeUpdates section does not work. It seems that the gateway did not receive any messages from tb. Did I miss something? How should I troubleshoot the problem?

Here are some screenshots of the details:
[image]
[image]
[image]
[image]

When the value is updated, gateway does not receive any messages

**Configuration**
```   
// mqtt.json

  "attributeUpdates": [
    {
      "retain": false,
      "deviceNameFilter": "SN-001",
      "attributeFilter": "FirmwareVersion",
      "topicExpression": "/sensor/${deviceName}/${attributeKey}",
      "valueExpression": "{\"${attributeKey}\":\"${attributeValue}\"}"
    }
  ]
````

**Connector name:**
MQTT Connector

**Versions (please complete the following information):**
 - OS: [Win11]
 - Thingsboard IoT Gateway version [3.2]
 - Python version[3.9]
 - Thingsboard CE version [3.4.0]

Thank you,Best regards.

coder7675, 2023-05-24:
I found that if the device is connected and then the gateway restarts, when the device try to request connected again, TB is not connected to the device unless the device request disconnected first. I am not sure if this behavior is normal. I solved my problem by modifying the code for method add_device.

imbeacon, 2023-05-24:
Hi @coder7675 ,
Thank you for your message and investigation, we will check this point.

imbeacon, 2024-06-19:
The issue relates to the device activity and it was caused by removing the device session on the ThingsBoard side, it was fixed in ThingsBoard 3.7, but if your device doesn't publish data more often than once per 5 minutes - it is recommended to enable "Overwrite device activity for connected devices" parameter on UI for the gateway device.

closed 2024-06-19 by imbeacon