thingsboard/thingsboard-gateway #1126: [HELP]Install failing in windows by python3

kiraten191010, 2023-06-14:
Hello , Is this  "thingsboard-gateway" is possible to install in Windows10 by python3 ?

I use command     pip3 install thingsboard-gateway

but it keep failing.

It always have this error

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for grpcio
  Running setup.py clean for grpcio
Failed to build grpcio
ERROR: Could not build wheels for grpcio, which is required to install pyproject.toml-based projects

samson0v, 2023-07-06:
Hi @kiraten191010, thanks for your interest in ThingsBoard IoT Gateway!
Try to use this solution https://stackoverflow.com/questions/56357794/unable-to-install-grpcio-using-pip-install-grpcio
Also, try to google: how to install grpcio on Windows if the above doesn't help.

samson0v, 2023-09-29:
Hi @kiraten191010, can we close the issue?

kiraten191010, 2023-10-02:
Thank you very much

closed 2023-10-02 by kiraten191010