thingsboard/thingsboard-gateway #1129: [HELP] Can't send RPC command to gateway after 5 minutes

UprightNeether, 2023-06-20:
Hello tbGateway team:
   When docker container tb-gateway started. If I send a RPC command to gateway every 1 minute,it will be successfully executed
  Then if stop calling and wait for 5 minutes,nothing will be send to gateway and thingsboard will print log like this

`2023-06-20 11:31:14,334 [tb-rule-engine-consumer-40-thread-23 | QK(Main,TB_RULE_ENGINE,system)-6] INFO  o.t.s.s.q.DefaultTbRuleEngineConsumerService - [d0760ae0-b425-11ed-ac44-7beb42f012a8] Failed to process message:
 TbMsg(queueName=Main, id=0ed144ac-c23d-4853-8d7b-c9ec3a557585, ts=1687231874327, type=RPC_CALL_FROM_SERVER_TO_DEVICE, originator=d1b03bc0-0c0d-11ee-afa6-e72c4d533fe1, customerId=null
, metaData=TbMsgMetaData(data={deviceType=ElectricityMeter, originServiceId=zjypwy-app001, requestUUID=e9f63a27-3d54-45bb-9db3-070196aee828, expirationTime=1687231874325, persistent=false
, deviceName=120015206173020054, oneway=false}), dataType=JSON, data={"error":"NO_ACTIVE_CONNECTION"}, ruleChainId=d0c47900-b425-11ed-ac44-7beb42f012a8
, ruleNodeId=d0dcbbf0-b425-11ed-ac44-7beb42f012a8, ctx=org.thingsboard.server.common.msg.TbMsgProcessingCtx@a758973
, callback=org.thingsboard.server.common.msg.queue.TbMsgCallback$1@2fc0bac9), Last Rule Node: null`

I pull data from device and push to thingsboard every 15minutes,it works well,so I dont think the network was broken

samson0v, 2023-07-06:
Hi @UprightNeether, thanks for your interest in ThingsBoard IoT Gateway!
Try to set `Overwrite activity time for connected device` to true on the Gateway device:
[image]

mdameda, 2023-07-12:
I have the same problem. Solution explained does not work (using "Overwrite activity time for connected device")

Any solution ?

Altamirano426, 2023-07-13:
hello, 
same problem here, the solution provided, "Overwrite activity time for connected device" doesn't work

skewty, 2023-09-07:
I think this is an upstream ThingsBoard platform issue.  I have checked gateway code right through to the line that pushes data out MQTT in the gateway (all correct).  The widget system using upstream's DeviceService in TypeScript / Angular component is getting a 500 series (server) error / no response. The response was pushed out via MQTT in less than 1 second so it wasn't a timeout.  This is intermittent.  Restarting my gateway sometimes fixes it -- not even always.

imbeacon, 2024-06-19:
This issue relates to the session timeout parameter on ThingsBoard, it was fixed in the ThingsBoard 3.7 version. Please try to use it, but don't forget to enable overwrite device activity time for connected devices, to allow the gateway to manage activity for related devices. It is necessary to have it enabled for devices, connected through connectors that cannot report about the device activity like TCP/UDP, serial, etc.

closed 2024-06-19 by imbeacon

UprightNeether, 2024-06-19:
> This issue relates to the session timeout parameter on ThingsBoard, it was fixed in the ThingsBoard 3.7 version. Please try to use it, but don't forget to enable overwrite device activity time for [...]

Many thanks to you and team mambers,but It's too risky to me to upgrade to 3.7,because I madn a lot of changes,so which commit shuold I refer to?

imbeacon, 2024-06-19:
@UprightNeether you may find related changes in the following PR - https://github.com/thingsboard/thingsboard/pull/10497