thingsboard/thingsboard-gateway #1130: [BUG]

guoziqin, 2023-06-21:
**Describe the bug**
Clear and concise description of what the bug is.

**Connector name (If bug in the some connector): 
ModbusConnector

**Error traceback (If available):**
```
ConvertWorker close error
```

**Versions (please complete the following information):**
 - OS: MacOS 13 m2 Max
 - Thingsboard IoT Gateway version 3.2
 - Python version: 3.10

description:
thingsboard cloud update shared attribute  trigger thingsborad gateway update remoteconfiguration callback __apply_new_connectors_configuration method,  self.__gateway.available_connectors[connector_name].close() 
ModbusConnector close methods only close socket ConvertWoker thread not close? this is bug ?

guoziqin, 2023-06-21:
[image]
thread debugger

guoziqin, 2023-06-21:
load_slave in put queue,but close no execute queue.clear()

samson0v, 2023-07-07:
Hi @guoziqin, thanks for your interest in ThingsBoard IoT Gateway!
Can you, please, describe your bug more clearly?
Also, write steps to reproduce, pls.

samson0v, 2023-09-29:
Hi @guoziqin, can we close the issue?

closed 2023-10-13 by samson0v