thingsboard/thingsboard-gateway #1137: [BUG] tb gateway shut down

zhbtman123, 2023-07-05:
**Describe the bug**
|ERROR| - [tb_utility.py] - tb_utility - validate_converted_data - 69 - deviceType is empty in data: {"deviceName": "test", "deviceType": "", "attributes": [{"model": "T1000"}, {"T1000": "on"}], "telemetry": [{"temperature": "4172"}, {"humidity": "58"}, {"combine": "58:4172"}]}"
2023-07-05 22:47:54 ""2023-07-05 14:47:54" - |ERROR| - [tb_gateway_service.py] - tb_gateway_service - __send_to_storage - 389 - Data from MQTT Broker Connector connector is invalid."

-----------------------------------------------------

I setup gateway by docker lastest version.

when after this error gateway container must be restart to working ..else it's not working ..I dont know why

zhbtman123, 2023-07-05:
2023-07-05 22:53:39 ""2023-07-05 14:53:39" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 167 - Number of accepted mapping handlers: 3"
2023-07-05 22:53:39 ""2023-07-05 14:53:39" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 171 - Number of rejected mapping handlers: 0"
2023-07-05 22:53:39 ""2023-07-05 14:53:39" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 167 - Number of accepted serverSideRpc handlers: 2"
2023-07-05 22:53:39 ""2023-07-05 14:53:39" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 171 - Number of rejected serverSideRpc handlers: 0"
2023-07-05 22:53:39 ""2023-07-05 14:53:39" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 167 - Number of accepted connectRequests handlers: 2"
2023-07-05 22:53:39 ""2023-07-05 14:53:39" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 171 - Number of rejected connectRequests handlers: 0"
2023-07-05 22:53:40 ""2023-07-05 14:53:40" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 167 - Number of accepted disconnectRequests handlers: 2"
2023-07-05 22:53:40 ""2023-07-05 14:53:40" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 171 - Number of rejected disconnectRequests handlers: 0"
2023-07-05 22:53:40 ""2023-07-05 14:53:40" - |ERROR| - [mqtt_connector.py] - mqtt_connector - load_handlers - 152 - Mandatory key 'topicExpression' missing from attributeRequests handler: {"retain": false, "topicFilter": "v1/devices/me/attributes/request", "deviceNameTopicExpression": "${SerialNumber}", "attributeNameJsonExpression": "${sensorModel}"}"
2023-07-05 22:53:40 ""2023-07-05 14:53:40" - |ERROR| - [mqtt_connector.py] - mqtt_connector - load_handlers - 152 - Mandatory key 'valueExpression' missing from attributeRequests handler: {"retain": false, "topicFilter": "v1/devices/me/attributes/request", "deviceNameTopicExpression": "${SerialNumber}", "attributeNameJsonExpression": "${sensorModel}"}"
2023-07-05 22:53:40 ""2023-07-05 14:53:40" - |ERROR| - [mqtt_connector.py] - mqtt_connector - load_handlers - 159 - attributeRequests handler is missing some mandatory keys => rejected: {"retain": false, "topicFilter": "v1/devices/me/attributes/request", "deviceNameTopicExpression": "${SerialNumber}", "attributeNameJsonExpression": "${sensorModel}"}"
2023-07-05 22:53:40 ""2023-07-05 14:53:40" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 167 - Number of accepted attributeRequests handlers: 0"
2023-07-05 22:53:40 ""2023-07-05 14:53:40" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 171 - Number of rejected attributeRequests handlers: 1"
2023-07-05 22:53:40 ""2023-07-05 14:53:40" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 167 - Number of accepted attributeUpdates handlers: 1"
2023-07-05 22:53:40 ""2023-07-05 14:53:40" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 171 - Number of rejected attributeUpdates handlers: 0"
2023-07-05 22:53:40 ""2023-07-05 14:53:40" - |INFO| - [tb_device_mqtt.py] - tb_device_mqtt - _on_connect - 138 - connection SUCCESS"
2023-07-05 22:53:40 ""2023-07-05 14:53:40" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 152 - Gateway started."
2023-07-05 22:53:40 ""2023-07-05 14:53:40" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_connect - 244 - MQTT Broker Connector connected to 10.4.10.206:9883 - successfully."
2023-07-05 22:53:40 ""2023-07-05 14:53:40" - |INFO| - [tb_loader.py] - tb_loader - import_module - 66 - Import JsonMqttUplinkConverter from /usr/local/lib/python3.7/site-packages/thingsboard_gateway-2.9-py3.7.egg/thingsboard_gateway/connectors/mqtt."
2023-07-05 22:53:40 ""2023-07-05 14:53:40" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_connect - 286 - Connector "MQTT Broker Connector" subscribe to /sensor/data"
2023-07-05 22:53:40 ""2023-07-05 14:53:40" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_connect - 286 - Connector "MQTT Broker Connector" subscribe to /sensor/+/data"
2023-07-05 22:53:40 ""2023-07-05 14:53:40" - |INFO| - [tb_loader.py] - tb_loader - import_module - 66 - Import CustomMqttUplinkConverter from /usr/local/lib/python3.7/site-packages/thingsboard_gateway-2.9-py3.7.egg/thingsboard_gateway/extensions/mqtt."
2023-07-05 22:53:40 ""2023-07-05 14:53:40" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_connect - 286 - Connector "MQTT Broker Connector" subscribe to /custom/sensors/+"
2023-07-05 22:53:40 ""2023-07-05 14:53:40" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 326 - ()"
2023-07-05 22:53:40 ""2023-07-05 14:53:40" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 335 - "MQTT Broker Connector" subscription success to topic /sensor/data, subscription message id = 1"
2023-07-05 22:53:40 ""2023-07-05 14:53:40" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 326 - ()"
2023-07-05 22:53:40 ""2023-07-05 14:53:40" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 335 - "MQTT Broker Connector" subscription success to topic /sensor/+/data, subscription message id = 2"
2023-07-05 22:53:40 ""2023-07-05 14:53:40" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 326 - ()"
2023-07-05 22:53:40 ""2023-07-05 14:53:40" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 335 - "MQTT Broker Connector" subscription success to topic /custom/sensors/+, subscription message id = 3"
[... cut ...]
2023-07-05 22:53:40 ""2023-07-05 14:53:40" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 335 - "MQTT Broker Connector" subscription success to topic sensor/+/disconnect, subscription message id = 7"
2023-07-05 22:53:41 ""2023-07-05 14:53:41" - |INFO| - [tb_updater.py] - tb_updater - check_for_new_version - 78 - 3.3"
2023-07-05 22:53:41 ""2023-07-05 14:53:41" - |INFO| - [tb_updater.py] - tb_updater - check_for_new_version - 81 - 
2023-07-05 22:53:41 
2023-07-05 22:53:41 [===UPDATE===]
2023-07-05 22:53:41 
2023-07-05 22:53:41  New version 3.3 is available! 
2023-07-05 22:53:41 
2023-07-05 22:53:41 [===UPDATE===]
2023-07-05 22:53:41 "

samson0v, 2023-07-06:
Hi @zhbtman123, thanks for your interest in ThingsBoard IoT Gateway!
Please send your configuration files (mqqt.json and tb_gateway.yml) and the data you send to Gateway.

samson0v, 2023-09-29:
Hi @zhbtman123, is this issue still actual for you or we can close the issue?

closed 2023-10-13 by samson0v