thingsboard/thingsboard-gateway #1144: [HELP] Can't send RPC command {"error":"NO_ACTIVE_CONNECTION"}

Altamirano426, 2023-07-07:
Hello everybody,
When the Docker container for tb-gateway (version 3.2) is started, if I send an RPC command to a device provisioned on ThingsBoard(latest version), I am correctly notified on the gateway. However when the gateway is restarted, RPC messages are no longer notified to the devices through the gateway. In such a scenario, ThingsBoard returns a connection timeout error and prints the following log.

2023-07-07 13:48:28,657 [device-dispatcher-1-1] DEBUG o.t.server.actors.TbActorMailbox - [DEVICE|715b8fd0-1cad-11ee-81c2-9146c2529dc6] Going to process message: TimeoutMsg(id=2, timeout=4966)
2023-07-07 13:48:28,657 [device-dispatcher-1-1] DEBUG o.t.s.a.service.ContextAwareActor - Processing msg: TimeoutMsg(id=2, timeout=4966)

2023-07-07 13:48:28,688 [device-dispatcher-1-1] DEBUG o.t.s.a.d.DeviceActorMessageProcessor - [715b8fd0-1cad-11ee-81c2-9146c2529dc6] RPC request [2] timeout detected!

2023-07-07 13:48:28,729 [JpaExecutorService-13-5] DEBUG o.t.server.dao.sql.JpaAbstractDao - Saving entity AuditLogEntity(tenantId=21050430-1c12-11ee-8fe7-4d4a58887b1f, customerId=13814000-1dd2-11b2-8080-808080808080, entityType=DEVICE, entityId=715b8fd0-1cad-11ee-81c2-9146c2529dc6, entityName=PA1_2_2, userId=21936180-1c12-11ee-8fe7-4d4a58887b1f, userName=tenant@thingsboard.org, actionType=RPC_CALL, actionData={"entityId":"715b8fd0-1cad-11ee-81c2-9146c2529dc6","oneWay":true,"method":"downlinkCommand","params":"{\"entityId\":\"715b8fd0-1cad-11ee-81c2-9146c2529dc6\",\"deviceId\":\"00137a100000a988\",\"payload\":\"04000100000000000001\"}"}, actionStatus=FAILURE, actionFailureDetails=RPC Error: NO_ACTIVE_CONNECTION)

2023-07-07 13:48:28,762 [tb-rule-engine-consumer-46-thread-6 | QK(Main,TB_RULE_ENGINE,system)-5] DEBUG o.t.s.s.q.p.BurstTbRuleEngineSubmitStrategy - [Main] submitting [1] messages to rule engine

2023-07-07 13:48:28,764 [app-dispatcher-54-thread-1] DEBUG o.t.server.actors.TbActorMailbox - [TENANT|13814000-1dd2-11b2-8080-808080808080] Going to process message: QueueToRuleEngineMsg(super=TbRuleEngineActorMsg(msg=TbMsg(queueName=Main, id=99617226-de2b-40f9-ada2-330cc1cb1646, ts=1688737708745, type=RPC_CALL_FROM_SERVER_TO_DEVICE, originator=715b8fd0-1cad-11ee-81c2-9146c2529dc6, customerId=null, metaData=TbMsgMetaData(data={deviceType=waterDetector, originServiceId=db596bf4670e, requestUUID=96781a42-47cd-4346-ac65-7a66090a8d3a, expirationTime=1688737708656, persistent=false, deviceName=PA1_2_2, oneway=true}), dataType=JSON, data={"error":"NO_ACTIVE_CONNECTION"}, ruleChainId=215f0b10-1c12-11ee-8fe7-4d4a58887b1f, ruleNodeId=2172ba20-1c12-11ee-8fe7-4d4a58887b1f, ctx=org.thingsboard.server.common.msg.TbMsgProcessingCtx@131cf2e1, callback=org.thingsboard.server.service.queue.TbMsgPackCallback@1f2664c1)), tenantId=21050430-1c12-11ee-8fe7-4d4a58887b1f, relationTypes=[Failure], failureMessage=NO_ACTIVE_CONNECTION)

mdameda, 2023-07-12:
I have the same problem. Solution using "**Overwrite activity time for connected device**", does not work

samson0v, 2023-07-14:
Hi @Altamirano426 @mdameda, do you have enabled Remote Logging?

Altamirano426, 2023-07-14:
hi @samson0v, yes i have all the logs in case it's needed.

tatuiot, 2024-06-30:
Hi Team I am running RPi device which is the mqtt broker for the Gateway itself. It is working fine, at least that part. I have the following issue, I can send telemetry MQTT->GW->TB->Dashboard.
But I cannot receive any RPC call from server. I have the default config. and remote logging (which has a handler issue).
and I am wondering what am I missing here , why I cannot receive any RPC from server/dashboard round switch.

This is the portion of my mqtt.json config related to serverSideRpc. and a second doubt would be, is needed the rpc_connector.json connector to make it work? I am kind of lost here with this...  thanks I am not understanding thanks in advance for any comments or suggestion.

 "serverSideRpc": [
    {
      "deviceNameFilter": ".*",
      "methodFilter": "echo",
      "requestTopicExpression": "/sensor/${deviceName}/request/${methodName}/${requestId}",
      "responseTopicExpression": "/sensor/${deviceName}/response/${methodName}/${requestId}",
      "responseTimeout": 10000,
      "valueExpression": "${params}"
    },
    {
      "deviceNameFilter": ".*",
      "methodFilter": "no-reply",
      "requestTopicExpression": "/sensor/${deviceName}/request/${methodName}/${requestId}",
      "valueExpression": "${params}"
    }
  ],