thingsboard/thingsboard-gateway #1145: [BUG] module 'urllib3.util.ssl_' has no attribute 'DEFAULT_CIPHERS'

tomtommahout, 2023-07-08:
**Describe the bug**
urllib2>2 throws the error below. Pin to urllib3<=2 resolves the issue for now. 
Appears to be from requests lib updates (https://github.com/psf/requests/issues/6443)

**Connector name (If bug in the some connector):**
Request Connector (Custom connector in the error listed below, but not changed from the default request_connector.py)

**Error traceback (If available):**
```

""2023-07-08 08:04:46" - |ERROR| - [tb_loader.py] - tb_loader - import_module - 75 - module 'urllib3.util.ssl_' has no attribute 'DEFAULT_CIPHERS'"
Traceback (most recent call last):
  File "/thingsboard_gateway/tb_utility/tb_loader.py", line 65, in import_module
    module_spec.loader.exec_module(module)
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/thingsboard_gateway/extensions/request/custom_request_connector.py", line 43, in <module>
    requests.packages.urllib3.util.ssl_.DEFAULT_CIPHERS += ':ADH-AES128-SHA256'
AttributeError: module 'urllib3.util.ssl_' has no attribute 'DEFAULT_CIPHERS'
```

**Versions (please complete the following information):**
 - OS: Docker on Windows
 - Thingsboard IoT Gateway version: 3.3 
 - Python version: 3.10.12

KubaDavid, 2023-07-12:
I just got this as well in the same situation.
Resolution as per this comment: https://github.com/psf/requests/issues/6443#issuecomment-1535667256

closed 2023-08-11 by imbeacon

z1z2, 2023-08-25:
the problem still exist in the docker version.
https://hub.docker.com/r/thingsboard/tb-gateway