thingsboard/thingsboard-gateway #1148: [HELP]

sukang00, 2023-07-12:
**Describe the issue**
I started two gateways using Docker, but only the first gateway [. tb gateway] started normally, and the second gateway [. tb gateway mdj] has a connection status of "Inactive". Attached is my configuration file. Could you please help me check what the problem is? Strangely, the "/mdj/Sleep" in the mqtt.json configuration of the second gateway [. tb gateway mdj] can take effect.

My startup command is
```
sudo docker run -it -e host=172.168.105.7 -e port=1883 -e accessToken=vha045MY96DFuwy7m841 -v ~/.tb-gateway/logs:/thingsboard_gateway/logs -v ~/.tb-gateway/extensions:/thingsboard_gateway/extensions -v ~/.tb-gateway/config:/thingsboard_gateway/config --name tb-gateway --restart always thingsboard/tb-gateway

sudo docker run -it -e host=172.168.105.7 -e port=1883 -e accessToken=gyDkWqtZAV8AA5W08SSQ -v ~/.tb-gateway-mdj/logs:/thingsboard_gateway/logs -v ~/.tb-gateway-mdj/extensions:/thingsboard_gateway/extensions -v ~/.tb-gateway-mdj/config:/thingsboard_gateway/config --name tb-gateway-mdj --restart always thingsboard/tb-gateway
```
If you want to build multiple gateways, which configurations should be modified to prevent gateway configuration conflicts? Can you point them out to me

**Configuration** (Attach your configuration file)
**Notate: Remove Access token from file if you want to attach a tb_gateway.yaml**

[config.zip](https://github.com/thingsboard/thingsboard-gateway/files/12025813/config.zip)

**Connector name (If you need help with some connector/converter):**
[e.g. MQTT Connector]

**Error traceback (If it was raised):**
```
'deviceName'
Traceback (most recent call last):
  File "<input>", line 2, in <module>
KeyError: 'deviceName'
```

**Versions (please complete the following information):**
 - OS: Ubuntu 20.04
 - Thingsboard IoT Gateway version 3.2

samson0v, 2023-08-11:
Hi @sukang00, thanks for your interest in ThingsBoard IoT Gateway!
I downloaded your `config.zip` but there is not files.
Please, send config files and debug logs (change all `INFO` to `DEBUG` level in logs.conf file).

samson0v, 2023-09-18:
Hi @sukang00, is this bug still actual for you or can we close it?

closed 2023-10-13 by samson0v