thingsboard/thingsboard-gateway #1151: [BUG]

ultimate-smile, 2023-07-19:
**Describe the bug**
Clear and concise description of what the bug is.
Directly read the serial device and report an error:
'[Input/Output] No Response received from the remote slave/Unable to decode response'

**Connector name (If bug in the some connector):**
[e.g. OPC-UA Connector]
{
  "master": {
    "slaves": [
      {
        "type": "serial",
        "method": "rtu",
        "port": "/dev/cu.usbmodem1414201",
        "baudrate": 115200,
        "timeout": 35,
        "stopbits": 1,
        "bytesize": 8,
        "byteOrder": "BIG",
        "wordOrder": "BIG",
        "retries": true,
        "retryOnEmpty": true,
        "retryOnInvalid": true,
        "pollPeriod": 5000,
        "unitId": 1,
        "deviceName": "Modbus Temperature Sensor",
        "sendDataOnlyOnChange": false,
        "connectAttemptTimeMs": 5000,
        "connectAttemptCount": 5,
        "waitAfterFailedAttemptsMs": 300000,
        "attributes": [
          {
            "tag": "dht11_temperature",
            "type": "bits",
            "functionCode": 1,
            "objectsCount": 1,
            "address": 0
          }
        ]
      }
    ]
  }
}
**Error traceback (If available):**
```
'[Input/Output] No Response received from the remote slave/Unable to decode response'
```

**Versions (please complete the following information):**
 - OS: macOS
 - Thingsboard IoT Gateway version master
 - Python version[3.11]

samson0v, 2023-08-11:
Hi @ultimate-smile, thanks for your interest in ThingsBoard IoT Gateway!
From the error, it seems that some of the following params are invalid:
- port
- baudrate
- stopbits
- bytesize

Please, make sure that the params about correspond to the device settings.

samson0v, 2023-09-29:
Hi @ultimate-smile, can we close the issue?

closed 2023-10-13 by samson0v