thingsboard/thingsboard-gateway #1152: [BUG]

mincequi, 2023-07-25:
The request connector does not parse this json correctly:
```
{
"temperature":33.46,
"overtemperature":false,
"tmp":{"tC":33.46,"tF":92.23, "is_valid":true},
"temperature_status":"Normal",
"ext_sensors":{"temperature_unit":"C"},
"ext_temperature":{
    "0":{
        "hwID":"284b0b4d182006ed","tC":47.62,"tF":117.73
    }
}
}
```

```
        "attributes": [
          {
            "key": "temp1",
            "type": "double",
            "value": "${tmp.tC}"
          },
          {
            "key": "temp2",
            "type": "double",
            "value": "${ext_temperature.0.tC}"
          }
        ]
```

I can actually parse tmp.tC, however ext_temperature.0.tC is not parsed.
Thanks in advance.

samson0v, 2023-08-11:
Hi @mincequi, thanks for your interest in ThingsBoard IoT Gateway!
You can use the following expression: `"value": "${ext_temperature.*.tC}"`
[image]
You can't use number as a key of dict because parser expect array instead of a dict.

samson0v, 2023-09-29:
Hi @mincequi, can we close the issue?

mincequi, 2023-09-29:
@samson0v yes, this can be closed.

Thanks a lot!

closed 2023-10-13 by samson0v