thingsboard/thingsboard-gateway #1154: [HELP]

seanblaney, 2023-07-27:
**Describe the issue**
I have successfully connected to my FTP server and the device is added into ThingsBoard.

I'm struggling to parse a CSV for attributes and telemetry from within the FTP connector, would anyone be able to give me an example based on the example CSV attached?

Specifically, I cannot understand how to parse for attributes and telemetry given the example, I.e. how to change the example below to ingest my data.

       "attributes": [
         {
           "type": "int",
           "key": "key",
           "value": "${temp}"
         }
       ],
       "timeseries": [
         {
           "type": "int",
           "key": "${hum}",
           "value": "${temp}"
         },
         {
           "type": "int",
           "key": "temp",
           "value": "${hum}"
         }
       ]

Many thanks.

[89916-readings-current.csv](https://github.com/thingsboard/thingsboard-gateway/files/12182893/89916-readings-current.csv)

**Versions (please complete the following information):**
 - OS: Ubuntu 20.4.6 LTS
 - Thingsboard IoT Gateway version: 3.3
 - Python version: 3.8.10

samson0v, 2023-08-11:
Hi @seanblaney, thanks for your interest in ThingsBoard IoT Gateway! 
Unfortunately, the FTP connector doesn't support .csv parsing.

seanblaney, 2023-08-11:
Vitalii,

Ok, thank you for coming back to me. I will take another approach and run a Python script via a cron job to parse the CSV and post over HTTPS to Thingsboard.

Thanks,

Sean

samson0v, 2023-08-11:
@seanblaney, ok.
Can we close the issue?

closed 2023-08-11 by seanblaney