thingsboard/thingsboard-gateway #1157: [BUG]

zwoabier, 2023-08-03:
**Describe the bug**
The docker image is not working since it is installing python modules at runtime. You can see in the attached logs, that the mmh3 module is installed successfully `Successfully installed mmh3-4.0.1` but later where it tries to import it it still fails.
```
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/sdk_utils.py", line 29, in <module>
    from mmh3 import hash, hash128
ModuleNotFoundError: No module named 'mmh3'
```

I guess this is because you install the dependencies at runtime but you do not update path variable. So the runing sytem does not know about the dependencies that were installed at runtime.
Maybe this stackoverflow discussion is helpfull
https://stackoverflow.com/questions/17234762/how-to-install-and-import-python-modules-at-runtime

If you do not remove the crashed container and restart it it just will work, but that shouldn't be the solution since this breaks almost all deployments.

I really would appreciate if you consider to solve this and the related topic https://github.com/thingsboard/thingsboard-gateway/issues/1075 because installing packages at runtime does really not sound like a good idea

To reproduce this error you just have to run the `docker run` command from the docs

```
docker run -it -v ~/.tb-gateway/logs:/var/log/thingsboard-gateway -v ~/.tb-gateway/extensions:/var/lib/thingsboard_gateway/extensions -v ~/.tb-gateway/config:/etc/thingsboard-gateway/config --name tb-gateway --restart always thingsboard/tb-gateway
```

**Connector name (If bug in the some connector):**
[e.g. OPC-UA Connector]

**Error traceback (If available):**
```
Unable to find image 'thingsboard/tb-gateway:latest' locally
latest: Pulling from thingsboard/tb-gateway
5b5fe70539cd: Pull complete 
f4b0e4004dc0: Pull complete 
c5424f0ac885: Pull complete 
9d21fe1624ec: Pull complete 
de0194aa1c21: Pull complete 
b010216a2cd9: Pull complete 
828c03412214: Pull complete 
25e0d8db7d34: Pull complete 
30e030194c6a: Pull complete 
4f4fb700ef54: Pull complete 
417a27a35f0c: Pull complete 
Digest: sha256:0b8ab29baf83c306ca4868d27919a7bbf3108a8e925365f46441f9f34a871264
Status: Downloaded newer image for thingsboard/tb-gateway:latest
Collecting mmh3
  Downloading mmh3-4.0.1-cp310-cp310-manylinux_2_5_x86_64.manylinux1_x86_64.manylinux_2_17_x86_64.manylinux2014_x86_64.whl (72 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 72.6/72.6 kB 1.9 MB/s eta 0:00:00
Installing collected packages: mmh3
Successfully installed mmh3-4.0.1
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv

[notice] A new release of pip is available: 23.1.2 -> 23.2.1
[notice] To update, run: pip install --upgrade pip
tb-mqtt-client library not found - installing...
Requirement already satisfied: tb-mqtt-client in /usr/local/lib/python3.10/site-packages (1.5)
Requirement already satisfied: tb-paho-mqtt-client>=1.6.3 in /usr/local/lib/python3.10/site-packages (from tb-mqtt-client) (1.6.3)
Requirement already satisfied: requests>=2.31.0 in /usr/local/lib/python3.10/site-packages (from tb-mqtt-client) (2.31.0)
Requirement already satisfied: simplejson in /usr/local/lib/python3.10/site-packages (from tb-mqtt-client) (3.19.1)
Requirement already satisfied: charset-normalizer<4,>=2 in /usr/local/lib/python3.10/site-packages (from requests>=2.31.0->tb-mqtt-client) (3.1.0)
Requirement already satisfied: idna<4,>=2.5 in /usr/local/lib/python3.10/site-packages (from requests>=2.31.0->tb-mqtt-client) (3.4)
Requirement already satisfied: urllib3<3,>=1.21.1 in /usr/local/lib/python3.10/site-packages (from requests>=2.31.0->tb-mqtt-client) (2.0.3)
Requirement already satisfied: certifi>=2017.4.17 in /usr/local/lib/python3.10/site-packages (from requests>=2.31.0->tb-mqtt-client) (2023.5.7)
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv

[notice] A new release of pip is available: 23.1.2 -> 23.2.1
[notice] To update, run: pip install --upgrade pip
Requirement already satisfied: mmh3 in /root/.local/lib/python3.10/site-packages (4.0.1)
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv

[notice] A new release of pip is available: 23.1.2 -> 23.2.1
[notice] To update, run: pip install --upgrade pip
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/sdk_utils.py", line 29, in <module>
    from mmh3 import hash, hash128
ModuleNotFoundError: No module named 'mmh3'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/thingsboard_gateway/gateway/tb_client.py", line 28, in <module>
    from tb_gateway_mqtt import TBGatewayMqttClient, TBDeviceMqttClient
  File "/usr/local/lib/python3.10/site-packages/tb_gateway_mqtt.py", line 19, in <module>
    from tb_device_mqtt import TBDeviceMqttClient
  File "/usr/local/lib/python3.10/site-packages/tb_device_mqtt.py", line 27, in <module>
    from sdk_utils import verify_checksum
  File "/usr/local/lib/python3.10/site-packages/sdk_utils.py", line 31, in <module>
    from pymmh3 import hash, hash128
ModuleNotFoundError: No module named 'pymmh3'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/sdk_utils.py", line 29, in <module>
    from mmh3 import hash, hash128
ModuleNotFoundError: No module named 'mmh3'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "//./thingsboard_gateway/tb_gateway.py", line 17, in <module>
    from thingsboard_gateway.gateway.tb_gateway_service import TBGatewayService
  File "/thingsboard_gateway/gateway/tb_gateway_service.py", line 41, in <module>
    from thingsboard_gateway.gateway.tb_client import TBClient
  File "/thingsboard_gateway/gateway/tb_client.py", line 32, in <module>
    from tb_gateway_mqtt import TBGatewayMqttClient, TBDeviceMqttClient
  File "/usr/local/lib/python3.10/site-packages/tb_gateway_mqtt.py", line 19, in <module>
    from tb_device_mqtt import TBDeviceMqttClient
  File "/usr/local/lib/python3.10/site-packages/tb_device_mqtt.py", line 27, in <module>
    from sdk_utils import verify_checksum
  File "/usr/local/lib/python3.10/site-packages/sdk_utils.py", line 31, in <module>
    from pymmh3 import hash, hash128
ModuleNotFoundError: No module named 'pymmhdocker on3'
```

**Versions (please complete the following information):**
 - OS: Docker version 24.0.5, build ced0996 on Ubuntu 22.04
 - Thingsboard IoT Gateway version 3.3
 - Python 3.10.6

samson0v, 2023-08-11:
Hi @zwoabier, thanks for your interest in ThingsBoard IoT Gateway!
We know about this limitation and will fix it in the nearest future.

LukaszDuss, 2023-08-17:
I got the same error.  Waiting for solution :)

zwoabier, 2023-08-17:
So the easiest workaround for now is to just clone the repo add `mmh3` to `requirements.txt` and build the image by yourself

samson0v, 2023-09-29:
Hi @zwoabier, thanks for your interest in ThingsBoard IoT Gateway!
This issue was fixed. Try the latest docker image build and let us know about the result.

closed 2023-10-13 by samson0v