thingsboard/thingsboard-gateway #1162: [HELP] Adding additional python library to custom connector

tomtommahout, 2023-08-13:
**Describe the issue**

I'm trying to build a custom Request connector/converter for Thingsboard Gateway using the XML2dict library. The request endpoint returns xml only - I’m attempting to convert it to json and use the existing request code to reduce the amount of custom code.

Thingsboard-gw is running in a linux container hosted on Windows. Adding the additional library throws an error and I can't work out how to install it within the container

I've tried adding the library in the custom connector file using the following code block, but get the following errors in the log:

`try:
    import XML2dict
except ImportError:
    print("TG ADDED - xml2dict not found - installing...")
    TBUtility.install_package('xml2dict')
    import XML2dict `

`""2023-08-12 04:20:57" - |ERROR| - [tb_loader.py] - tb_loader - import_module - 72 - No module named 'XML2dict'"
Traceback (most recent call last):
  File "/thingsboard_gateway/extensions/request/custom_request_connector.py", line 43, in <module>
    import XML2dict
ModuleNotFoundError: No module named 'XML2dict'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/thingsboard_gateway/tb_utility/tb_loader.py", line 65, in import_module
    module_spec.loader.exec_module(module)
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/thingsboard_gateway/extensions/request/custom_request_connector.py", line 47, in <module>
    import XML2dict
ModuleNotFoundError: No module named 'XML2dict' `

In the same code block - this import for the requests library works. I’m not sure why the import for the xml library does not. 

`try:
    from requests import Timeout, request
except ImportError:
    print("Requests library not found - installing...")
    TBUtility.install_package("requests")
    from requests import Timeout, request `

**Versions (please complete the following information):**
 - OS: [Docker on windows (Linux container)]
 - Thingsboard IoT Gateway version [e.g. latest]
 - Python version[e.g. 3.10]

samson0v, 2023-08-18:
Hi @tomtommahout, thanks for your interest in ThingsBoard IoT Gateway!
I can't find this library too)
Make sure you correctly write the library name!
If you are using this library XML2Dict 0.2.2, you have to use the following name:
`pip install XML2Dict`

tomtommahout, 2023-08-19:
Ah yes - my mistake! 

The api I’m querying is almost older than that library! Back to the drawing board for xml to json conversion.

Thanks for your help - and for the great product!

closed 2023-08-19 by tomtommahout