thingsboard/thingsboard-gateway #1163: [HELP] Is there a way to config clientSideRpc for mqtt connector

xiaomalover, 2023-08-14:
**Describe the issue**
 Is there a way to config clientSideRpc for mqtt connector

**Configuration** (Attach your configuration file)
**Notate: Remove Access token from file if you want to attach a tb_gateway.yaml**

**Connector name (If you need help with some connector/converter):**
MQTT Connector

**Error traceback (If it was raised):**
```
'deviceName'
Traceback (most recent call last):
  File "<input>", line 2, in <module>
KeyError: 'deviceName'
```

**Versions (please complete the following information):**
 - OS: [e.g. Ubuntu 20.04]
 - Thingsboard IoT Gateway version [e.g. 3.0.1]
 - Python version[e.g. 3.9]

samson0v, 2023-09-18:
Hi @xiaomalover, thanks for your interest in ThingsBoard IoT Gateway!
Unfortunately, Mqtt Connector doesn't have this functionality.

samson0v, 2023-09-29:
Hi @xiaomalover, can we close the issue?

closed 2023-10-13 by samson0v