thingsboard/thingsboard-gateway #1165: [BUG] ThingsBoard Gateway Widget - Save Button Not Functional

AIOTEL-sanjanaa, 2023-08-17:
**Describe the bug**
I am using demo.thingsboard.io connected to ThingsBoard Gateway version 3.3.4. I wanted to remotely configure my OPC-UA, so I tried using the Gateway widgets provided from the widget bundle. However, unfortunately, the save button does not seem to be working or is non-functional.
[image]

**Versions (please complete the following information):**
 - Thingsboard IoT Gateway version - 3.3.4
 - Python version - 3

samson0v, 2023-08-18:
Hi @AIOTEL-sanjanaa, thanks for your interest in ThingsBoard IoT Gateway!
We will release a fully new Gateway Remote Configuration dashboard soon.
The old widgets will remove.

AIOTEL-sanjanaa, 2023-08-18:
> Hi @AIOTEL-sanjanaa, thanks for your interest in ThingsBoard IoT Gateway! We will release a fully new Gateway Remote Configuration dashboard soon. The old widgets will remove.

Okay thank you.

closed 2023-09-18 by imbeacon