thingsboard/thingsboard-gateway #1168: [Question] Thingsboard Gateway deployment scenario

shashank-r25, 2023-08-21:
Hello,

What are the deployment scenarios for Thingsboard Gateway?
How many devices and requests can be handled?
Gateway should be deployed in a new server or within the existing server.
what is the persistent file storage format when it's stored in the gateway?
persistent file storage should be with the Thingsboard gateway&cloud server or in a separate server?

Thank You,

samson0v, 2023-09-18:
Hi @shashank-r25, thanks for your interest in ThingsBoard IoT Gateway!
You can deploy Gateway as another regular service (docker, source, deb, etc.).
About storages: https://thingsboard.io/docs/iot-gateway/configuration/#storage-configuration

shashank-r25, 2023-09-19:
Hi @samson0v, Thank you for the reply.

closed 2023-09-19 by shashank-r25