thingsboard/thingsboard-gateway #1171: [HELP] Remote Converter Update feature : JSON Configuration

AIOTEL-sanjanaa, 2023-08-30:
**Remote Converter Update:**
- In the [Remote Converter Update Documentation](https://thingsboard.io/docs/iot-gateway/guides/how-to-use-remote-converter-update/) there is no mention of the json format to follow to use the feature.
-  I tried connecting it to my gateway and send shared attributes for my OPC-UA connector. 
[image]
- But the configuration is not updated.
- This is my logs from thingsboard-gateway:
[image]
- This is the tb_gateway.yaml configuration.
[image]

**Questions:**
- What is the proper format of shared attributes?
- Is there any other configurations that I have to add in the tb_gateway.yaml file?
- What additional steps are required for a functional Remote Converter Update feature?

**Versions:**
 - OS: Ubuntu
 - Thingsboard IoT Gateway version: 3.3
 - Thingsboard: 3.5
 - Python version: 3.10

samson0v, 2023-09-18:
Hi @AIOTEL-sanjanaa, thanks for your interest in ThingsBoard IoT Gateway!
[image]

AIOTEL-sanjanaa, 2023-09-19:
Okay, thank you

closed 2023-09-19 by AIOTEL-sanjanaa