thingsboard/thingsboard-gateway #1172: [HELP] MODBUS RTU over TCP multiple devices on one TCP connection

LukaszDuss, 2023-08-31:
**Describe the issue**
I need to configure 15 sensors that are connected to device with modbus rtu over tcp. I can only open one TCP connection due to limitation of my device. 

Is it possible/how to configure modbus.json (v3) like modbus.json (v2), so my device will open just one TCP connection but still recognize 15 different sensors??

**Configuration** (Attach your configuration file)
**Notate: Remove Access token from file if you want to attach a tb_gateway.yaml**

**Connector name (If you need help with some connector/converter):**
MODBUS TCP with RTU

**Versions (please complete the following information):**
 - OS: Ubuntu 22.04
 - Thingsboard IoT Gateway v3.3
 - Python version 3.10.12

samson0v, 2023-09-18:
Hi @LukaszDuss, thanks for your interest in ThingsBoard IoT Gateway!
We know about this limitation and we will fix it in the next release.

LukaszDuss, 2023-09-18:
> Hi @LukaszDuss, thanks for your interest in ThingsBoard IoT Gateway! We know about this limitation and we will fix it in the next release.

OK

closed 2023-09-18 by LukaszDuss

clementlgl, 2024-05-28:
@samson0v Is it now possible to configure the gateway to fetch data from multiple sensors through one TCP connection ?

Thank you